MagicMirrorOrg/MagicMirror #2119: cannot start the mirror

wirdman03, 2020-08-28:
`pi@raspberrypi:~/MagicMirror` $ npm start

> magicmirror@2.12.0 start /home/pi/MagicMirror
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

[2020-08-28 14:24:26.604] [LOG]    Starting MagicMirror: v2.12.0
[2020-08-28 14:24:26.619] [LOG]    Loading config ...
[2020-08-28 14:24:26.632] [LOG]    Loading module helpers ...
[2020-08-28 14:24:26.635] [LOG]    No helper found for module: alert.
[2020-08-28 14:24:26.736] [LOG]    Initializing new module helper ...
[2020-08-28 14:24:26.738] [LOG]    Module helper loaded: updatenotification
[2020-08-28 14:24:26.740] [LOG]    No helper found for module: clock.
[2020-08-28 14:24:27.532] [LOG]    Initializing new module helper ...
[2020-08-28 14:24:27.532] [LOG]    Module helper loaded: calendar
[2020-08-28 14:24:27.534] [LOG]    No helper found for module: compliments.
[2020-08-28 14:24:27.535] [LOG]    No helper found for module: currentweather.
[2020-08-28 14:24:27.536] [LOG]    No helper found for module: weatherforecast.
[2020-08-28 14:24:29.823] [LOG]    Initializing new module helper ...
[2020-08-28 14:24:29.825] [LOG]    Module helper loaded: MMM-GoogleAssistant
[2020-08-28 14:24:29.828] [LOG]    No helper found for module: MMM-SwissCommute.
[2020-08-28 14:24:29.834] [ERROR]  WARNING! Could not load config file. Starting with default configuration. Error found: TypeError: Module is not a constructor
[2020-08-28 14:24:29.836] [LOG]    Loading module helpers ...
[2020-08-28 14:24:29.838] [LOG]    No helper found for module: alert.
[2020-08-28 14:24:29.845] [LOG]    Initializing new module helper ...
[2020-08-28 14:24:29.846] [LOG]    Module helper loaded: updatenotification
[2020-08-28 14:24:29.848] [LOG]    No helper found for module: clock.
[2020-08-28 14:24:29.849] [LOG]    Initializing new module helper ...
[2020-08-28 14:24:29.850] [LOG]    Module helper loaded: calendar
[2020-08-28 14:24:29.851] [LOG]    No helper found for module: compliments.
[2020-08-28 14:24:29.852] [LOG]    No helper found for module: currentweather.
[2020-08-28 14:24:29.853] [LOG]    No helper found for module: weatherforecast.
[2020-08-28 14:24:29.854] [LOG]    Initializing new module helper ...
[2020-08-28 14:24:29.855] [LOG]    Module helper loaded: MMM-GoogleAssistant
[2020-08-28 14:24:29.856] [LOG]    No helper found for module: MMM-SwissCommute.
[2020-08-28 14:24:29.857] [ERROR]  App threw an error during load
[2020-08-28 14:24:29.858] [ERROR]  TypeError: Module is not a constructor
    at loadModule (/home/pi/MagicMirror/js/app.js:125:12)
    at loadNextModule (/home/pi/MagicMirror/js/app.js:158:5)
    at /home/pi/MagicMirror/js/app.js:160:6
    at loadModule (/home/pi/MagicMirror/js/app.js:143:4)
    at loadNextModule (/home/pi/MagicMirror/js/app.js:158:5)
    at /home/pi/MagicMirror/js/app.js:160:6
    at Class.loaded (/home/pi/MagicMirror/js/node_helper.js:18:3)
    at loadModule (/home/pi/MagicMirror/js/app.js:141:6)
    at loadNextModule (/home/pi/MagicMirror/js/app.js:158:5)
    at /home/pi/MagicMirror/js/app.js:160:6
[2020-08-28 14:24:29.862] [ERROR]  Whoops! There was an uncaught exception...
[2020-08-28 14:24:29.863] [ERROR]  TypeError: Module is not a constructor
    at loadModule (/home/pi/MagicMirror/js/app.js:125:12)
    at loadNextModule (/home/pi/MagicMirror/js/app.js:158:5)
    at /home/pi/MagicMirror/js/app.js:160:6
    at loadModule (/home/pi/MagicMirror/js/app.js:143:4)
    at loadNextModule (/home/pi/MagicMirror/js/app.js:158:5)
    at /home/pi/MagicMirror/js/app.js:160:6
    at Class.loaded (/home/pi/MagicMirror/js/node_helper.js:18:3)
    at loadModule (/home/pi/MagicMirror/js/app.js:141:6)
    at loadNextModule (/home/pi/MagicMirror/js/app.js:158:5)
    at /home/pi/MagicMirror/js/app.js:160:6
[2020-08-28 14:24:29.871] [ERROR]  MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
[2020-08-28 14:24:29.872] [ERROR]  If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
[2020-08-28 14:24:30.028] [LOG]    Launching application.``

sdetweil, 2020-08-28:
@wirdman03   what other modules are listed in config.js?  after MMM-SwissCommute

seems there  is a problem with the module structure

wirdman03, 2020-08-28:
it is MMM-Assistant2Display

sdetweil, 2020-08-28:
is that the only one after MMM-SwissCommute?

wirdman03, 2020-08-28:
no got all of it

sdetweil, 2020-08-28:
ok, can u show the node_helper.js from the MMM-Assistant2Display folder?

bugsounet, 2020-08-28:
can you learn to read a log ?

[2020-08-28 **14:24:29.856**] [LOG] No helper found for module: MMM-SwissCommute.
[2020-08-28 **14:24:29.857**] [ERROR] App threw an error during load

so it's MMM-SwissCommute.

in all case, it's not an issur of MagicMirror core

sdetweil, 2020-08-28:
its after that 
[2020-08-28 14:24:29.856] [LOG] No helper found for module: MMM-SwissCommute.
(this is normal)

the NEXT module it tries to load, hasn't reported name yet.. but fails

still not MM problem..

sdetweil, 2020-09-07:
@wirdman03  did u resolve this problem?

closed 2020-09-18 by MichMich