MagicMirrorOrg/MagicMirror #2123: Can not change anything including adding modules

Plainbroke, 2020-09-04:
## I'm not sure if this is a bug

Here goes,
I thought I just had an issue with not being able to edit config.js and custom.css.
Now it seems nothing is allowed to change.
I just deleted config.js and custom.css (I have backups).
No need for backups,  Instead of saying the message about no config.js after a reboot,
All came up working in my old settings.  
This all seemed to start right after last update.
Any clues for me to look at or patches I am not finding?

rejas, 2020-09-04:
I had something similar once. No change I did on the filesystem was there after a reboot.
The cause was the SD card which was faulty and didnt allow any real write operations anymore. Had to replace the card.

Plainbroke, 2020-09-04:
Thats what I thought.   So making a new SD card right now.

Jeff Burdick

DivingIvan, 2020-09-07:
@Plainbroke,

I have had this happen to me before also. For some reason, the SD card just became readonly so I could no longer modify anything. My only option was to copy the SD card to a new one.

closed 2020-09-18 by MichMich