MagicMirrorOrg/MagicMirror #2124: Interest in OpenWeatherMap OneCall API and Hourly Forecasts

JimHutchins, 2020-09-07:
I wanted to use MagicMirror as a giant clock and weather info center. I loved what was already there in the weather module, but I wanted even more. I've implemented support for OpenWeatherMap's new onecall API endpoint. I've got the code working well enough for my purposes and I read the contribution guide so I'm posting an issue here. 

Is there enough interest in the module for me to clean up the code an submit it? As implemented right now, it is an addition to the standard Weather module and standard openweathermap.js provider. It definitely needs some polish before I would submit it, so I wanted to throw it out here before I did any more. Here's a screenshot of my implementation (including a lot of entries in custom.css).

[image]

sdetweil, 2020-09-07:
has been added already, https://github.com/MichMich/MagicMirror/pull/2078

switch to the develop branch to access the new code not yet released

this may not be the same as what u want

closed 2020-09-18 by MichMich

hangorazvan, 2021-04-21:
@JimHutchins Can you share with me your code please? New weather module It lacks a lot of information given by onecall

ThSGM, 2021-08-23:
@JimHutchins Just chiming in as well to say that I was interested in the script that generates the detailed weather info.

JimHutchins, 2021-11-22:
The files I modified are attached.

I suspect there may be a better (more official) way to do this, but it worked for me.

Sorry for the slow reply. I've been buried.

/* Magic Mirror Config Sample
 *
 * By Michael Teeuw https://michaelteeuw.nl
 * MIT Licensed.
 *
 * For more information on how you can configure this file
 * See https://github.com/MichMich/MagicMirror#configuration
 *
 */

var config = {
	address: "localhost", 	// Address to listen on, can be:
							// - "localhost", "127.0.0.1", "::1" to listen on loopback interface
							// - another specific IPv4/6 to listen on a specific interface
							// - "0.0.0.0", "::" to listen on any interface
							// Default, when address config is left out or empty, is "localhost"
	port: 8080,
	basePath: "/", 	// The URL path where MagicMirror is hosted. If you are using a Reverse proxy
					// you must set the sub path here. basePath must end with a /
	ipWhitelist: ["127.0.0.1", "::ffff:127.0.0.1", "::1"], 	// Set [] to allow all IP addresses
															// or add a specific IPv4 of 192.168.1.5 :
															// ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.1.5"],
															// or IPv4 range of 192.168.3.0 --> 192.168.3.15 use CIDR format :
															// ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.3.0/28"],

	useHttps: false, 		// Support HTTPS or not, default "false" will use HTTP
	httpsPrivateKey: "", 	// HTTPS private key path, only require when useHttps is true
	httpsCertificate: "", 	// HTTPS Certificate path, only require when useHttps is true

	language: "en",
	logLevel: ["INFO", "LOG", "WARN", "ERROR"],
	timeFormat: 24,
	units: "metric",
	// serverOnly:  true/false/"local" ,
	// local for armv6l processors, default
	//   starts serveronly and then starts chrome browser
	// false, default for all NON-armv6l devices
	// true, force serveronly mode, because you want to.. no UI on this device

	modules: [
		{
			module: "alert",
		},
		{
			module: "updatenotification",
			position: "top_bar"
		},
		{
			module: "clock",
			position: "top_left",
			config: {
				timeFormat: "12",
				dateFormat: "ddd, MMM D, YYYY",
				showPeriodUpper: "true",				
				showSunTimes: "true",
				lat: "39.000",
				lon: "-86.000"
				}
		},
		{
			module: "weather",
			position: "bottom_left",
			config: {
				weatherProvider: "openweathermap",
[... 614 lines cut ...]
/**
 * Static method to register a new weather provider.
 */
WeatherProvider.register = function (providerIdentifier, providerDetails) {
	WeatherProvider.providers[providerIdentifier.toLowerCase()] = WeatherProvider.extend(providerDetails);
};

/**
 * Static method to initialize a new weather provider.
 */
WeatherProvider.initialize = function (providerIdentifier, delegate) {
	providerIdentifier = providerIdentifier.toLowerCase();

	var provider = new WeatherProvider.providers[providerIdentifier]();

	provider.delegate = delegate;
	provider.setConfig(delegate.config);

	provider.providerIdentifier = providerIdentifier;
	if (!provider.providerName) {
		provider.providerName = providerIdentifier;
	}

	return provider;
};

JimHutchins, 2021-11-22:
The files I modified are attached.

I suspect there may be a better (more official) way to do this, but it worked for me.

Sorry for the slow reply. I've been buried.