MagicMirrorOrg/MagicMirror #2127: Issue with MMM-Snowboy and GoogleAssistant

mnooijen, 2020-09-19:
[image]

I hope someone can help me with this problem. I have installed the 2 modules for my smart mirror and that is MMM-Snowboy and Google Assistant. At first start everything worked properly but got a message that I had to update the Google Assistant and I did. After this I restarted my smart mirror. Now my mirror does not come on and I get a green message in my terminal with the following message that the module “snowboy” is not found. Anyone an idea how to fix this? See my screenshot link so you can see the notification I see. My screenshot link: https://paste.pics/A5S87

Regards,

Marcel

bugsounet, 2020-09-19:
Hi and thanks for your participation :) 

@MichMich can't response to this issue ...
So i will respond :) 

MMM-GoogleAssistant it's a great assitant but you can only use ONE mic recorder at time :)
he have an record embed, so MMM-Snowboy is not needed. you can delete it ;)
In terminal try `num run update`
and answer `y` for electron rebuild of snowboy part :)

of course no after actions is needed after : no `npm Fix`, no `npm audit fix` or other ...
Note to @michMich: Because MagicMirror Don't USE in core same version of node !!!!!

by default `Jarvis` is the keyword to call the assistant ;)

@bugsounet vs @MichMich 

Please don't forget to mention @MichMich in your post ... thx :)

mnooijen, 2020-09-19:
In terminal try num run update (Please explain where to do that in terminal as I am getting a -bash error?)
Do you have a good step-by-step plan for me?

MichMich, 2020-09-22:
@bugsounet Is there something I can do to fix this issue?

bugsounet, 2020-09-22:
No... I have just say.. Every of my module is banned of MagicMirror

bugsounet, 2020-09-22:
So, In all case you can close it ;)

MichMich, 2020-09-22:
@mnooijen This isn't a core issue, and @bugsounet has chosen to unpublished his module, so I'm afraid I can't help you with this. Is it ok to close this issue?

tdh8316, 2020-09-25:
Is exiting MagicMirror whose idea? That's the worst idea I have seen.

Please don't do irresponsible behavior.
Please be aware that there are a lot of people around the world who use the MagicMirror and Google Assistant module. And due to your irresponsible behavior, these innocent people are being harmed.

I hope this problem is solved smoothly.

Dear developers, please remember how valuable your work is.
A lot of people enjoy building their smart mirrors and using the Google Assistant module. Don't take away their pleasure.

I hope everything goes well.

MichMich, 2020-09-30:
I'm closing this issue for now. If you feel this issue should remain open, feel free to let me know so I can reopen it. For now there is not much the MagicMirror² core team can do to solve this issue.

closed 2020-09-30 by MichMich