MagicMirrorOrg/MagicMirror #2128: Having problems with sound

SushiRollsss, 2020-09-19:
whenever I activate the google assistant, there is always a sound cut off at the very beginning of the phrase google would speak
for example, when I ask Google what time it is, it would say 9:57 am. But, the only thing anyone can hear is the last 'm'. It is not a delay as a delay would still say the entire sentence. With this, the sound only turns on when the sentence is almost finished. Any help would be appreciated! :)

bugsounet, 2020-09-19:
Hi @SushiRollsss , thanks to your participation !

@MichMich, do you have any idea why Electron cut it ???

in this cas you have 2 solution, maybe you use `useNative:true,` feature
any sound will be played with mpg321 play 

* try with HTML5 supoprt ( useNative:false)
* try to install mpg123
```
sudo apt-get install mpg123
```
in this case `playProgram="mpg123",`

Maybe @MichMich have a better solution, for this bug of MagicMirror ?

Note: Please don't forget to mention @MichMich in your post ... thx :)
One mention = One Mail :)

MichMich, 2020-09-19:
This is an issue with an external MagicMirror module. Please open an issue in the repository of that module.

Thanks!

closed 2020-09-19 by MichMich

bugsounet, 2020-09-19:
always on live ?

bugsounet, 2020-09-19:
ALL ISSUES OF MY MODULES WILL BE OPEN THERE !!!!!

bugsounet, 2020-09-19:
it's the begining ! ... so respond in forum @MichMich !