MagicMirrorOrg/MagicMirror #2129: ANY issue of @bugsounet Modules (22 Modules  will come there !)

bugsounet, 2020-09-19:
So then ... @MichMich ... you don't want to repond to MY question ? maybe it's so difficult for you ?

Maybe I'm a `little develloper` for MagicMirror...

I redirect ALL issue there ! yeah 22 modules !
let's start so...

1) https://github.com/bugsounet/MMM-GoogleAssistant/issues/162
2) https://github.com/bugsounet/MMM-Assistant2Display/issues/128
3) https://github.com/bugsounet/MMM-Freebox/issues/28
4) https://github.com/bugsounet/MMM-NewPIR/issues/36
5) https://github.com/bugsounet/MMM-GABackground/issues/12
6) https://github.com/bugsounet/MMM-TelegramBot/issues/64
7) https://github.com/bugsounet/MMM-Tools/issues/58
8) https://github.com/bugsounet/MMM-Spotify/issues/111
9) https://github.com/bugsounet/MMM-FreeboxTV/issues/12
10) https://github.com/bugsounet/MMM-UpdateNotification/issues/28
11) https://github.com/bugsounet/MMM-Pronote/issues/1
12) https://github.com/bugsounet/MMM-NotificationReceived/issues/2
13) https://github.com/bugsounet/MMM-AssistantMk2/blob/master/README.md
14) https://github.com/bugsounet/MMM-News/issues/14
15) https://github.com/bugsounet/MMM-Hotword/issues/151
16) https://github.com/bugsounet/MMM-Snowboy/issues/12
17) https://github.com/bugsounet/MMM-Timetable/issues/10
18) https://github.com/bugsounet/MMM-Xbox/issues/4
19) https://github.com/bugsounet/MMM-AirParif/issues/2
20) https://github.com/bugsounet/MMM-HomeStatus/issues/6
21) https://github.com/bugsounet/MMM-JarvisFace/issues/1
22) https://github.com/bugsounet/MMM-RemoteShoppingList/issues/1

@MichMich: should I redirect my 14 NPM library designed especially for MagicMirror also here too ???

Or you respond to MY [question](https://forum.magicmirror.builders/post/81698) ?

rejas, 2020-09-19:
Might I ask what happened so that you fehlt the need to open this issue?

bugsounet, 2020-09-19:
@rejas = @MichMich ... no, i don't think so
So... :yawning_face:

rejas, 2020-09-19:
No reason to be rude to me....

bugsounet, 2020-09-19:
@MichMich ... one mention = one mail ....

related to [this](https://github.com/MichMich/MagicMirror/issues/2127)

MichMich, 2020-09-19:
Hey man, if this is the way you communicate within an open source project, I don't feel obligated to your question. 

The forum is ment as a community support platform. It's not there for me to give support. 

Thanks.

closed 2020-09-19 by MichMich

bugsounet, 2020-09-19:
Hey... Mister Michael Teeuw... I done a PR ... you will not read .. I ask some question ... You don't respnd ...

So ?

Who don't want to comunicate ?

Me ???

Ok you search what ? lost 22 modules ? 14 npm library ?

bugsounet, 2020-09-19:
Result : 22 Modules / 14 npm library lost ! Yeah :) AND a little developer !

Bye ;)

fastastic, 2020-09-19:
You don't know what do your repositories mean to the MagicMirror community, so, I encourage you to PLEASE reupload your repositories. At least don't mantain them but please...

masoudl, 2020-09-20:
I am not sure what has happened between Bugsounet and MichMich, but the net result is that the community is suffering. My daughter was so enthusiastic about Magic Mirror and Assistant modules, she has been learning coding over the past 3 months, and working hard to set up her Magic Module, and we just got it all working just to see modules disappear and stop working. 
This is truly heart breaking. I wish you would both reconsider your approach, and in the spirit of an open source community find a better way to address the disagreement.

pnobrega80, 2020-09-20:
Please guys try to fix the issue. It's the community and a lot of projects that get hurt. Please consider that.

bugsounet, 2020-09-20:
Don't worry, it's solved for me ! 

[read there](https://github.com/bugsounet/MagicMirror/blob/MagicMirror/README.md)

@bugsounet 

See you later ! (maybe... if you follow me !)

rejas, 2020-09-20:
@masoudl I am not sure how to resolve this with @bugsounet current attitude. @MichMich could not be any more welcoming. It was bugsounet who choose to delete his code claiming that michmich sees him only as a little developer and didn't want his code. Never have I read something like that from him neither did I personally feel not welcome here. 
Yes, PRs get stale sometime or take time to get merged, but as a father you surely understand that there isn't always time for those things. Especially if it is only a hobby that doesn't pay the bills.