MagicMirrorOrg/MagicMirror #2131: MagicMirror run error due to MMM-Google Assistant error

alectunney, 2020-09-20:
## I found a bug in MagicMirror

Please make sure to only submit reproducible issues. You can safely remove everything above the dividing line.
When submitting a new issue, please supply the following information:

**Platform**:  Raspberry Pi 4 with Magic Mirror 2.12, MMM-GoogleAssistant Version: 2.4.11,    [A2D] MMM-Assistant2Display Version: 2.3.15
(Updated GA Module and MM today)

**Node Version**: Make sure it's version 8 or later.

**MagicMirror Version**: 2.12
**Description**: MM Runs and then stops due to errors:
[2020-09-20 15:37:43.963] [LOG]    [A2D] MMM-Assistant2Display Version: 2.3.15
[2020-09-20 15:37:44.158] [INFO]   Checking git for module: MMM-GoogleAssistant
[2020-09-20 15:37:44.198] [INFO]   Checking git for module: MMM-Assistant2Display
[2020-09-20 15:37:44.266] [INFO]   Checking git for module: MMM-Globe
[2020-09-20 15:37:44.314] [INFO]   Checking git for module: MMM-ModuleScheduler
[2020-09-20 15:37:44.374] [INFO]   Checking git for module: MMM-GoogleBirthdaysProvider
[2020-09-20 15:37:44.422] [INFO]   Checking git for module: MMM-Screencast
[2020-09-20 15:37:44.458] [INFO]   Checking git for module: MMM-Remote-Control
[2020-09-20 15:37:44.503] [INFO]   Checking git for module: MMM-Formula1
[2020-09-20 15:37:44.554] [INFO]   Checking git for module: MMM-GooglePhotos
[2020-09-20 15:37:44.611] [INFO]   Checking git for module: MMM-CurrencyExchange
[2020-09-20 15:37:44.658] [INFO]   Checking git for module: MMM-MoonPhase
[2020-09-20 15:37:46.069] [LOG]    [CAST] Cast v1.0.0 Initialized...
[2020-09-20 15:37:46.178] [LOG]    [SPOTIFY] Launch Librespot...
[2020-09-20 15:37:46.193] [LOG]    [NPM] Initialized v0.9.9
[2020-09-20 15:37:46.631] [LOG]    [A2D] Assistant2Display is initialized.
[2020-09-20 15:37:46.633] [ERROR]  Whoops! There was an uncaught exception...
[2020-09-20 15:37:46.634] [ERROR]  Error: listen EADDRINUSE: address already in use :::8569
    at Server.setupListenHandle [as _listen2] (net.js:1226:14)
    at listenInCluster (net.js:1274:12)
    at Server.listen (net.js:1362:7)
    at Function.listen (/home/pi/MagicMirror/modules/MMM-Assistant2Display/node_modules/express/lib/application.js:618:24)
    at DialServer.start (/home/pi/MagicMirror/modules/MMM-Assistant2Display/node_modules/@bugsounet/cast/index.js:90:23)
    at Class.addons (/home/pi/MagicMirror/modules/MMM-Assistant2Display/node_helper.js:1:5119)
    at Class.initialize (/home/pi/MagicMirror/modules/MMM-Assistant2Display/node_helper.js:1:3606)
    at MD5.cb (/home/pi/MagicMirror/modules/MMM-Assistant2Display/node_helper.js:1:847)
    at MD5.start (/home/pi/MagicMirror/modules/MMM-Assistant2Display/node_modules/@bugsounet/md5/index.js:1:1820) {
  code: 'EADDRINUSE',
  errno: 'EADDRINUSE',
  syscall: 'listen',
  address: '::',
  port: 8569
}
[2020-09-20 15:37:46.645] [ERROR]  MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
[2020-09-20 15:37:46.646] [ERROR]  If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
[2020-09-20 15:37:46.706] [ERROR]  [MD5] MMM-GoogleAssistant sources edited !
[2020-09-20 15:37:46.707] [ERROR]  [MD5] You don't have permission to do this
[2020-09-20 15:37:46.707] [ERROR]  [MD5] Maybe... You don't have permission to USE MMM-GoogleAssistant in MagicMirror! (thx to @MichMich)
[2020-09-20 15:37:46.708] [ERROR]  [MD5] Exiting MagicMirror...

**Steps to Reproduce**: ~/MagicMirror $ npm run start
(Since getting update notification for GA and GA2D

**Expected Results**: MM Run

**Actual Results**: [2020-09-20 15:37:46.645] [ERROR]  MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
[2020-09-20 15:37:46.646] [ERROR]  If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
[2020-09-20 15:37:46.706] [ERROR]  [MD5] MMM-GoogleAssistant sources edited !
[2020-09-20 15:37:46.707] [ERROR]  [MD5] You don't have permission to do this
[2020-09-20 15:37:46.707] [ERROR]  [MD5] Maybe... You don't have permission to USE MMM-GoogleAssistant in MagicMirror! (thx to @MichMich)
[2020-09-20 15:37:46.708] [ERROR]  [MD5] Exiting MagicMirror...

**Configuration**: 

/* Magic Mirror Config Sample
 *
 * By Michael Teeuw https://michaelteeuw.nl
 * MIT Licensed.
[... 382 lines cut ...]
		{
			module: "newsfeed",
			position: "top_bar",
			config: {
				feeds: [
					{
						title: "News24 SA",
						url: "http://feeds.news24.com/articles/News24/TopStories/rss"
					}
				],
				showSourceTitle: true,
				showPublishDate: true,
				broadcastNewsFeeds: true,
				broadcastNewsUpdates: true
			}
		},
	]
};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.p
[image]
[image]

rejas, 2020-09-20:
I see these lines:
"
[2020-09-20 15:37:46.706] [ERROR] [MD5] MMM-GoogleAssistant sources edited !
[2020-09-20 15:37:46.707] [ERROR] [MD5] You don't have permission to do this
[2020-09-20 15:37:46.707] [ERROR] [MD5] Maybe... You don't have permission to USE MMM-GoogleAssistant in MagicMirror! (thx to @MichMich)
"

The MMM-GoogleAssistant seems to complain that you edited the sources. Not sure what the core of MagicMirror can do here, so please redirect your issue to the MMM-GoogleAssistant repository of @bugsounet...

bugsounet, 2020-09-20:
`Maybe... You don't have permission to USE MMM-GoogleAssistant in MagicMirror!` 

I force update GA to not using it in Original MagicMirror and close the MM app (Yes ... the Little developer can do this, sorry)

@rejas: You can close this issue ;)

closed 2020-09-20 by MichMich