MagicMirrorOrg/MagicMirror #2140: Asking for cookies when opening full article out of newsfeed module

Bountyman1234, 2020-09-26:
First of all thanks for this incredible project, I am so exited about 💯 

But I am not that experienced and so I have a problem now with your default newsfeed module.
Every time I want to open the full article (via the notification ARTICLE_MORE_DETAILS) the site pops up like it should but than a pop up pops up and asks me to accept cookies. And I have to do this every time I open the Article site because it does not safe my cookie settings. And as my mirror should not have a touch screen nor a mouse thats not ideal.
Is there any solution?

Regards,

sdetweil, 2020-09-29:
I'm not aware of any mechanism to override the cookie prompt, its a security thing..
yes, its a pain when u are using a browser in a non-typical environment like this..

MichMich, 2020-09-30:
I'm closing this since it's not a MagicMirror² core issue and can not be solved by the MagicMirror² framework. If you feel this issue should remain open, please let me know.

closed 2020-09-30 by MichMich