MagicMirrorOrg/MagicMirror #2147: 2.13 black screen on some systems, security warning

sdetweil, 2020-10-02:
## I'm not sure if this is a bug

use sees in the browser developer window console log

For more information and help, consult
https://electronjs.org/docs/tutorial/security.
 This warning will not show up
once the app is packaged.
/home/pi/MagicMirror…ity-warnings.js:145 Electron Security Warning (Insecure Content-Security-Policy) This renderer process has either no Content Security
    Policy set or a policy with "unsafe-eval" enabled. This exposes users of
    this app to unnecessary security risks.

fix for this is add the following meta tag to index.html 
```
	<meta name="apple-mobile-web-app-capable" content="yes">
	<meta name="apple-mobile-web-app-status-bar-style" content="black">
	<meta name="format-detection" content="telephone=no">
	<meta name="mobile-web-app-capable" content="yes">
	<meta http-equiv="Content-Security-Policy" content="script-src 'self';">    <!--- add this line -- >
```

BerndKohl, 2020-10-03:
Had the same problem after upgrading MM to the latest version.
Adding the line from above solved the issue.

alexandruchitic, 2020-10-03:
Unfortunately I still get a black screen (even after using this solution). Anyone knows how I can fix it?

sdetweil, 2020-10-03:
@alexandruchitic can u post the message from npm start and the developers window?

alexandruchitic, 2020-10-03:
Sure, here's the npm start messaging:

`> magicmirror@2.13.0 start /home/pi/MagicMirror
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

[2020-10-03 21:26:26.494] [LOG]    Starting MagicMirror: v2.13.0
[2020-10-03 21:26:26.502] [LOG]    Loading config ...
[2020-10-03 21:26:26.507] [LOG]    Loading module helpers ...
[2020-10-03 21:26:26.509] [LOG]    No helper found for module: alert.
[2020-10-03 21:26:26.551] [LOG]    Initializing new module helper ...
[2020-10-03 21:26:26.551] [LOG]    Module helper loaded: updatenotification
[2020-10-03 21:26:26.552] [LOG]    No helper found for module: clock.
[2020-10-03 21:26:27.282] [LOG]    Initializing new module helper ...
[2020-10-03 21:26:27.283] [LOG]    Module helper loaded: MMM-TelegramBot
[2020-10-03 21:26:27.538] [LOG]    Initializing new module helper ...
[2020-10-03 21:26:27.539] [LOG]    Module helper loaded: MMM-3Day-Forecast
[2020-10-03 21:26:27.541] [LOG]    Initializing new module helper ...
[2020-10-03 21:26:27.542] [LOG]    Module helper loaded: MMM-rain-forecast
[2020-10-03 21:26:27.554] [LOG]    Initializing new module helper ...
[2020-10-03 21:26:27.555] [LOG]    Module helper loaded: MMM-SystemStats
[2020-10-03 21:26:27.557] [LOG]    Initializing new module helper ...
[2020-10-03 21:26:27.558] [LOG]    Module helper loaded: MMM-Widget
[2020-10-03 21:26:27.581] [LOG]    Initializing new module helper ...
[2020-10-03 21:26:27.582] [LOG]    Module helper loaded: newsfeed
[2020-10-03 21:26:27.583] [LOG]    No helper found for module: MMM-windy.
[2020-10-03 21:26:27.584] [LOG]    All module helpers loaded.
[2020-10-03 21:26:27.688] [LOG]    Starting server on port 8080 ...
[2020-10-03 21:26:27.701] [LOG]    Server started ...
[2020-10-03 21:26:27.702] [LOG]    Connecting socket for: updatenotification
[2020-10-03 21:26:27.703] [LOG]    Connecting socket for: MMM-TelegramBot
[2020-10-03 21:26:27.704] [LOG]    Connecting socket for: MMM-3Day-Forecast
[2020-10-03 21:26:27.705] [LOG]    MMM-3Day-Forecast helper, started...
[2020-10-03 21:26:27.706] [LOG]    Connecting socket for: MMM-rain-forecast
[2020-10-03 21:26:27.706] [LOG]    Starting node helper for: MMM-rain-forecast
[2020-10-03 21:26:27.707] [LOG]    Connecting socket for: MMM-SystemStats
[2020-10-03 21:26:27.707] [LOG]    Connecting socket for: MMM-Widget
[2020-10-03 21:26:27.708] [LOG]    Connecting socket for: newsfeed
[2020-10-03 21:26:27.709] [LOG]    Starting node helper for: newsfeed
[2020-10-03 21:26:27.709] [LOG]    Sockets connected & modules started ...
[2020-10-03 21:26:27.788] [LOG]    Launching application.
`
Not sure what dev window you're making a reference to?

sdetweil, 2020-10-03:
developers window, on mirror, ctrl-shift-i on the keyboard, then select the console tab and scroll up to see any errors

alexandruchitic, 2020-10-03:
[image]
[image]
[image]
[image]

Looks like I got two errors & one alert. 

Pasted the snapshot on all of them, including what's on line 22 in the js file

sdetweil, 2020-10-03:
yeh, looks like the translation de.json file in mmm-3day-forecsst has a bug
no comments allowed

sdetweil, 2020-10-03:
put the contents on one of the online json validators

alexandruchitic, 2020-10-03:
Sure, will do
 Thanks for the help!

PieBa, 2020-10-04:
Same issue for me today after mm update.
Adding the line to index.html fixed it.

sdetweil, 2020-10-04:
seems there should be some other messages..    can u post the text of the window where u do npm start?

sdetweil, 2020-10-04:
@alexandruchitic looks like the MMM-3day-Forecast module was updated 3 days ago to fix this problem 
do a git pull in the module folder to get the latest

alexandruchitic, 2020-10-04:
@sdetweil yup, the last update on the MMM-3day-Forecast did the trick. However I had to take out the line you added in this thread to make the black screen disappear.

sdetweil, 2020-10-04:
@alexandruchitic I didn't think that line applied anyhow...

alexandruchitic, 2020-10-04:
> 
> 
> @alexandruchitic I didn't think that line applied anyhow...

Sure, thanks for the help anyway! My issue was solved

PieBa, 2020-10-04:
> seems there should be some other messages.. can u post the text of the window where u do npm start?
> and the developers window,
> 
> on MM keyboard, ctrl-shift-i, select the tab labeled 'console' and scroll up to see any errors..

This is the output after npm start. It stops after "Launching application". No further output there.
Windows of the MM is completely black.

pi@MagicMirror:~/MagicMirror $ npm start

> magicmirror@2.13.0 start /home/pi/MagicMirror
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

```
[2020-10-04 18:16:19.762] [LOG]    Starting MagicMirror: v2.13.0
[2020-10-04 18:16:19.779] [LOG]    Loading config ...
[2020-10-04 18:16:19.795] [LOG]    Loading module helpers ...
[2020-10-04 18:16:19.799] [LOG]    No helper found for module: alert.
[2020-10-04 18:16:19.954] [LOG]    Initializing new module helper ...
[2020-10-04 18:16:19.958] [LOG]    Module helper loaded: updatenotification
[2020-10-04 18:16:19.960] [LOG]    No helper found for module: clock.
[2020-10-04 18:16:19.961] [LOG]    No helper found for module: currentweather.
[2020-10-04 18:16:20.540] [LOG]    Initializing new module helper ...
[2020-10-04 18:16:20.542] [LOG]    Module helper loaded: MMM-PublicTransportHafas
[2020-10-04 18:16:21.300] [LOG]    Initializing new module helper ...
[2020-10-04 18:16:21.301] [LOG]    Module helper loaded: MMM-Travelynx
[2020-10-04 18:16:21.311] [LOG]    Initializing new module helper ...
[2020-10-04 18:16:21.314] [LOG]    Module helper loaded: MMM-COVID-19-RKI
[2020-10-04 18:16:21.499] [LOG]    Initializing new module helper ...
[2020-10-04 18:16:21.500] [LOG]    Module helper loaded: calendar
[2020-10-04 18:16:22.327] [LOG]    Initializing new module helper ...
[2020-10-04 18:16:22.328] [LOG]    Module helper loaded: MMM-FHEM
[2020-10-04 18:16:22.330] [LOG]    No helper found for module: weatherforecast.
[2020-10-04 18:16:23.649] [LOG]    Initializing new module helper ...
[2020-10-04 18:16:23.652] [LOG]    Module helper loaded: MMM-WeConnect
[2020-10-04 18:16:23.662] [LOG]    Initializing new module helper ...
[2020-10-04 18:16:23.663] [LOG]    Module helper loaded: MMM-go-eCharger
[2020-10-04 18:16:24.034] [LOG]    Initializing new module helper ...
[2020-10-04 18:16:24.036] [LOG]    Module helper loaded: MMM-FRITZ-Box-Callmonitor
[2020-10-04 18:16:24.039] [LOG]    No helper found for module: MMM-Callmonitor-Current-Call.
[2020-10-04 18:16:24.046] [LOG]    Initializing new module helper ...
[2020-10-04 18:16:24.050] [LOG]    Module helper loaded: MMM-Lunartic
[2020-10-04 18:16:24.834] [LOG]    Initializing new module helper ...
[2020-10-04 18:16:24.835] [LOG]    Module helper loaded: MMM-EARTH
[2020-10-04 18:16:24.837] [LOG]    No helper found for module: compliments.
[2020-10-04 18:16:24.905] [LOG]    Initializing new module helper ...
[2020-10-04 18:16:24.909] [LOG]    Module helper loaded: newsfeed
[2020-10-04 18:16:24.911] [LOG]    All module helpers loaded.
[2020-10-04 18:16:25.233] [LOG]    Starting server on port 8080 ...
[2020-10-04 18:16:25.262] [LOG]    Server started ...
[2020-10-04 18:16:25.264] [LOG]    Connecting socket for: updatenotification
[2020-10-04 18:16:25.266] [LOG]    Connecting socket for: MMM-PublicTransportHafas
[2020-10-04 18:16:25.268] [LOG]    Connecting socket for: MMM-Travelynx
[2020-10-04 18:16:25.269] [LOG]    MMM-Travelynx helper started...
[2020-10-04 18:16:25.271] [LOG]    Connecting socket for: MMM-COVID-19-RKI
[2020-10-04 18:16:25.273] [LOG]    Starting node_helper for: MMM-COVID-19-RKI
[2020-10-04 18:16:25.276] [LOG]    Connecting socket for: calendar
[2020-10-04 18:16:25.278] [LOG]    Starting node helper for: calendar
[2020-10-04 18:16:25.281] [LOG]    Connecting socket for: MMM-FHEM
[2020-10-04 18:16:25.283] [LOG]    Connecting socket for: MMM-WeConnect
[2020-10-04 18:16:25.285] [LOG]    MMM-WeConnect: Starting node helper
[2020-10-04 18:16:25.288] [LOG]    Connecting socket for: MMM-go-eCharger
[2020-10-04 18:16:25.290] [LOG]    Starting node_helper for: MMM-go-eCharger
[2020-10-04 18:16:25.291] [LOG]    Connecting socket for: MMM-FRITZ-Box-Callmonitor
[2020-10-04 18:16:25.293] [LOG]    Starting module: MMM-FRITZ-Box-Callmonitor
[2020-10-04 18:16:25.294] [LOG]    Connecting socket for: MMM-Lunartic
[2020-10-04 18:16:25.295] [LOG]    Starting node_helper for: MMM-Lunartic
[2020-10-04 18:16:25.296] [LOG]    Connecting socket for: MMM-EARTH
[2020-10-04 18:16:25.297] [LOG]    Starting module: MMM-EARTH
[2020-10-04 18:16:25.298] [LOG]    Connecting socket for: newsfeed
[... 34 lines cut ...]
  
 
For more information and help, consult
https://electronjs.org/docs/tutorial/security.
 This warning will not show up
once the app is packaged.
warnAboutInsecureResources @ /home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/renderer/security-warnings.js:95
/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/renderer/security-warnings.js:145 Electron Security Warning (Insecure Content-Security-Policy) This renderer process has either no Content Security
    Policy set or a policy with "unsafe-eval" enabled. This exposes users of
    this app to unnecessary security risks.
 
For more information and help, consult
https://electronjs.org/docs/tutorial/security.
 This warning will not show up
once the app is packaged.

(anonymous) @ /home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/renderer/security-warnings.js:145
jsonpointer.js:91 Uncaught ReferenceError: exports is not defined
    at jsonpointer.js:91

(index):5 Uncaught SyntaxError: Unexpected token / in JSON at position 79
    at JSON.parse (<anonymous>)
    at XMLHttpRequest.xhr.onreadystatechange (translator.js:22)`

I hope this is what you were asking for?

sdetweil, 2020-10-04:
@PieBa  yep, there it is
```
(index):5 Uncaught SyntaxError: Unexpected token / in JSON at position 79
    at JSON.parse (<anonymous>)
    at XMLHttpRequest.xhr.onreadystatechange (translator.js:22)
```

sdetweil, 2020-10-04:
I think this is a dup of 2149, will wait a little.. PR submitted to stop the crash at least

sdetweil, 2020-10-05:
closing as dup

closed 2020-10-05 by sdetweil