MagicMirrorOrg/MagicMirror #2159: Google calendar have some problems

hangorazvan, 2020-10-11:
From today something happend on Google calendar. Recurring anual events are not shown.
Back in 2.12 version calendarfetcher.js is working fine with ical even with node-ical
Attached is romanian national holidays calendar with recurring events
[basic.zip](https://github.com/MichMich/MagicMirror/files/5361224/basic.zip)

sdetweil, 2020-10-11:
that code is unchanged since 2.12..

I loaded your calendar in  2.13 develop branch and see this 
[image]

to switch to the develop branch do 
```
cd ~/MagicMirror
git checkout develop
```
if you used my install script do 
```
cd ~/MagicMirror
git fetch origin develop develop
git checkout develop
```

then restart mm

there are NO recurring events in the ics u provided.. (they would need an RRULE line) 
the ics is just a text file

hangorazvan, 2020-10-11:
I already use develop branch.
this is normal calendars including the attached one that now it works without above code
[image]

and this is with the code (until this morning is working as well) and national holidays not shown (in blue) or birhdays
[image]

sdetweil, 2020-10-11:
I don't have a link to the national holiday calendar

the basic.ics does not have the word 'Fear' anywhere in it.

I am using the latest develop branch 

i have 2.11 here and it has the events.slice() clause
the default number of events is 10 
maximumEntries: 10, not changed since before 2.11

sdetweil, 2020-10-11:
none of the dates in the romanian national holidays has any of those words shown
and does not contain any repeating events

https://www.officeholidays.com/countries/romania/2020

hangorazvan, 2020-10-11:
I have to explain to you like a novice, although you don't have to explain to me what's going on, something is wrong and not working. I made an issue, but I can very well use the code that works.
If you can deduce from the different colors and icons it is about several calendars, the one with blue is with the national holidays, the one with yellow the birthdays, the one with magenta and the movie icon is for movies, and so on. It is obvious that in the one with national holidays and birthdays the events are recursive. And they don't show up. In the attached basic.ics the flag for recurrence no longer appears, that's why I tell you that maybe google has changed something.
this is the link for national holidays calendar:
https://calendar.google.com/calendar/ical/ro.romanian%23holiday%40group.v.calendar.google.com/public/basic.ics
and google url:
https://calendar.google.com/calendar/embed?src=ro.romanian%23holiday%40group.v.calendar.google.com&ctz=Europe%2FBucharest

sdetweil, 2020-10-11:
ok, I am missing something..  following the link to the basic.ics I get a calendar with no repeating rules (RRULE)
but I DO get this display with the default calendar settings
[image]

without fade 
[image]

we cannot control if the calendar provider changes their data.. I had to get calendarlabs to remove a bad RRULE:RRULE line last year.. 

one of your pictures shows the 1st cal entry with the word Fear, I don't know what cal that came from, but I do not have it. 
it is under your comment 
"The national holidays is public Google calendar, thats why I think there is maybe some error in their calendar"

hangorazvan, 2020-10-11:
The one with Fear is another calendar with movies made by me in google calendar, and the events are not recurring
this is my config.js calendar settings
```
calendars: [
	{
		symbol: "calendar-check-o", symbolClass: "skyblue", titleClass: "skyblue", timeClass: "skyblue",
		url: "https://calendar.google.com/calendar/ical/ro.romanian%23holiday%40group.v.calendar.google.com/public/basic.ics"
	},
	{
		symbol: "birthday-cake", symbolClass: "gold", titleClass: "gold", timeClass: "gold",
		url: "https://calendar.google.com/calendar/ical/xxxxxxxxxxxxxxxxxxxxxx%40group.calendar.google.com/private-xxxxxxxxxxxxxxxxxxxxxxxxxx/basic.ics"
	},
	{
		symbol: "film", symbolClass: "magenta", titleClass: "magenta", timeClass: "magenta",
		url: "https://calendar.google.com/calendar/ical/emm3k4f4t7dihvfb0c65st0ijo%40group.calendar.google.com/public/basic.ics"
	}
]
```

sdetweil, 2020-10-11:
looks the same to me as the  2,13 develop using that cal def cut and paste
[image]

hangorazvan, 2020-10-12:
You must load these calendars from Google, not from local folder. Try any public calendar or make one with yearly recurring events.

hangorazvan, 2020-10-12:
OK, don't bother anymore, I will make changes that works for me.
Thank you anyway!

sdetweil, 2020-10-12:
I am sorry. I don't understand. I am using your calendar definition, cut and paste.

on the slice()
you need to change the config maxEntries value. no code change required.

on the other, full day,  I don't see a difference.
full day events are those that have no time value associated in their DTSTART/DTEND records

hangorazvan, 2020-10-12:
I have maximumEntries: 14. But ignore the fact that is full day or recurring event.
The wierd thing is that without event.slice I can see all calendars and with event.slice I can see only 2 of 4 calendars, one is national holidays, the blue one.

sdetweil, 2020-10-12:
yes, the final sort of the events doesn't occur until the (maxEntries) 'list' is returned to the top side display module.  
thanks, I had opened an issue on that over a year ago, but guess it wasn't resolved. 

looks like the code has the list truncation still on in the fetcher which it shouldn't.   good catch...

sdetweil, 2020-10-12:
well, i think I was wrong.. 

each calendar is restricted to maxEntries, either the global value configured above the calendar array, or specifically IN a calendar entry 
```
 config {  
     maxEntries:14
     calendars : [
           {
               maxEntries:2
               url: ...... some url 
           }
     ]
}
```

this says the MAX to display is 14
but the max from this calendar is 2 (regardless of how many it has) 

if maxEntries is NOT specified on the the individual calendar entry, then the global maxEntries will used (14 here)

can u share the actual config with me privately, I won't share it anywhere.. 
same id at gmail

hangorazvan, 2020-10-12:
sent, thank you!

hangorazvan, 2020-10-12:
yes, specifying for each calendar maximumEntries: is working with event.slice

sdetweil, 2020-10-12:
ok.. . the 'problem' was in your config,
```
			excludedEvents: [],
			broadcastPastEvents: true,    / < ----- 
			nextDaysRelative: true,
```

so the old cal events consumed all the space for 10 entries max, but the front doesn't display them. 
(but some other module might care?)

here with global maximumEntries : 15  on 2.13 (no maximumEtries on any cal)
[image]

hangorazvan, 2020-10-12:
You are right

closed 2020-10-12 by hangorazvan