MagicMirrorOrg/MagicMirror #2164: Weather modules dont show up anymore

rejas, 2020-10-13:
After https://github.com/MichMich/MagicMirror/pull/2161 got merged into develop my weather modules (current & forecast) dont show up anymore.

Not sure what happens but IMHO that PR seems to be incomplete and breaks existing configs and should therefor should be reworked.

Snille, 2020-10-13:
Hmm... Sorry, don't know why its not working for you, all is good here, my weather modules keeps showing them self even if I have hidden them with another module. Do you not experience this?

rejas, 2020-10-13:
I dont hide my weather modules from other modules. How do you do that?

sdetweil, 2020-10-13:
@rejas 
MMM-Voice will allow you to hide individual modules on command
my MMM-SleepWake provides an option to hide all modules instead of turning off the screen (where not possible)

there are others

Snille, 2020-10-13:
@rejas I'm using my own MMM-Modulebar to hide and show modules. And also I'm using "profiles" MMM-ProfileSwitcher to show different modules for different users (@ different times of the day). But the problem is that both the weather modules do not "respect" that they have been hidden. They keep showing up when they update (default every 10 minutes).

sdetweil, 2020-10-13:
their getDom() routines should become aware of the state and not draw when hidden
they should have suspend() and resume() methods that track the state

MichMich, 2020-10-13:
This might be an issue in the core itself. getDom should not be called when they are hidden.

sdetweil, 2020-10-13:
@MichMich yep. updateDom() does not check the visibility state before calling getDom()

MichMich, 2020-10-14:
@Snille @rejas Does https://github.com/MichMich/MagicMirror/pull/2166 solve the issue?

rejas, 2020-10-14:
No, I only see a big "undefined" in the place of the module. See my comment in the PR for possible reason.

MichMich, 2020-10-14:
I added a new commit returning an empty promise. Could you give it another try, @rejas?

rejas, 2020-10-14:
The undefined was an error on my side. Still, returning an empty promise is the better way I think. So: thumbs up from me for this.

sdetweil, 2020-10-14:
looks good..  i  think the real bug was in updateDomWithContent, 
but this approach is cleaner

```
var updateDomWithContent = function (module, speed, newHeader, newContent) {
		return new Promise(function (resolve) {
			if (module.hidden || !speed) {                       // <--- maybe ig(!module.hidden && !speed)// speed handled later
```

Snille, 2020-10-14:
Hmm.. It's not working for me, I think, if it's in the branch "develop-hidden-module-update-restriction", the weather modules still comes back on refresh and also the "clock" stops updating if it's been hidden and then shown again... :)

sdetweil, 2020-10-14:
@Snille 
>the weather modules still comes back on refresh
what does that mean?

Snille, 2020-10-14:
@sdetweil Oh, sorry for being unclear. :) I mean when the module "refreshes" it's content (default every 10 minutes)...

sdetweil, 2020-10-14:
can u examine the file ~/MagicMirror/js/main.js 

search for updateDom
should start like this
```
var updateDom = function (module, speed) {
		if (module.hidden) {
			return new Promise();
		}
```

Snille, 2020-10-14:
@sdetweil Yepp, that is correct, I'm on "develop-hidden-module-update-restriction", there is is as you show above. When I hide and show the clock, it stops updating (seconds stops counting) and the weather modules shows up again (I have set the update interval to 1 minute) when they refresh. I'm using my "MMM-Modulebar" to hide and show modules... Is there something I can have missed?

sdetweil, 2020-10-14:
what weather provider are you using?

MichMich, 2020-10-14:
Is that your module? Or someone else's?

MichMich, 2020-10-14:
How does your module show/hide modules? Do you use the MagicMirror² built in functionality for this?

Snille, 2020-10-14:
> Is that your module? Or someone else's?

To hide and show modules I'm using my own module MMM-Modulebar.

Snille, 2020-10-14:
> How does your module show/hide modules? Do you use the MagicMirror² built in functionality for this?

That's what I think I do... :) (https://github.com/Snille/MMM-Modulebar/blob/master/MMM-Modulebar.js)

Snille, 2020-10-14:
> what weather provider are you using?

openweathermap.org

MichMich, 2020-10-14:
Not sure what triggers the showing of the module. But to prevent this, you might want to look into the lockString option during hiding/showing. This prevents modules from showing when your module hides it.

sdetweil, 2020-10-14:
can u open the developers console. ctrl-shift-i, and select console, add weather to the filter field and then dock the console on the bottom of the screen (you can do this on your pc if u have address:"0.0.0.0", and ipWhiltelist:[], in config.js to allow your pc to connect to pi 

the code in weather.js does this 
```
	// What to do when the weather provider has new information available?
	updateAvailable: function () {
		Log.log("New weather information available.");
		this.updateDom(0);
		this.scheduleUpdate();
	},
```
when the providers signal updateAvailable()
so, you should see a message on the console before every update.
you can also add this to the bottom of weather,js

```
	},  // < notice the comma 
	resume() {                         // so it looks like thiss
		Log.log('weather resume called')
	}            <  last lien of add
});
```

Snille, 2020-10-14:
> Not sure what triggers the showing of the module. But to prevent this, you might want to look into the lockString option during hiding/showing. This prevents modules from showing when your module hides it.

Well, I don't want to lock it, because then other modules (like IFTTT och even MMM-Remote-Control) cant un-hide them. Also I'm using "profiles" that switches at different times, there also modules are shown and hidden... :/

sdetweil, 2020-10-14:
> Also I'm using "profiles" that switches at **different times**

Snille, 2020-10-14:
@sdetweil Ok, done it all... :) Do you want the log here, like this..
`

Load script: modules/default/currentweather//currentweather.js
module.js:513 Module registered: currentweather
loader.js:148 Bootstrapping module: currentweather
loader.js:153 Scripts loaded for: currentweather
loader.js:195 Load stylesheet: vendor/node_modules/weathericons/css/weather-icons.css
loader.js:195 Load stylesheet: modules/default/currentweather/currentweather.css
loader.js:155 Styles loaded for: currentweather
loader.js:157 Translations loaded for: currentweather
loader.js:176 Load script: modules/default/weatherforecast//weatherforecast.js
module.js:513 Module registered: weatherforecast
loader.js:148 Bootstrapping module: weatherforecast
loader.js:153 Scripts loaded for: weatherforecast
loader.js:236 File already loaded: weather-icons.css
loader.js:195 Load stylesheet: modules/default/weatherforecast/weatherforecast.css
loader.js:155 Styles loaded for: weatherforecast
loader.js:157 Translations loaded for: weatherforecast
currentweather.js:97 Starting module: currentweather
weatherforecast.js:90 Starting module: weatherforecast
MMM-Modulebar.js:117 Hiding currentweather ID: 4
module.js:199 currentweather is suspended.
MMM-Modulebar.js:117 Hiding weatherforecast ID: 5
module.js:199 weatherforecast is suspended.
module.js:206 currentweather is resumed.
module.js:206 weatherforecast is resumed.
module.js:206 currentweather is resumed.
module.js:206 weatherforecast is resumed.

`

Snille, 2020-10-14:
It's not the weather provider that cases the resume... As far as I can understand?...

sdetweil, 2020-10-14:
so, I would look at profiles, and add a logging entry around its module.show()
sadly show/hide can't tell us who is invoking those calls. 

no, all good how u are running.. nuc is 'pi'.. i do all my dev on ubuntu desktop, and have 5 SBCs and VMs and local machine all running instances..

Snille, 2020-10-14:
@sdetweil Do you have this problem at all on your setup? I have had it from start (over 2 years)... That's why I started to "dig" in to it...

sdetweil, 2020-10-14:
i do not.. my webcam based motion detector (MMM-SleepWake)  hides modules (as my TV doesn't support turn off/on)  and I never see anything pop up 
used on pi, odroid, jetson nano. linux, catalina,

Snille, 2020-10-14:
> so, I would look at profiles, and add a logging entry around its module.show()
> sadly show/hide can't tell us who is invoking those calls.
> 
> no, all good how u are running.. nuc is 'pi'.. i do all my dev on ubuntu desktop, and have 5 SBCs and VMs and local machine all running instances..

Hmm... Do you think it can have to do with the "classes" (that wrongly I call profiles)?

sdetweil, 2020-10-14:
> Do you think it can have to do with the "classes" (that wrongly I call profiles)?

i don't think there are very many places that hide/show will be called.. if u have access to the class name at the time..  u can log it..

sdetweil, 2020-10-14:
hm.. but I see an error with the new code 
```
main.js:115 Uncaught TypeError: Promise resolver undefined is not a function
    at new Promise (<anonymous>)
    at updateDom (main.js:115)
    at Object.updateDom (main.js:560)
    at Class.updateDom (module.js:364)
    at Class.socketNotificationReceived (calendar.js:153)
    at module.js:250
    at r.<anonymous> (socketclient.js:39)
    at r.emit (index.js:83)
    at r.onevent (index.js:83)
    at r.MMSocket.self.socket.onevent (socketclient.js:33)
```
as the promise doesn't have a resolve method

Snille, 2020-10-14:
> > Do you think it can have to do with the "classes" (that wrongly I call profiles)?
> 
> i don't think there are very many places that hide/show will be called.. if u have access to the class name at the time.. u can log it..

I just removed all the classes from all modules... Same thing... Well, I got to get some sleep... :)
My config now looks like this, and it still have the same problems.. :)  
````

/* Magic Mirror Config Sample
 *
 * By Michael Teeuw http://michaelteeuw.nl
 * MIT Licensed.
 *
 * For more information how you can configurate this file
 * See https://github.com/MichMich/MagicMirror#configuration
 *
 */

var config = {
	port: 8080,
	ipWhitelist: [],
	address: "0.0.0.0",
	language: "sv",
//	language: "en",
	timeFormat: 24,
	units: "metric",

	modules: [
		// 00
		{
			module: "alert",
		},
		// 01
		{
			module: "updatenotification",
			position: "top_bar"
		},
		// 02
		{
			module: "clock",
			position: "top_left"
		},

// Test here!

		// 03
		{
			disabled: false,
			module: "MMM-Modulebar",
			position: "bottom_bar",
			config: {
				showBorder: false,
				allowForce: true,
				buttons: {
				"1": {
						module: "clock",
						symbol: "clock"
					},
				"2": {
						module: "currentweather",
						symbol: "cloud"
					},
				"3": {
						module: "weatherforecast",
						symbol: "moon"
					}
				}
			}
		},
		// 04
		{
			disabled: false,
			module: "currentweather",
			position: "top_right",
			header: "Just nu Åstorp",
			config: {
				updateInterval: 60*1000,
				location: "Astorp",
				appendLocationNameToHeader: false,
				locationID: "2724435",
				appid: "--SeCreT--",
				showPeriodUpper: true,
				showHumidity: true
			}
		},
		// 05
		{
			disabled: false,
			module: "weatherforecast",
			position: "top_right",
			header: "Väderprognos för Åstorp",
			config: {
				updateInterval: 60*1000,
				location: "Astorp",
				appendLocationNameToHeader: false,
				locationID: "2724435",
				appid: "--SeCreT--",
				showRainAmount: true
			}
		},

// End Here!

		// 06
		{
			disabled: false,
			module: "MMM-Remote-Control",
			position: "top_left",
			config: {
			}
		}
	]

};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}

````

Snille, 2020-10-14:
> hm.. but I see an error with the new code
>
> ```
> main.js:115 Uncaught TypeError: Promise resolver undefined is not a function
>     at new Promise (<anonymous>)
>     at updateDom (main.js:115)
>     at [...]

I don't get that error... ?

sdetweil, 2020-10-14:
ok, looks like I was always running forecast, not current.. I see the show() problem  with my MMM-SleepWake. 

doah, its right there in currentWeather.

right at the end of updateWeather
```
		this.show(this.config.animationSpeed, { lockString: this.identifier });  < ---- shows itself on every update
		this.loaded = true;
		this.updateDom(this.config.animationSpeed);
		this.sendNotification("CURRENTWEATHER_DATA", { data: data });
	},
```
change to this 
```
		if(!this.loaded){   // if data not yet loaded once
                        // unhide
			this.show(this.config.animationSpeed, { lockString: this.identifier });
                        // indicate data loaded
			this.loaded = true;
		}
```

sdetweil, 2020-10-15:
well, forecast does the same.

MichMich, 2020-10-15:
@rejas & @Snille: does https://github.com/MichMich/MagicMirror/pull/2168 solve the issue for you?

Snille, 2020-10-15:
In what branch has it been merged? :)

MichMich, 2020-10-15:
```
git fetch origin pull/2168/head:develop-hidden-module-update-restriction
git checkout develop-hidden-module-update-restriction
```

sdetweil, 2020-10-15:
i think he already has that branch  @Snille  just change the branch name, add a 1 or something

git fetch origin pull/2168/head:develop-hidden-module-update-restriction1
git checkout develop-hidden-module-update-restriction1

rejas, 2020-10-15:
:+1: looks good on my mirror!

Snille, 2020-10-15:
@MichMich @sdetweil Looks good! Thank you! It's working. :)

MichMich, 2020-10-15:
Merged!

closed 2020-12-08 by MichMich

nishmeht7, 2021-02-13:
@MichMich @sdetweil I'm running into the same issue, although the proposed soln. doesn't work for me. 

Also why was the change reverted in [2165](https://github.com/MichMich/MagicMirror/pull/2165)?

MichMich, 2021-02-13:
Because it caused an issue. Please supply more info about your issue so we can see if this is the same issue.