MagicMirrorOrg/MagicMirror #2170: trying to synch back to my fork for next update , get build error

sdetweil, 2020-10-16:
The DANGER_GITHUB_API_TOKEN/DANGER_BITBUCKETSERVER_HOST environmental variable is missing
Without an api token, danger will be unable to comment on a PR
Error:  Error: Cannot use authenticated API requests.

MichMich, 2020-10-16:
Danger is used for the automatic message to don't forget the Changelog. You don't need it locally nor do you need it for a fork. Is it blocking your build?

sdetweil, 2020-10-16:
yes, I am just trying to pull the changes from develop back to my fork so I can update my local and 
maybe I am doing it wrong

yours ->pull to fork, pull to my local
update
commit
push to my fork
pull request back to you

MichMich, 2020-10-16:
You should be able to merge (and just ignore the error). Not sure why Travis is running on your fork.

sdetweil, 2020-10-16:
was able to  merge, closing

closed 2020-10-16 by sdetweil