MagicMirrorOrg/MagicMirror #2172: Clock moon illuminatedFractionString

hangorazvan, 2020-10-17:
when page refresh and Math.round(moonIllumination.fraction * 100) = 0.526781678199768 or something between 0 and 1
```
clock.js:208 Uncaught (in promise) ReferenceError: illuminatzedFractionString is not defined
    at Class.getDom (clock.js:208)
    at main.js:118
    at new Promise (<anonymous>)
    at updateDom (main.js:117)
    at Object.updateDom (main.js:558)
    at Class.updateDom (module.js:366)
    at notificationTimer (clock.js:49)
```
illuminatedFractionString = Math.floor(moonIllumination.fraction * 100) resolve this issue

see fix floating point bug in moon illumination calc #1939

hangorazvan, 2020-10-17:
nevermind, my error, sorry!

hangorazvan, 2020-10-17:
nope, is not my error is a big bug,!
BAD CODING!

rejas, 2020-10-17:
When you resubmit your PR could you add a test case to it so we can be sure it stays fixed?

hangorazvan, 2020-10-17:
when page refresh and Math.round(moonIllumination.fraction * 100) = 0.526781678199768 or something between 0 and 1

clock.js:208 Uncaught (in promise) ReferenceError: illuminatzedFractionString is not defined
    at Class.getDom (clock.js:208)
    at main.js:118
    at new Promise (<anonymous>)
    at updateDom (main.js:117)
    at Object.updateDom (main.js:558)
    at Class.updateDom (module.js:366)
    at notificationTimer (clock.js:49)
illuminatedFractionString = Math.floor(moonIllumination.fraction * 100) resolve this issue

now  Math.round or Math.floor not working ( I think the moonIllumination.fraction is about 1%)

closed 2020-10-17 by hangorazvan

rejas, 2020-10-17:
so, it isnt an issue at all? or what happened? anythign we can do to help?

hangorazvan, 2020-10-17:
is wierd but I need more time to test, I will try to simulate the new moon status to see behavior of illuminatedFractionString
thank you anyway