MagicMirrorOrg/MagicMirror #2173: Default compliments module; your -remoteFile- option does not work with Github Gist Raw link!

will4378, 2020-10-20:
Hello.

For days I am trying to use your -remoteFile- option in combination with a Github compliments.json gist.
Making a secret gist or a public gist and inserting the RAW link in config.js simply does not work.

In your discription you wrote  to use the RAW link as a remoteFile link.

Using Developer Tools (Console) it responds with a Unexpected Token. Meanwhile the default compliments are loaded.

I really appreciate your view on this issue.

Ron

sdetweil, 2020-10-20:
I setup this config 
```
		{
			module: "compliments",
			disabled: false,
			position: "lower_third",
			config: {
				remoteFile:"https://raw.githubusercontent.com/sdetweil/testcompliments/main/mycompliments.json"
			}
		},
```
and it worked as expected

will4378, 2020-10-21:
Hi Sam.

Thank you so much for your reply. I really appreciate your help, Sam.
I am pretty sure your config.js will work. ;-) Unfortunately there is one difference.

The Developer (I guess  MichMich) explained in -configuration options of a Compliment File hosted on Github-, the following:

See screen shot:

[image]

MichMich explained to use a gist (secret or public) with Tokens. (See MagicMirror webpage: https://docs.magicmirror.builders/modules/compliments.html#configuration-options)

Besides the [user] en [Token] values, the link to my Github gist is exactly the same.
Like this: https://gist.githubusercontent.com/[my username]/[my Token]/raw/[My Token]/compliments.json'

I copied the "RAW" link (Tokens included) and paste it in a web browser. I noticed compliments.json is readable.
Inserting that same link in config.js  apparently doe not work. Developer Tools shows an error: "Unexpected Token"
The text MagicMirror shows is the default compliments.json text.

Of course you may correct me if I am wrong:
To my opinion your "RAW" link is not a secret link. I pasted your link in a webbrowser and yes it is public readable.
I don't want my gist link public readable!

Perhaps I am totally wrong. Correct me please.
I did exactly the same procedure as MichMich described. Perhaps I forgot something.

Learning from eachother is my goal. :-)

By the way: MagicMirror is an awesome project! Top.

Greetz,

Ron

MichMich, 2020-12-08:
Of course you can also put a file on a local webserver if that is suits your needs better. Closing this issue since this is not a core bug. Feel free to reopen.

closed 2020-12-08 by MichMich

sdetweil, 2020-12-11:
@ashishtank the file must contain valid json. One can use an online validation tool to check the content. 

In json, all strings must be quoted  at least.

sdetweil, 2020-12-11:
@ashishtank correct, as I said, valid json

KingKalan, 2021-01-18:
Tried everything and still can't get this to work. Doesn't even go to default compliments it just leaves the module blank

rejas, 2021-01-18:
@KingKalan are you trying to display via remoteFile to oor is your problem worth a new issue ticket?

ashishtank, 2021-01-21:
@KingKalan can you post your config ? and url of compliments ? Do you see any error in console or network tab of developer tools of browser ?