MagicMirrorOrg/MagicMirror #2174: How can I edit the module position?

baalan92, 2020-10-22:
I want to edit the Module location. The current position is not suitable me. Therefore I want to edit the horizontal and vertical location. How can I do this, please?

sdetweil, 2020-10-22:
generally, you use the position: value as part of the module definition 

there are multiple positions available for modules, without editing any of the module code
see this post in the forums
https://forum.magicmirror.builders/topic/286/regions

after that you would use the style sheet to adjust the classes used to position the code, knowing that you may cause collisions with other modules.. 
see this topic
https://forum.magicmirror.builders/topic/6808/css-101-getting-started-with-css-and-understanding-how-css-works?_=1603138829483

sdetweil, 2020-10-26:
@baalan92  did you get what you needed??

arcangeloerrico, 2020-11-19:
@baalan92 you can just use this code in the custom.css too:

.MMM-xxx {
Position: absolute;
Left: 200px;
Top: 200px;
Z-index: 1;
}

accountnumber3, 2020-12-01:
Sorry if this has been asked. Is there a way to do left_side and right_side? I'm displaying my mirror on a chromecast so it's locked into landscape mode at 720p and there isn't a whole lot of room for each of the regions. I'd rather replace top_left and bottom_left with left_side, and just use the entire edge.

MichMich, 2020-12-08:
Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2020-12-08 by MichMich

jdlucke, 2021-02-07:
> I want to edit the Module location. The current position is not suitable me. Therefore I want to edit the horizontal and vertical location. How can I do this, please?

BAalan92, this module helped me as well .. 1. MMM-Admin Interface – Gives you a very nice looking web app to control you mirror. It’s similar to the Remote control module but a bit more robust.  https://github.com/ItayXD/MMM-Admin-Interface

sdetweil, 2021-02-07:
you would diplicate and modify the position styles in main.css to custom.css... knowing u may mess up other modules

pastormingle, 2021-08-23:
I have a really stupid question. How do you use it? I've installed it and can't seem to find away to actually start it.

sdetweil, 2021-08-23:
@pastormingle start what? MagicMirror?

without other tools

cd ~/MagicMirror
npm start

if u used my script and selected pm2, then

from any folder  pm2 status, and then

pm2 start x
where x is the name or number of the entry

pm2 start 0  ( for example)

pastormingle, 2021-08-23:
> @pastormingle start what? MagicMirror?
> 
> without other tools
> 
> cd ~/MagicMirror
> npm start
> 
> if u used my script and selected pm2, then
> 
> from any folder pm2 status, and then
> 
> pm2 start x
> where x is the name or number of the entry
> 
> pm2 start 0 ( for example)

Thanks.. But i am looking for is https://github.com/TheBodger/MMM-ModulePosition/blob/master/README.md 
the mmm moduleposition tool.

sdetweil, 2021-08-23:
ok. I assume you use your mouse.
turn off cusror:none in main.css to see it

pastormingle, 2021-08-23:
I have
html {
  cursor: none;
  overflow: hidden;
  background: var(--color-background);
  user-select: none;
  font-size: var(--font-size);
}
set to none but i don't see the grid..

sdetweil, 2021-08-23:
sorry, no idea... this is not a MagicMirror issue.. so u have to take it to the forums or ask the author..
I 'assume' you might have to CLICK on a module