MagicMirrorOrg/MagicMirror #2182: calendar module off by one day (tomorrow's events show as today)

rgabrielson11, 2020-11-02:
## I found a bug in MagicMirror

Please make sure to only submit reproducible issues. You can safely remove everything above the dividing line.
When submitting a new issue, please supply the following information:

**Platform**: Docker/Unraid 

**Node Version**: Make sure it's version 8 or later.

**MagicMirror Version**: 2.13.0

**Description**: After daylight savings time ended, calendar events are showing for today but are actually tomorrow.  relative time is counting down as if today.  this is on both Calendar and MMMCalendar module.

**Steps to Reproduce**: install module and retrieve from google calendar.  also tried with different ical (non-google), same result.

**Expected Results**: calendar events should be accurate

**Actual Results**: calendar events are inaccurate

**Configuration**: 
`/* Magic Mirror Config Sample
 *
 * By Michael Teeuw http://michaelteeuw.nl
 * MIT Licensed.
 *
 * For more information how you can configurate this file
 * See https://github.com/MichMich/MagicMirror#configuration
 *
 */

var config = {
	address: "0.0.0.0", // Address to listen on, can be:
	                      // - "localhost", "127.0.0.1", "::1" to listen on loopback interface
	                      // - another specific IPv4/6 to listen on a specific interface
	                      // - "", "0.0.0.0", "::" to listen on any interface
	                      // Default, when address config is left out, is "localhost"
	port: 8080,
	ipWhitelist: [], // Set [] to allow all IP addresses
	                                                       // or add a specific IPv4 of 192.168.1.5 :
	                                                       // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.1.5"],
	                                                       // or IPv4 range of 192.168.3.0 --> 192.168.3.15 use CIDR format :
	                                                       // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.3.0/28"],

	language: "en",
	timeFormat: 12,
	units: "imperial",

	modules: 
[        
//	{	
//	module: "MMM-DailyPokemon",
//	position: "top_left",
//	config: {
//		updateInterval: 600000,
//		minPoke: 4,
//		maxPoke: 151,
//		grayscale: true,
//		showType: true,
//		language: "en",
//		genera: true, 
//		gbaMode: true,            
//		nameSize: 26            
//		}
//	},

	{
		module: "clock",
		timezone: "America/Los_Angeles",
		position: "top_right"
	},
	{
		module: "calendar",
		position: "bottom_left",	// This can be any of the regions. Best results in left or right regions.
		config: 
		{
			colored: true,
			coloredSymbolOnly: false,
			timeFormat: "relative",
			getRelative: 0,
			wrapEvents: true,
            fetchInterval: 30000,
			maximumNumberOfDays: 1,
			maxTitleLength: 30,
			showEnd: false,
			dateEndFormat: "LT",
			fade: false,
			dateFormat: "LT",
			calendars: 
			[
				{
					url: "xxxxxxxx.ics"

				},

			],
		}
	}	
]

};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}
`
**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

sdetweil, 2020-11-03:
if you switch to the develop branch there are fixes, more coming ...
see here for instructions
https://forum.magicmirror.builders/topic/13770/2-13-calendar-fix-for-office-365-users

rgabrielson11, 2020-11-03:
did the fix, did not resolve.

updated the ical library

cd ~/MagicMirror/node_modules
mv node-ical old-node-ical
git clone https://github.com/jens-maus/node-ical.git
cd node-ical
npm install

sdetweil, 2020-11-03:
there is ONE more fix not yet integrated with MM.. 
[calendarfetcher.js.txt](https://github.com/MichMich/MagicMirror/files/5482860/calendarfetcher.js.txt)

replace the file in ~/MagicMirror/modules/default/calendar
save the old one, remove the .txt extension on this one

sdetweil, 2020-11-06:
the latest fixes are available now in the develop branch

rgabrielson11, 2020-11-06:
yep that did it, thanks

closed 2020-11-06 by rgabrielson11