MagicMirrorOrg/MagicMirror #2183: Calendar Module: Daylight Savings Offset for Created and Modified Timeframes

doubled136, 2020-11-05:
Seem to have an issue with an hour offset for recurring events created before time change.

So the sync with Google Calendar provided the automatic offset to the correct time.

It seems like the Creation date is being used to offset.

**I would entertain that this could be a bug with my calendar sync program instead.**

Example entry with issue:
BEGIN:VEVENT
DTSTART;TZID=America/New_York:20200525T093000
DTEND;TZID=America/New_York:20200525T103000
RRULE:FREQ=WEEKLY;UNTIL=20201231T143000Z;BYDAY=MO,TH
EXDATE;TZID=America/New_York:20200615T093000
DTSTAMP:20201105T130632Z
UID:[Redacted]
CREATED:20200728T184622Z
DESCRIPTION:[Redacted]
LAST-MODIFIED:20201103T131543Z
LOCATION:[Redacted]
SEQUENCE:0
STATUS:CONFIRMED
SUMMARY:[Redacted]
TRANSP:OPAQUE
END:VEVENT

Example entry without issue: 
BEGIN:VEVENT
DTSTART;TZID=America/New_York:20200702T150000
DTEND;TZID=America/New_York:20200702T153000
DTSTAMP:20201105T130632Z
UID:[Redacted]
RECURRENCE-ID;TZID=America/New_York:20200702T150000
CREATED:20200728T184647Z
DESCRIPTION:[Redacted]
LAST-MODIFIED:20201102T121945Z
LOCATION:[Redacted]
SEQUENCE:0
STATUS:CONFIRMED
SUMMARY:[Redacted]
TRANSP:OPAQUE
END:VEVENT

hangorazvan, 2020-11-05:
read https://github.com/MichMich/MagicMirror/issues/2182

sdetweil, 2020-11-06:
new fixes are available in the develop branch

doubled136, 2020-11-09:
I have develop version 2.14.0  Is it a fix past that?

sdetweil, 2020-11-09:
no, but it is the collection of fixes before next release, and we keep updating it 

so you will need to go git pull and then npm I stall again

doubled136, 2020-11-09:
Ok, thank you!

closed 2020-11-09 by doubled136