MagicMirrorOrg/MagicMirror #2188: Annual Events not shown properly

dvelabs, 2020-11-08:
Events set as 'Annual' in google calendar do not show properly

Following event is shown in my mirror as 'Birthday of ZZZ  in 2 days' even though it's a Bday set for April 2021

> BEGIN:VEVENT
> DTSTART;VALUE=DATE:20200409
> DTEND;VALUE=DATE:20200410
> RRULE:FREQ=**YEARLY**;WKST=MO;INTERVAL=1;BYMONTHDAY=9
> DTSTAMP:20201108T095611Z
> UID:XXXXXXXXXXXXXXXXXXX@google.com
> CREATED:20200713T201307Z
> DESCRIPTION:
> LAST-MODIFIED:20200713T201307Z
> LOCATION:
> SEQUENCE:1
> STATUS:TENTATIVE
> SUMMARY:Birthday of ZZZ
> TRANSP:OPAQUE
> END:VEVENT

Probably this has to do with the DTSTART and DTEND fields 
**Following event works properly**

> BEGIN:VEVENT
> DTSTART:20201109T090000Z
> DTEND:20201109T100000Z
> DTSTAMP:20201108T104414Z
> UID:YYYYYYYYYYYYYYYY@google.com
> CREATED:20201030T225632Z
> DESCRIPTION:
> LAST-MODIFIED:20201105T212545Z
> LOCATION:
> SEQUENCE:5
> STATUS:CONFIRMED
> SUMMARY:Dentist
> TRANSP:OPAQUE
> END:VEVENT

sdetweil, 2020-11-14:
did u switch to the develop branch which has a bunch of calendar fixes? 

please see the instructions here 
https://forum.magicmirror.builders/topic/13770/2-13-calendar-fix-for-office-365-users/2

sdetweil, 2020-11-15:
i put this with the new fixes and it shows dentist entry, but I had to change the date on Birthday to Dec (1209)  to get it to display

closed 2021-01-23 by stale[bot]