MagicMirrorOrg/MagicMirror #2192: Calendar Enhancement - End Date for dateheaders Time Format

doubled136, 2020-11-10:
Not really sure how to pull or merge for a feature enhancement.  In the Calendar module, I noticed that when "dateheaders" is selected, "showEnd" doesn't do anything but it would be reasonable to allow the option to show an end time, especially since showEnd is false by default.

From line 273 in calendar.js, I added an if statement to check for showEnd and publish the end time of an event:
					timeWrapper.innerHTML = moment(event.startDate, "x").format("LT");
					//Add for End Date from showEnd for dataheaders
					if (this.config.showEnd) {
					timeWrapper.innerHTML += " - " + moment(event.endDate, "x").format("LT");
					}

Not sure if this is helpful but I figured I would put it out there.

MichMich, 2020-12-08:
Could you please send a PR (if its not already fixed in the develop branch). Thanks!

sbehrends, 2021-01-08:
I am happy to collaborate on this fix 👍 

@doubled136, could you describe a bit how to reproduce the issue?

wrongwaymarv, 2021-01-28:
I was able to reproduce this issue. I have the following set in my config.js for the default calendar module:

```
timeFormat: "dateheaders",
showEnd: true,

```
But the calendar on my MagicMirror only shows the start time of events and not the end time. I can confirm that adding the above code works (although for me it was line 312 in calendar.js, not line 273).

Reddax, 2022-05-07:
> Not really sure how to pull or merge for a feature enhancement. In the Calendar module, I noticed that when "dateheaders" is selected, "showEnd" doesn't do anything but it would be reasonable to [...]

This fix works for me too.

rejas, 2022-05-08:
I might take a stab at this and provide a PR, could you assign this issue to me @MichMich ?

rejas, 2022-05-10:
Created a PR for @doubled136 fix, maybe you can check it out?

khassel, 2022-10-03:
seems to be fixed with the above PR, so should be closed @doubled136 @MichMich

PR #2848 (2022-05-18): Show endDate for calendar events when using dateHeaders
Fixes #2192

As reported by @doubled136 this adds the endDate for events when using the dateHeaders layout.
closed 2022-10-04 by MichMich