MagicMirrorOrg/MagicMirror #2203: change base branch fir pull requests

sdetweil, 2020-11-16:
Mich, could u change the base branch in the repo settings so we can't accidentally PR to master..

rejas, 2020-11-16:
As long as it gets documented in the docs, I approve this!

sdetweil, 2020-11-16:
I don't think anything about PRs is in the docs now..  and everyone does pr to master, which is wrong.. 

lets prevent all that trouble

sdetweil, 2020-11-16:
but it says this

The default branch is considered the “base” branch in your repository, against which all pull requests and code commits are automatically made, **unless you specify a different branch.**

MichMich, 2020-11-16:
Yes. But it also specifies which branch somebody sees at first glance when he/she visits the repo. Or which branch they pull when they don't define a specific branch while pulling.

sdetweil, 2020-11-16:
booooooo

sdetweil, 2020-11-16:
closing...dang it!..

closed 2020-11-16 by sdetweil

MichMich, 2020-11-16:
But maybe good to know. You can always simply change the branch to which you made a PR. You don't need to make a new PR. 

And usually when I see it's the wrong branch, I just change it. 

(Can't make a screenshot on how to do it since I'm on mobile.)

Also good to know: the master branch is protected. So I can never accidentally merge into master.

sdetweil, 2020-11-16:
yes, I knew all that, but just another annoying thing to take care of.. 

my problem is catching up to where develop is.. without having to make a new fork..

MichMich, 2020-11-16:
Never tried it but can't you do something like `git merge origin/master`?

sdetweil, 2020-11-16:
not master.. develop..   I got behind, you did a bunch of merges, I want to submit a new PR but am behind..  so I pull from develop to my local, should be in synch.. now push to my fork... oops..  unknown commits.. so when I create a PR they all come too... look at the closed PR 2201

MichMich, 2020-11-16:
Yeah. Sorry develop. Edited my reply but you were to quick. 
Maybe you can do a rebase in stead of a merge.

sdetweil, 2020-11-16:
will try rebase next time