MagicMirrorOrg/MagicMirror #2211: Google Assistant is not launching on my Magic Mirror, gives me an eaddrinuse issue

akeysny, 2020-11-25:
## I'm not sure if this is a bug

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)

## I'm having troubles installing or configuring MagicMirror

Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

## I found a bug in the MagicMirror installer

If you are facing an issue or found a bug while trying to install MagicMirror via the installer please report it in the respective GitHub repository:
[https://github.com/sdetweil/MagicMirror_scripts](https://github.com/sdetweil/MagicMirror_scripts)

## I found a bug in the MagicMirror Docker image

If you are facing an issue or found a bug while running MagicMirror inside a Docker container please create an issue in the GitHub repository of the MagicMirror Docker image:
[https://github.com/bastilimbach/docker-MagicMirror](https://github.com/bastilimbach/docker-MagicMirror)

---

## I found a bug in MagicMirror

Please make sure to only submit reproducible issues. You can safely remove everything above the dividing line.
When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version _and_ your hardware (Raspberry Pi 2/3, Windows, Mac, Linux, System V UNIX).

**Node Version**: Make sure it's version 8 or later.

**MagicMirror Version**: Please let us now which version of MagicMirror you are running. It can be found in the `package.log` file.

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce**: List the step by step process to reproduce the issue.

**Expected Results**: Describe what you expected to see.

**Actual Results**: Describe what you actually saw.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

**Additional Notes**: Trying to launch Google assistant and when i start my mirror this 

happens

[image]

sdetweil, 2020-11-25:
Not a MagicMirror problem. Eaddressinuse means mm is already running on the same port.

If using pm2, do

pm2 stop  all

Then u can start a new instance

pm2 status will show u if something is running

MichMich, 2020-12-08:
Please open an issue in the repo of the module.

closed 2020-12-08 by MichMich