MagicMirrorOrg/MagicMirror #2221: MMM-weatherforecast. Icon problem - still displays the same.

majdzik84, 2020-12-02:
Today I noticed that the icons in the module are not updating. It’s still wi-night-cloudy.
I don’t know at what point it broke. I know it used to work properly.
Can I fix it somehow?

[image]
[image]

MichMich, 2020-12-08:
Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

majdzik84, 2020-12-08:
I believe the problem should remain open.
It also appears in the main module of magic mirror (weatherforcast). I found that this is not a problem with non-changing icons as they change but only in NIGHT mode. The icons do not switch to DAY. So I always have a moon icon - even at 12.00 when the sun should be.
Could you please check with yourself?

MichMich, 2020-12-08:
The 401 error is caused by the limitation of your (free account) which can be solved by limiting the number of days via the config. Please refer to the forum for more info. The icons seems to work fine here. Are  there more reports by users with the same issue? Is it a confirmed bug?

majdzik84, 2020-12-08:
I suggest not to look at the error because I have already figured it out.

The icons that do not change (day / night) are a mistake that I confirmed when shopping for 2 people because I asked so many

12:30
[image]

14:40
[image]

MichMich, 2020-12-08:
Reopened.

ashishtank, 2020-12-09:
Hi @majdzik84 

You are referring to below icons ? If yes, then I could also reproduce the issue and I know the cause so might also able fix it.
[image]

PS: Screenshot taken at 10:47 AM in the morning.
--
Ashish

majdzik84, 2020-12-09:
that's what these icons are all about. We would appreciate the repair.

ashishtank, 2020-12-09:
[Fixed and Created pull request](https://github.com/MichMich/MagicMirror/pull/2224)
Issue was due to locale, Hours comparison with 8-17 hours range was failing.

closed 2020-12-21 by MichMich

majdzik84, 2020-12-27:
Unfortunately, the problem persists for me. I updated to the latest version but the icons are still night time.

ashishtank, 2020-12-27:
@majdzik84 : can you check if you have below code in your current weather module file ? Looks like you are using https://github.com/jharttech/MMM-weatherforecast and not default current weather module. In https://github.com/jharttech/MMM-weatherforecast pull request is not merged yet.

```javascript
if (forecast.dt_txt) {
				day = moment(forecast.dt_txt, "YYYY-MM-DD hh:mm:ss").format("ddd");
				hour = moment(forecast.dt_txt, "YYYY-MM-DD hh:mm:ss").toDate().getHours(); //This line
			} else {
				day = moment(forecast.dt, "X").format("ddd");
				hour = moment(forecast.dt, "X").toDate().getHours(); //And this line
			}

```

majdzik84, 2020-12-27:
@ashishtank I am using the default module. I have the code you entered in the default weather module.

[image]

[image]

ashishtank, 2020-12-28:
@majdzik84 Can you check in dev tools if debugger is reaching to below lines ? The line 404 sets the correct icon. 

[image]

Another option is to check with English language for MM, if that solves the issue then we can confirm that my findings were correct.

majdzik84, 2020-12-28:
After changing to English, the icons are still in the night mode change. :(

[image]

ashishtank, 2020-12-28:
Can you send me JSON data that module is receiving from weather provider ?

majdzik84, 2020-12-28:
@ashishtank  no problem if you write me how can i get it / where to look?

ashishtank, 2020-12-28:
@majdzik84 
1. Write below line in weatherforcast.js processWeather function as shown in image below
2. console.log(JSON.stringify(data));
```javascript
processWeather: function (data) {
		// Forcast16 (paid) API endpoint provides this data.  Onecall endpoint
		// does not.
		console.log(JSON.stringify(data)); //This line
```
3. Open dev tools, refresh the mirror
4. check the dev tools console tab
5. click on copy button as highlighted in image 
6. paste it here or share using pastebin.org
7. Also send me your config language

[image]

majdzik84, 2020-12-28:
Thank you for the instructions.

language: "pl",

`{"cod":"200","message":0,"cnt":25,"list":[{"dt":1609189200,"main":{"temp":4.61,"feels_like":-1.32,"temp_min":3.71,"temp_max":4.61,"pressure":989,"sea_level":989,"grnd_level":961,"humidity":68,"temp_kf":0.9},"weather":[{"id":802,"main":"Clouds","description":"rozproszone chmury","icon":"03n"}],"clouds":{"all":32},"wind":{"speed":5.47,"deg":132},"visibility":10000,"pop":0,"sys":{"pod":"n"},"dt_txt":"2020-12-28 21:00:00"},{"dt":1609200000,"main":{"temp":5.14,"feels_like":-1.57,"temp_min":5.05,"temp_max":5.14,"pressure":987,"sea_level":987,"grnd_level":959,"humidity":72,"temp_kf":0.09},"weather":[{"id":803,"main":"Clouds","description":"pochmurno z przejaśnieniami","icon":"04n"}],"clouds":{"all":65},"wind":{"speed":6.86,"deg":118},"visibility":10000,"pop":0,"sys":{"pod":"n"},"dt_txt":"2020-12-29 00:00:00"},{"dt":1609210800,"main":{"temp":5.41,"feels_like":-0.54,"temp_min":5.41,"temp_max":5.41,"pressure":984,"sea_level":984,"grnd_level":956,"humidity":78,"temp_kf":0},"weather":[{"id":500,"main":"Rain","description":"słabe opady deszczu","icon":"10n"}],"clouds":{"all":94},"wind":{"speed":6.08,"deg":129},"visibility":10000,"pop":0.59,"rain":{"3h":0.6},"sys":{"pod":"n"},"dt_txt":"2020-12-29 03:00:00"},{"dt":1609221600,"main":{"temp":2.44,"feels_like":-4.74,"temp_min":2.42,"temp_max":2.44,"pressure":986,"sea_level":986,"grnd_level":958,"humidity":94,"temp_kf":0.02},"weather":[{"id":500,"main":"Rain","description":"słabe opady deszczu","icon":"10n"}],"clouds":{"all":99},"wind":{"speed":7.77,"deg":231},"visibility":258,"pop":0.91,"rain":{"3h":1.6},"sys":{"pod":"n"},"dt_txt":"2020-12-29 06:00:00"},{"dt":1609232400,"main":{"temp":3.53,"feels_like":-4.56,"temp_min":3.53,"temp_max":3.53,"pressure":992,"sea_level":992,"grnd_level":964,"humidity":74,"temp_kf":0},"weather":[{"id":500,"main":"Rain","description":"słabe opady deszczu","icon":"10d"}],"clouds":{"all":67},"wind":{"speed":8.58,"deg":215},"visibility":10000,"pop":0.53,"rain":{"3h":0.13},"sys":{"pod":"d"},"dt_txt":"2020-12-29 09:00:00"},{"dt":1609243200,"main":{"temp":6.25,"feels_like":-0.57,"temp_min":6.25,"temp_max":6.25,"pressure":995,"sea_level":995,"grnd_level":967,"humidity":62,"temp_kf":0},"weather":[{"id":803,"main":"Clouds","description":"pochmurno z przejaśnieniami","icon":"04d"}],"clouds":{"all":67},"wind":{"speed":6.8,"deg":209},"visibility":10000,"pop":0.38,"sys":{"pod":"d"},"dt_txt":"2020-12-29 12:00:00"},{"dt":1609254000,"main":{"temp":4.09,"feels_like":-0.59,"temp_min":4.09,"temp_max":4.09,"pressure":997,"sea_level":997,"grnd_level":968,"humidity":73,"temp_kf":0},"weather":[{"id":800,"main":"Clear","description":"bezchmurnie","icon":"01n"}],"clouds":{"all":8},"wind":{"speed":3.78,"deg":180},"visibility":10000,"pop":0,"sys":{"pod":"n"},"dt_txt":"2020-12-29 15:00:00"},{"dt":1609264800,"main":{"temp":3.96,"feels_like":0.06,"temp_min":3.96,"temp_max":3.96,"pressure":999,"sea_level":999,"grnd_level":971,"humidity":77,"temp_kf":0},"weather":[{"id":802,"main":"Clouds","description":"rozproszone chmury","icon":"03n"}],"clouds":{"all":30},"wind":{"speed":2.8,"deg":206},"visibility":10000,"pop":0,"sys":{"pod":"n"},"dt_txt":"2020-12-29 18:00:00"},{"dt":1609275600,"main":{"temp":3.99,"feels_like":-1.08,"temp_min":3.99,"temp_max":3.99,"pressure":1002,"sea_level":1002,"grnd_level":973,"humidity":84,"temp_kf":0},"weather":[{"id":803,"main":"Clouds","description":"pochmurno z przejaśnieniami","icon":"04n"}],"clouds":{"all":67},"wind":{"speed":4.74,"deg":209},"visibility":10000,"pop":0,"sys":{"pod":"n"},"dt_txt":"2020-12-29 21:00:00"},{"dt":1609286400,"main":{"temp":2.81,"feels_like":-1.04,"temp_min":2.81,"temp_max":2.81,"pressure":1002,"sea_level":1002,"grnd_level":974,"humidity":90,"temp_kf":0},"weather":[{"id":803,"main":"Clouds","description":"pochmurno z przejaśnieniami","icon":"04n"}],"clouds":{"all":65},"wind":{"speed":2.95,"deg":169},"visibility":10000,"pop":0,"sys":{"pod":"n"},"dt_txt":"2020-12-30 00:00:00"},{"dt":1609297200,"main":{"temp":3.94,"feels_like":0,"temp_min":3.94,"temp_max":3.94,"pressure":1002,"sea_level":1002,"grnd_level":974,"humidity":89,"temp_kf":0},"weather":[{"id":500,"main":"Rain","description":"słabe opady deszczu","icon":"10n"}],"clouds":{"all":100},"wind":{"speed":3.31,"deg":147},"visibility":10000,"pop":0.52,"rain":{"3h":0.58},"sys":{"pod":"n"},"dt_txt":"2020-12-30 03:00:00"},{"dt":1609308000,"main":{"temp":3.94,"feels_like":0.17,"temp_min":3.94,"temp_max":3.94,"pressure":1003,"sea_level":1003,"grnd_level":975,"humidity":87,"temp_kf":0},"weather":[{"id":500,"main":"Rain","description":"słabe opady deszczu","icon":"10n"}],"clouds":{"all":100},"wind":{"speed":2.99,"deg":155},"visibility":8132,"pop":0.72,"rain":{"3h":1},"sys":{"pod":"n"},"dt_txt":"2020-12-30 06:00:00"},{"dt":1609318800,"main":{"temp":3.78,"feels_like":0.53,"temp_min":3.78,"temp_max":3.78,"pressure":1004,"sea_level":1004,"grnd_level":976,"humidity":94,"temp_kf":0},"weather":[{"id":500,"main":"Rain","description":"słabe opady deszczu","icon":"10d"}],"clouds":{"all":100},"wind":{"speed":2.48,"deg":196},"visibility":10000,"pop":0.91,"rain":{"3h":2.46},"sys":{"pod":"d"},"dt_txt":"2020-12-30 09:00:00"},{"dt":1609329600,"main":{"temp":3.95,"feels_like":1.37,"temp_min":3.95,"temp_max":3.95,"pressure":1004,"sea_level":1004,"grnd_level":976,"humidity":95,"temp_kf":0},"weather":[{"id":500,"main":"Rain","description":"słabe opady deszczu","icon":"10d"}],"clouds":{"all":100},"wind":{"speed":1.6,"deg":208},"visibility":10000 [... cut ...] emp_min":-0.11,"temp_max":-0.11,"pressure":1010,"sea_level":1010,"grnd_level":981,"humidity":95,"temp_kf":0},"weather":[{"id":804,"main":"Clouds","description":"całkowite zachmurzenie","icon":"04n"}],"clouds":{"all":93},"wind":{"speed":1.18,"deg":22},"visibility":10000,"pop":0,"sys":{"pod":"n"},"dt_txt":"2020-12-31 21:00:00"}],"city":{"id":3094597,"name":"Krupski Młyn","coord":{"lat":50.5734,"lon":18.6225},"country":"PL","population":0,"timezone":3600,"sunrise":1609137993,"sunset":1609166856}}
`

ashishtank, 2020-12-29:
@majdzik84 Thanks for the weather data, it helped in getting the root cause. In this data for each day each time slot also have icon with it. we have logic to set the icon from 8 to 17 hours time frame to show as main icon as below.

```javascript
// Since we don't want an icon from the start of the day (in the middle of the night)
// we update the icon as long as it's somewhere during the day.
if (hour >= 8 && hour <= 17) {
	forecastData.icon = this.config.iconTable[forecast.weather[0].icon];
}
```
But if any last time slot data contains the night time icon then it will display night icon only. In your provided data I noticed the same. for time slot 2020-12-30 15:00:00 to 2020-12-30 18:00:00 it is night icon and thus we are setting the night icons. To prove this I did changed the condition to <=14 hours and then it solved the issue but this is not foolproof solution.
```javascript
if (hour >= 8 && hour <= 14) {
	forecastData.icon = this.config.iconTable[forecast.weather[0].icon];
}
```
[image]

Last time slot is having night icons(though it is day time, may be its time for sun set ? )
[image]

I see two solutions to this problem.
1. We should have configurable day start and end time
2. We should always convert the night type icon to day time in this day slot. 10n should become 10d and 02n -> 02d.

I am not sure which one we should implement. Some senior member of the MM community should suggest.

CC: @MichMich @rejas

MichMich, 2020-12-29:
I don't think it needs to be configurable. The weather app knows which time the sun comes up and goes down. We can use those timestamps (and fallback to 06.00 and 18.00).

ashishtank, 2020-12-30:
@MichMich Thanks for the suggestion. I propose below fix.

```javascript

var dayStarts = 8;
var dayEnds = 17;

if (data.city && data.city.sunrise && data.city.sunset) {
	dayStart = moment.unix(data.city.sunrise).toDate().getHours();
	dayEnds = moment.unix(data.city.sunset).toDate().getHours();
}

.....
.....

if (hour > dayStarts && hour < dayEnds) {
   forecastData.icon = this.config.iconTable[forecast.weather[0].icon];
}

```
note that I have changed condition from hours >= offset to hours > offset (removed =) same for hour < offset. Reason is that when sunset time is 15:47 still open weather return night icon for 15:00 hour slot.

Let me know if this fix is correct, I will send pull request.

CC : @majdzik84

MichMich, 2020-12-30:
Aside from a small typo inside your if statement (dayStart in stead of dayStarts) it looks good. 

A PR is welcome. Would be great if you can also add tests to check this feature.

ashishtank, 2020-12-30:
@MichMich Thanks, I also noticed typo after post. I will try to do unit test as well.

ashishtank, 2021-01-10:
PR : #2413

ashishtank, 2021-01-10:
One test case is failing, and i am not sure how to fix that.. The test is unrelated to my changes (or at least it looks like)

rejas, 2021-01-12:
Does the test run fine locally on your branch?

ashishtank, 2021-01-12:
Yes, they are working fine. Also working for another pull request.
[image]

MichMich, 2021-01-12:
I just restarted the tests. Let's see if that helps.

ashishtank, 2021-01-12:
Looks like it works now !