MagicMirrorOrg/MagicMirror #2227: Blackscreen after new module (mmm-systemtemperature)

jlnxaer, 2020-12-11:
Hello,

I just tried to "install" a new Module called mmm-systemtemperature.
After inserting the config and rebooting the Mirror I just get a blackscreen. 

My PM2 log shows the following:
`0|MagicMir | npm ERR! A complete log of this run can be found in:`
`0|MagicMir | npm ERR!     /home/pi/.npm/_logs/2020-12-10T14_43_28_756Z-debug.log`
`0|MagicMir | npm ERR! code ELIFECYCLE`
`0|MagicMir | npm ERR! errno 1`
`0|MagicMir | npm ERR! magicmirror@2.13.0 start: DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js`
`0|MagicMir | npm ERR! Exit status 1`
`0|MagicMir | npm ERR!`
`0|MagicMir | npm ERR! Failed at the magicmirror@2.13.0 start script.`
`0|MagicMir | npm ERR! This is probably not a problem with npm. There is likely additional logging output above.`

Can anybody help me find a solution for this? :(

Thank you in advance!
Greetings, 
JLN

sdetweil, 2020-12-11:
I would guess that you didn't run npm install in the module folder. Or it failed

MichMich, 2020-12-11:
I must admit I don't really maintain that module very well. If I'm not mistaken there was a PR merged recently which might be the issue. I didn't check test the module after merging. Might be worth investigating that.

jlnxaer, 2020-12-11:
> I would guess that you didn't run npm install in the module folder. Or it failed

If it failed, can i just try again and see for any error messages or do I need to deinstall it again? And if I have to deinstall it, how do I do it?

sdetweil, 2020-12-11:
If it failed in the module folder. Just erase the node_modules folder (in the module folder) then run npm install again

sdetweil, 2020-12-11:
There should be a log file in the ~/.npm folder

jlnxaer, 2020-12-12:
So I reinstalled it now and that's the error that I get:
```
npm WARN mocha-logger@1.0.7 requires a peer of mocha@^8.1.3 but none is installed. You must install peer dependencies yourself.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@2.1.3 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.1.3: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
```

plus a few vulnerabilities witch I tried to fix and got this error:
```
npm ERR! code EAI_AGAIN
npm ERR! errno EAI_AGAIN
npm ERR! request to https://registry.npmjs.org/@babel%2fpolyfill failed, reason: getaddrinfo EAI_AGAIN registry.npmjs.org registry.npmjs.org:443
npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-12-12T10_38_01_165Z-debug.
```

sdetweil, 2020-12-12:
The first are not errors, just warnings. 

EAI_AGAIN is a name server error.

I have seen this most often when someone is using an advertising filter like pinhole.