MagicMirrorOrg/MagicMirror #2362: calendar fetch fails with header size exceeded, breaking node js change

sdetweil, 2020-12-19:
forum topic 
https://forum.magicmirror.builders/topic/14257/overflow-with-calendar

see https://github.com/nodejs/node/issues/24692

node has made the max header size 8000 down from 80,000.  we have some bad servers which are sending the data in the headers too.. 
```
[2020-12-19 08:28:33.818] [ERROR]  Calendar Error. Could not fetch calendar:  https://sfxbrugge.smartschool.be/index.php/deeplink/0426/f90a2a81-5fdf-416f-a52a-1316eee8efc9 Error: Parse Error
    at TLSSocket.socketOnData (_http_client.js:452:22)
    at TLSSocket.emit (events.js:200:13)
    at addChunk (_stream_readable.js:294:12)
    at readableAddChunk (_stream_readable.js:275:11)
    at TLSSocket.Readable.push (_stream_readable.js:210:10)
    at TLSWrap.onStreamRead (internal/stream_base_commons.js:166:17) {
  bytesParsed: 8264,
  code: 'HPE_HEADER_OVERFLOW',
  reason: 'Header overflow'
```

the url above is invalidated(by me)  so it could be posted here.. I can advise someone else the corrected url

the workaround  --max-http-header-size=?????? some number doesn't work 
```
 node ./node_modules/.bin/electrons/electron.js --max-http-header-size=10000
```
I upgraded to latest stable node, 14.15.3

same error

I added the  maxHeaderSize value to the request options

```
		const opts = {
			headers: {
				"User-Agent": "Mozilla/5.0 (Node.js " + nodeVersion + ") MagicMirror/" + global.version + " (https://github.com/MichMich/MagicMirror/)"
			},
			maxHeaderSize: 12000,
			gzip: true
		};
```

same error... 

i don't know any other approach to resolving this..  help

khassel, 2020-12-19:
The downgrade from 80KB to 8KB was made 2 years ago.

running `node --help` v14.15.3 says 16KB:
````
  --max-http-header-size=...                            set the maximum size of HTTP headers (default: 16384 (16KB))
````

Did you try `node --max-http-header-size=80000 ./serveronly`?

sdetweil, 2020-12-19:
no, will try, however  note the failure above at 8264  on node 14.15.3

need to do this on non-serveronly

we are using request in calendar.

sdetweil, 2020-12-20:
@khassel  for node 10.23.0 without the parm , serveronly fails, with the parm does NOT fail.
for 14.15.3 does not fail with or without the parm 

so, how do we specify this for the 
electron version?

khassel, 2020-12-20:
from the electron docs [this](https://www.electronjs.org/docs/api/environment-variables#node_options) should work.

sdetweil, 2020-12-20:
@khassel  that worked.. i looked thru the electron docs for parameters,  not options

khassel, 2020-12-20:
glad to hear this is solved. Do we need a fix in this repo? 

I think not, node `v14` is the current lts version and has already 16KB header size. The forum issue was the first after 2 years.
What do you think?

sdetweil, 2020-12-20:
i do not think we need to implement any fixes. 
I was waiting til the user confirmed it worked for him. but it worked for me.

and the workaround is easy to do

I appreciate your help

sdetweil, 2020-12-23:
user confirmed parameter resolved problem

closed 2020-12-23 by sdetweil