MagicMirrorOrg/MagicMirror #2363: URGENT: Looking for an alternative for Travis CI.

MichMich, 2020-12-20:
Currently we are using Travis for automated testing of all branches and the PR's. Unfortunately, Travis is no longer free and their price plan is pretty steep. Therefor it might be necessary to switch to an alternative solution.

Please let me know if you know (affordable) solution, of know how to solve this using GitHub actions. Any help on this topic is welcome! 

Good to know: I prefer to have a solution which doesn't require the maintenance of a self hosted service. So we are looking for solutions similar to Travis and Circle CI.

Thanks!

rejas, 2020-12-20:
I read on travis-ci.org that it is shutting down but didnt read anything more yet. 
First came this to my mind, their claim: "Free for open source. We love the Open Source Community, and to show how much we love it, upon validated request placed with our Support Team you may receive free OSS credits for your public builds." So is that a possibility, tell them about MM and if they give you credits for that?
Otherwise github actions would be probably my choice, but besides automatic deployment to the github.io page (for example in https://github.com/Neofonie/nemory) I havent much done yet with it (because travis was there :-)

MichMich, 2020-12-20:
I quickly whipped up a test using GitHub actions, which runs nicely, but due to electron, it runs into issues:
https://github.com/MichMich/MagicMirror/blob/ci-test/.github/workflows/node-ci.js.yml

It might be a good starting point. Feel free to send a PR to the `ci-test` branch if you have an idea how to solve this.

khassel, 2020-12-20:
Fixed with https://github.com/MichMich/MagicMirror/pull/2364.

Additionally the `.travis.yml` should be removed.
And we need the testing on `develop` too (`ci_test` has parent `master`).

dwebfan, 2020-12-21:
> I quickly whipped up a test using GitHub actions, which runs nicely, but due to electron, it runs into issues:
> https://github.com/MichMich/MagicMirror/blob/ci-test/.github/workflows/node-ci.js.yml [...]

Is it a good idea to remove dependency of core of magic mirror out of electron. Electron build would be similiar as server-only option. User can choose build mode. It would be also easier from testing point of view

MichMich, 2020-12-21:
Thanks khassel! So anyone knows what the deal is? Are github actions free? Is this a full replacement for Travis?

MichMich, 2020-12-21:
> Is it a good idea to remove dependency of core of magic mirror out of electron. Electron build would be similiar as server-only option. User can choose build mode. It would be also easier from testing point of view

Maybe a good idea to open a separate Issue for this so we can discuss the possibilities.

MichMich, 2020-12-21:
> Thanks khassel! So anyone knows what the deal is? Are github actions free? Is this a full replacement for Travis?

It seems it is: 
https://github.blog/2019-08-08-github-actions-now-supports-ci-cd/ 

🎉

rejas, 2020-12-21:
> Thanks khassel! So anyone knows what the deal is? Are github actions free? Is this a full replacement for Travis?

As far as I know its free for public repos: https://github.com/features/actions#pricing-details

MichMich, 2020-12-21:
Cool. The only thing missing now is the Danger.js integration to check if someone added a changelog. This is available as a GitHub action as well. So we can make that a separate action. 

I'll merge the Travis replacement today so we can test it on the open PR's.

MichMich, 2020-12-21:
It looks like the Github Action for testing is working. All open PR's passed. Just to test: could someone send a failing PR so we can test it for sure?

I tried implementing danger, but it turns out danger js can't work in GitHub Actions on a PR: https://github.com/PrismJS/prism/issues/2627

There are a few CHANGELOG enforcing actions available. So Will look into that in the future.

MichMich, 2020-12-21:
Just posted a PR from a secondary account which fails as expected: https://github.com/MichMich/MagicMirror/pull/2368#issuecomment-748921773

🎉

So the only thing missing now is enforcing CHANGELOG.

MichMich, 2020-12-21:
Changelogs are enforced now. Thanks for everyone's help.

closed 2020-12-21 by MichMich