MagicMirrorOrg/MagicMirror #2373: How to disable a specific (or group of) modules, when another is activated?

bert269, 2020-12-24:
I'm not sure if this is a bug, but I have the need to disable a specific module, when another gets activated.
I know that recently there was a global variable (disabled: true/false) introduced and I was thinking of disabling the required module(s) with this.
So I defined another global variable, called **_dismod_** with value 0 (maining false) . My plan was when the trigger module gets activated, aka a "screencast", to change the value of **_dismod_** to 1 and ensure that the required module gets disabled - but I cannot seem to make it work. When I test for the value of **_dismod_** in the specific module: if (dismod > 0) {disabled: true}    - I receive and error. The magic Mirror does not like the "("
What am I missing here - is this even possible?

sdetweil, 2020-12-24:
when  'hide' a module, it received a call to the is suspend() function.. which is 'like' disable, if the module implemented it. 
this is how pages works..

bert269, 2020-12-24:
Thanks for the quick reply. Let me see where I can find more about this and how to enable it for the specific module(s).

sdetweil, 2020-12-24:
@bert269 I use this in my MMM-SleepWake to blank the screen where can't turn off/on monitor with software 

https://github.com/sdetweil/MMM-SleepWake

doc
https://docs.magicmirror.builders/development/core-module-file.html#subclassable-module-methods
https://docs.magicmirror.builders/development/helper-methods.html#module-selection

in my code 
```
case "SLEEP_HIDE":
                        // loop thru all the modules running
			MM.getModules().enumerate((module) => {
				// if the module is already hidden				
				if(module.hidden==true)
					// save it for wake up
					{self.previously_hidden.push(module.identifier);}
				else
					// hide this module
					{module.hide(1000);}
			});
```

bert269, 2020-12-24:
Sorry - but I cannot figure this out/get my mind around it.
Thanks for your help anyway.

sdetweil, 2020-12-24:
@bert269  in the modulename.js (the js file whose filename matches the module name) , u can use the  MM.getModules()  function to return a list of all the loaded modules. 

u can then operate on their data (module.hidden, module.name, module.identifier)...
https://docs.magicmirror.builders/development/core-module-file.html#available-module-instance-properties

and then call methods on their instance, module.hide()... module.show()

in my code above, I want to hide all modules, but I want to remember the ones that were already hidden (by pages or whatever), 
so that when I show all the modules, I only show the ones my module hid.  (return screen to prior state)

closed 2020-12-24 by MichMich