MagicMirrorOrg/MagicMirror #2382: Tag disabled returns undefined on getModules()

ezeholz, 2020-12-28:
## I found a bug in MagicMirror

Please make sure to only submit reproducible issues. You can safely remove everything above the dividing line.
When submitting a new issue, please supply the following information:

**Platform**: VM Ubuntu 18.04 and https://github.com/Jopyth/MMM-Remote-Control/issues/242#issuecomment-751563402
Electron installed with MM version.

**Node Version**: v14.15.0

**MagicMirror Version**: v2.13.0

**Description**: MagicMirror gives identifiers and a place inside `getModules()` array to modules that are disabled with the `disabled` tag.

**Steps to Reproduce**: 
1) Install MagicMirror Manually
2) Add the `disabled: true,` tag inside any default module. Tried in news module.
3) Start MM with `npm start`
4) In Electron, open dev console using CTRL + SHIFT + I and write `MM.getModules()`

**Expected Results**: You should see every module described as a class, and those modules that are disabled should never be returned. Not even as `undefined`

**Actual Results**: 
[image]
You see not only above but in the numbers and identifiers that the number 5 is skipped as an identifier and as an integer in the array. 

**Configuration**:
```js
/*************** AUTO GENERATED BY REMOTE CONTROL MODULE ***************/

var config = 
{
  address: '0.0.0.0',
  electronOptions: {
    fullscreen: false
  },
  ipWhitelist: [],
  modules: [
    {
      module: 'MMM-Remote-Control',
      position: 'top_right',
      header: '',
      config: {
        showModuleApiMenu: true,
        customMenu: 'custom.json',
        secureEndpoints: false,
        classes: {
          test: {
            toggle: 'calendar'
          },
          'This is a test': {
            hide: [
              'calendar',
              'newsfeed'
            ],
            toggle: 'clock'
          },
          'this is another test': {
            show: [
              'calendar',
              'newsfeed'
            ],
            hide: [
              'clock'
            ]
          }
        }
      }
    },
    {
      module: 'updatenotification',
      position: 'top_bar',
      config: {}
    },
    {
      module: 'clock',
      position: 'top_left',
      header: '',
      config: {}
    },
    {
      module: 'calendar',
      header: 'US Holidays',
      position: 'top_left',
      config: {
        calendars: [
          {
            symbol: 'calendar-check',
            url: 'webcal://www.calendarlabs.com/ical-calendar/ics/76/US_Holidays.ics'
          }
        ]
      }
    },
    {
      module: 'compliments',
      position: 'lower_third',
      classes: 'test',
      config: {}
    },
    {
      module: 'newsfeed',
      position: 'bottom_bar',
      disabled: true, // Here it's the disabled module
      config: {
        feeds: [
          {
            title: 'New York Times',
            url: 'https://rss.nytimes.com/services/xml/rss/nyt/HomePage.xml'
          }
        ]
      }
    },
    {
      module: 'alert',
      header: '',
      config: {
        welcome_message: true
      }
    }
  ],
  useHttps: false,
  httpsPrivateKey: '',
  httpsCertificate: '',
  logLevel: [
    'INFO',
    'LOG',
    'WARN',
    'ERROR'
  ]
}

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== 'undefined') {module.exports = config;}
```

**Additional Notes**: This could break code if the developers iterate above the `MM.getModules()` and brings the identifier from an `undefined` module.

khassel, 2020-12-28:
This is caused by [this line](https://github.com/MichMich/MagicMirror/blob/develop/js/main.js#L505) of `main.js`.

For testing I replaced `modules[module.data.index] = module;` with `modules[m] = module;` and this worked for me.

But:
- the modules array started with `1`, with the fix it starts with `0` (we could add 1 to m so it starts again with `1`)
- is there any relation needed between the index in the modules array and the module.data.index?

[image]

@ezeholz may you can test if this fix works on your side?

ezeholz, 2020-12-29:
@khassel That solution would totally solve the array problem. But there's a more deep problem, that I don't really know if it's designed that way. You can see also in your picture that the fifth module identifier it's missing. That means the MM it's detecting the module and, despite being disabled, it's being loaded and given a identifier.

I believe that disabled modules doesn't even have to be loaded. Maybe MM should detect when the module is disabled, and continue to the next one. 
But that's one step closer to the solution :D

sdetweil, 2020-12-29:
i think this is a side effect of how the config data is used..

it is a source file and as such loaded and rendered as a js object... ALL elements of the modules array are rendered.
at a later time this array is used to cause modules to be loaded, if they are not disabled. 

at this point the module array of the  config loaded thru require, should be reduced to non-disabled modules

the module search algorithm uses an enumerator, which MUST cover all the elements of the array, but 
returns true for loaded (not disabled) , but that slot in the determined array cannot be left out (using enumeration)

khassel, 2020-12-29:
I agree to @sdetweil  that its o.k. to load the config data into objects and then decide to use it or not (if disabled).

> That solution would totally solve the array problem.

But I'm not quite sure, if that solution works without any side effects.
The Coder of this line must have thought something about using the `module.data.index` as array index (?).

May @sdetweil and @MichMich should comment if this fix should be implemented or not or if there is a better solution.

sdetweil, 2020-12-29:
@khassel 
>The Coder of this line must have thought something about using the module.data.index as array index (?).

or didn't understand the side effect...
i think your fix is acceptable, does what i wanted a different way..

rejas, 2021-01-06:
Can this be closed now since the PR got merged @ezeholz ?

ezeholz, 2021-01-06:
I still thinking that this could be a deeper issue, because of module numeration and all that. I guess that the main problem that I brought was resolved with that PR.

Still, I was waiting until it's merged in master, so this could be close automatically. Haha

PR #2386 (2020-12-29): removes undefined objects from modules array, see issue #2382
closes #2382 

Fix removes undefined objects from modules array.
closed 2021-01-06 by MichMich