MagicMirrorOrg/MagicMirror #2389: Translator can't substitute empty strings

MikeBishop, 2020-12-30:
## I found a bug in MagicMirror

**Platform**: Raspberry Pi 4

**Node Version**: 

**MagicMirror Version**: 10.23.0

**Description**: In translator.js, the code checks whether a variable is present with the following code:

    return variables[varName] || "{" + varName + "}";

However, when the content of the variable is an empty string, this erroneously substitutes `{varName}` back in.    

**Steps to Reproduce**: Call `this.translate("something with {VAR}", {VAR: ""})

**Expected Results**: If the variable is present with the empty string, the empty string should be substituted.  This could be achieved by using `return varName in variables ? variables[varName] : "{" + varName + "}";` instead.

**Actual Results**: Variable is not subsituted

rejas, 2020-12-31:
Thanks for the report. Would you like to submit a PR for this issue?

MichMich, 2020-12-31:
Could you give an example when this would be usefull?

MikeBishop, 2020-12-31:
I initially used a substitution that contains "{HOURS}{LISTSEP}{MINUTES}", where each element is an empty string if that value is zero, and LISTSEP is `", "` if both are non-zero but `""` if only one is non-zero.  (If both are zero, we won't reach this code.)

There are other ways to build it, of course --  you can insert the string `" "` and let HTML collapse the whitespace, you can pick from multiple possible strings for the different possible combinations, etc.  But more generally, it's simply unexpected behavior that a variable that's present isn't substituted, no matter its value.

MichMich, 2020-12-31:
Makes sense. Feel free to send a PR.

rejas, 2021-01-06:
This issue can be closed I guess?

PR #2393 (2020-12-31): Permit substituting a falsy value in translator variables
Fixes #2389; this no longer evaluates the provided value, merely the presence of the key.  That means values which are falsy to JS (empty string, undefined, number 0, etc.) now get substituted where before they didn't.

For some possible values, it's possible that's not what you wanted -- but leaving {VARNAME} in place probably wasn't either, so 🤷‍♂️.  It fails differently now.  For empty strings, zeros, etc. this fixes a bug.
closed 2021-01-06 by MichMich