MagicMirrorOrg/MagicMirror #2391: Modules without a remote master branch trigger "[2020-12-31 14:43:37.337] [ERROR]  fatal: your current branch 'master' does not have any commits yet"

0m4r, 2020-12-31:
## I found a bug in MagicMirror
Any module without a "master" branch will give the fatal error
**[ERROR]  fatal: your current branch 'master' does not have any commits yet**

**Platform**: 
Raspberrypi 3b+

**Node Version**: 
v10.23.0

**MagicMirror Version**: 
2.13.0

**Description**:
Any module without a "master" branch will give the fatal error
**[ERROR]  fatal: your current branch 'master' does not have any commits yet**

**Steps to Reproduce**:
- Install a module without the master branch (i.e. mine: https://github.com/0m4r/MMM-covid19, it has a "main" branch as per the default github config for new repositories)
- start magic mirror
- check the server logs

**Expected Results**: 
No fatal error is shown. Main is used as a reference branch

**Actual Results**: 
A fatal error is shown

**Configuration**: 
Not relevant

**Additional Notes**: 
All seems to work well even with this error, and I know I could rename my main branch to master, but I think it should not be the case.

khassel, 2020-12-31:
This works for me on the `develop` branch, looking in [simple-git changelog](https://github.com/steveukx/git-js/blob/cd244a6c99b587ee605cb406c45c1df5dae56e3b/CHANGELOG.md) this was fixed with v2.23.0.

So the new release v2.14.0 of MagicMirror (coming tomorrow) should work with your module.

rejas, 2020-12-31:
So we can say it was fixed with PR https://github.com/MichMich/MagicMirror/pull/2222

sdetweil, 2020-12-31:
is this the same as the cause for #2383 ?

0m4r, 2020-12-31:
I have just switched to develop, run "npm i" and this error is gone, but now I do get another one:
```
[31.12.2020 15:51.58.377] [ERROR] (node:1763) UnhandledPromiseRejectionWarning: TypeError: Object.fromEntries is not a function
    at userOptions (/home/pi/MagicMirror/node_modules/simple-git/src/lib/tasks/log.js:32:19)
    at parseLogOptions (/home/pi/MagicMirror/node_modules/simple-git/src/lib/tasks/log.js:62:31)
    at Git.log (/home/pi/MagicMirror/node_modules/simple-git/src/git.js:871:26)
    at sg.git.fetch.status (/home/pi/MagicMirror/modules/default/updatenotification/node_helper.js:78:13)
    at onSuccess (/home/pi/MagicMirror/node_modules/simple-git/src/lib/task-callback.js:8:9)
    at process._tickCallback (internal/process/next_tick.js:68:7)
[31.12.2020 15:51.58.378] [ERROR] (node:1763) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
```
simple git is 2.26.0
```
pi@raspberrypi3:~/MagicMirror $ npm ls simple-git
magicmirror@2.14.0-develop /home/pi/MagicMirror
└── simple-git@2.26.0 
```

sdetweil, 2020-12-31:
@0m4r yeh, reported in #2383 
I can't reproduce it 

i tested with simple-git@2.31 and didn't see it, but can't cause it again on 2.26

if u can repeat, install simple-git@2.31 and try again

0m4r, 2020-12-31:
as per #2383, upgrading simple-git to 2.31.0 solves the issue.
As far as I am concerned this one can be closed. thanks for the quick support!
Let me know if I can be of any help!

khassel, 2020-12-31:
at the moment simple-git v2.26.0 is in `package.json`. So should this be updatet to v2.31.0 before the release tomorrow?

0m4r, 2020-12-31:
I guess that would be a good idea. But I do not know what it takes in terms of effort to update it.
I think you may want/need to go through all your QA process and stuff??

sdetweil, 2020-12-31:
just submitted pr

0m4r, 2021-01-05:
the problem is gone with the new release.
I close the issue.

closed 2021-01-05 by 0m4r