MagicMirrorOrg/MagicMirror #2394: add text for upgrade to switch back to master

sdetweil, 2020-12-31:
we have been advising users  to use the develop branch for issues that we have fixes for...

now git pull, will not upgrade them to 2.14

SO, we need to add a couple sentences
```
If you are using the develop branch, please switch back to master
git checkout master
```
then 
```
git pull
npm install
```

MichMich, 2020-12-31:
If you pull on the dev branch they will have the latest as well including some new dev commits when available. Leaving them on the dev branch seems better since that way we have more beta testers. 

When can add info in the documentation about the differences and procedures. But I don't think this is a mandatory part of the upgrade process.

sdetweil, 2020-12-31:
hm.. my upgrade script will put them back on master..

MichMich, 2020-12-31:
That's ok.

sdetweil, 2020-12-31:
ok, closing

closed 2020-12-31 by sdetweil