MagicMirrorOrg/MagicMirror #2399: Blank Screen after Reboot

Anish2291, 2021-01-01:
## I'm not sure if this is a bug

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)

## I'm having troubles installing or configuring MagicMirror

Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

## I found a bug in the MagicMirror installer

If you are facing an issue or found a bug while trying to install MagicMirror via the installer please report it in the respective GitHub repository:
[https://github.com/sdetweil/MagicMirror_scripts](https://github.com/sdetweil/MagicMirror_scripts)

## I found a bug in the MagicMirror Docker image

If you are facing an issue or found a bug while running MagicMirror inside a Docker container please create an issue in the GitHub repository of the MagicMirror Docker image:
[https://github.com/bastilimbach/docker-MagicMirror](https://github.com/bastilimbach/docker-MagicMirror)

---

## I found a bug in MagicMirror

Please make sure to only submit reproducible issues. You can safely remove everything above the dividing line.
When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version _and_ your hardware (Raspberry Pi 2/3, Windows, Mac, Linux, System V UNIX).

**Node Version**: Make sure it's version 8 or later.

**MagicMirror Version**: Please let us now which version of MagicMirror you are running. It can be found in the `package.log` file.

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce**: List the step by step process to reproduce the issue.

**Expected Results**: Describe what you expected to see.

**Actual Results**: Describe what you actually saw.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

rejas, 2021-01-01:
Hi @Anish2291 could you please provide more information? Please read the issue tempalte before posting an issue....

Anish2291, 2021-01-01:
> magicmirror@2.13.0 start /home/pi/MagicMirror
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

[2021-01-01 21:22:56.446] [LOG]    Starting MagicMirror: v2.13.0
[2021-01-01 21:22:56.454] [LOG]    Loading config ...
[2021-01-01 21:22:56.458] [ERROR]  WARNING! Could not validate config file. Starting with default configuration. Please correct syntax errors at or above this line: /home/pi/MagicMirror/config/config.js:76
				roundTemp: true,
				^^^^^^^^^

SyntaxError: Unexpected identifier
    at Module._compile (internal/modules/cjs/loader.js:722:23)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:798:10)
    at Module.load (internal/modules/cjs/loader.js:645:32)
    at Function.Module._load (internal/modules/cjs/loader.js:560:12)
    at Module.require (internal/modules/cjs/loader.js:685:19)
    at require (internal/modules/cjs/helpers.js:16:16)
    at loadConfig (/home/pi/MagicMirror/js/app.js:70:12)
    at App.start (/home/pi/MagicMirror/js/app.js:217:3)
    at Object.<anonymous> (/home/pi/MagicMirror/js/electron.js:129:7)
    at Module._compile (internal/modules/cjs/loader.js:786:30)
[2021-01-01 21:22:56.459] [LOG]    Loading module helpers ...
[2021-01-01 21:22:56.504] [LOG]    Initializing new module helper ...
[2021-01-01 21:22:56.505] [LOG]    Module helper loaded: updatenotification
[2021-01-01 21:22:56.506] [LOG]    No helper found for module: helloworld.
[2021-01-01 21:22:56.507] [LOG]    All module helpers loaded.
[2021-01-01 21:22:56.639] [LOG]    Starting server on port 8080 ... 
[2021-01-01 21:22:56.653] [LOG]    Server started ...
[2021-01-01 21:22:56.654] [LOG]    Connecting socket for: updatenotification
[2021-01-01 21:22:56.655] [LOG]    Sockets connected & modules started ...
[2021-01-01 21:22:56.763] [ERROR]  Whoops! There was an uncaught exception...
[2021-01-01 21:22:56.764] [ERROR]  Error: listen EADDRINUSE: address already in use 127.0.0.1:8080
    at Server.setupListenHandle [as _listen2] (net.js:1226:14)
    at listenInCluster (net.js:1274:12)
    at GetAddrInfoReqWrap.doListen [as callback] (net.js:1413:7)
    at GetAddrInfoReqWrap.onlookup [as oncomplete] (dns.js:65:10) {
  code: 'EADDRINUSE',
  errno: 'EADDRINUSE',
  syscall: 'listen',
  address: '127.0.0.1',
  port: 8080
}
[2021-01-01 21:22:56.771] [ERROR]  MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
[2021-01-01 21:22:56.772] [ERROR]  If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
[2021-01-01 21:22:56.775] [LOG]    Launching application.

Above is the error

Anish2291, 2021-01-01:
@rejas can you help me?

rejas, 2021-01-01:
> [2021-01-01 21:22:56.458] [ERROR] WARNING! Could not validate config file. Starting with default configuration. Please correct syntax errors at or above this line: /home/pi/MagicMirror/config/config.js:76
> roundTemp: true,
> ^^^^^^^^^

There is your error: Your config file is not valid. Please make sure it is valid javascript. You can check it directly for errors by running `npm run config:check` in your MagicMirror directory.

Anish2291, 2021-01-01:
I used that still it appears blank and give a syntax error on the below line. i cnt understand whats wrong
if (typeof module !== "undefined") {module.exports = config;}

MichMich, 2021-01-01:
Please refer to http://forum.magicmirror.builders for support and troubleshooting. The GitHub issues are ment for core bug reports. Not for support.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2021-01-01 by MichMich