MagicMirrorOrg/MagicMirror #2402: Calendar module is spamming in debug mode

shevett, 2021-01-03:
## I found a bug in MagicMirror

**Platform**: Linux

**Node Version**: 
```
pi@raspberrypi:~/MagicMirror/config $ node --version
v10.23.0
```

**MagicMirror Version**: Please let us now which version of MagicMirror you are running. It can be found in the `package.log` file.

This file does not exist:

```
pi@raspberrypi:~/MagicMirror $ find . -name package.log
pi@raspberrypi:~/MagicMirror $
```

Here's git log:
```
commit 10dc315f3b96ebb5e980e47a21011d44bb9da979
Merge: 3dbe8bf 0f1457b
Author: Michael Teeuw <michael@xonaymedia.nl>
Date:   Fri Jan 1 19:16:03 2021 +0100

    Merge pull request #2400 from MichMich/develop

    Release 2.14.0
```

**Description**: Just did a git pull to the latest version, did an npm install, and restarted.  I've been having problems with the Calendar module not respecting the 'maximumEntries' value (it still doesn't), and noticed that the nodeserver.log was filling up with gobs of debugging information - it appears to be going back into the entire history of the calendar and printing outeach line:

```
[03.01.2021 16:52.43.685] [DEBUG]  start=Sat Nov 12 2011 18:00:00 GMT-0500 (Eastern Standard Time) end=Sat Nov 12 2011 21:00:00 GMT-0500 (Eastern Standard Time)
[03.01.2021 16:52.43.687] [DEBUG] have entries
[03.01.2021 16:52.43.689] [DEBUG]
```

**Steps to Reproduce**: It just happens.

**Expected Results**: Calendar entries, no debug information

**Actual Results**: No calendar entries (they say 'no upcoming events') and tons of debug info.

**Configuration**: Here's the relevant block:
```
                {
                        module: "calendar",
                        header: "Dave's Personal Calendar",
                        position: "top_left",
                        config: {
                                calendars: [
                                        {
                                                maximumEntries: 5,
                                                symbol: "calendar-check",
                                                url: "https://calendar.google.com/calendar/ical/REDACTED/basic.ics"
                                        }
                                ]
                        }
                },
```

**Additional Notes**: A previous run here had the maximumEntries as a text field (ie, in quotes), I thought that may have been the problem.  I changed it to a numeric  - now I don't see any entries.  Previously, I was seeing 10.   Just to check, I just put it back to this, and I'm still seeing gobs of debug messages (and seeing no calendar entries...

```
                                                maximumEntries: "5",
```

khassel, 2021-01-03:
you can set logLevel in your config, see https://github.com/MichMich/MagicMirror/blob/master/config/config.js.sample#L31

If no logLevel is set, all entries including DEBUG are displayed.

shevett, 2021-01-03:
That seems a weird default, but you're right, adding hte logLevel block to the config turned off the debug messages (when I left it ou of the array).

Naturally, now the display is showing only 5 entries.  So I guess this one is fixed?   :) :)

rejas, 2021-01-04:
Thanks for the report @shevett, I just created a PR to set the responsible log level default values.

rejas, 2021-01-05:
The PR got merged, so this issue can be closed I guess @shevett

shevett, 2021-01-05:
WFM!