MagicMirrorOrg/MagicMirror #2403: 2.14 regression, not installed module produces no error

sdetweil, 2021-01-03:
I was helping user with MMM-WiFiPassword, it just displays a qr code

he had misspelled the name

to recreate
add this to your config.js 
```
        {
            module: "MMM-WiFiPassword1",
            position: "center",
              config: {
              //See 'Configuration options' for more information.
              network: "xxxxx",
              password: "xxxxxx",
	            showAuthType: false,
	            showNetwork: false,
	            showPassword: false,
	            debug: false
        		}
        }
```
i got  
No helper found for module: MMM-WiFiPassword1.

but not failure, because the module was missing

rejas, 2021-01-04:
What was the behaviour pre-2.14 for this case?

sdetweil, 2021-01-04:
magicMirror could not validate the config file
will not stop.....
(i think) have to go test

MichMich, 2021-01-04:
I think this behavior did not change.

sdetweil, 2021-01-04:
does not fail on 2.11...
should fail

rejas, 2021-03-04:
Tried as far as v2.7.0 (below wouldnt start anymore on my Ubuntu). No warnings or error besides the " No helper found for module: MMM-WiFiPassword1. "

So not sure when / or if it was changed. But maybe something worth discussing: Shouldnt there be a more obvious warnin/error log that you have a config entry for a module that isnt there?

sdetweil, 2021-03-23:
@rejas > Shouldnt there be a more obvious warnin/error log that you have a config entry for a module that isnt there?

thats my opinion...   users coming from windows make mistakes on letter casing and nothing works.. but they don't know why..

rejas, 2022-10-30:
Re-opening since this might be useful feature (just ran into it myself)

khassel, 2023-01-17:
This is the part of `app.js`

```js
		let loadHelper = true;
		try {
			fs.accessSync(helperPath, fs.R_OK);
		} catch (e) {
			loadHelper = false;
			Log.log(`No helper found for module: ${moduleName}.`);
		}
```

It is correct to do only a `log` here because not every module has a `node_helper.js`.

We could test before this part e.g. if the directory exists (and may its not empty) and otherwise do a `Log.error()`.

Would this solve this issue?
Other ideas?

sdetweil, 2023-01-17:
directory and modulename js existing. not node_helper as optional

khassel, 2023-01-17:
would be this:

```js
	function loadModule(module, callback) {
		const elements = module.split("/");
		const moduleName = elements[elements.length - 1];
		let moduleFolder = `${__dirname}/../modules/${module}`;

		if (defaultModules.includes(moduleName)) {
			moduleFolder = `${__dirname}/../modules/default/${module}`;
		}
// begin new
		const moduleFile = `${moduleFolder}/${module}.js`;

		try {
			fs.accessSync(moduleFile, fs.R_OK);
		} catch (e) {
			Log.error(`No ${moduleFile} found for module: ${moduleName}.`);
		}
// end new

		const helperPath = `${moduleFolder}/node_helper.js`;

		let loadHelper = true;
		try {
			fs.accessSync(helperPath, fs.R_OK);
		} catch (e) {
			loadHelper = false;
			Log.log(`No helper found for module: ${moduleName}.`);
		}
...
```

sdetweil, 2023-01-17:
I guess. They should be able to resolve no file if the folder is missing too.  altho many times it's a case mismatch

some have no idea that all these things must match.

PR #3015 (2023-01-26): added error message if `<modulename>.js` file is missing …
… in module folder to get a hint in the logs

fixes #2403
commit 157e74ce (2023-01-26): added error message if `<modulename>.js` file is missing … (#3015)

… in module folder to get a hint in the logs

fixes #2403
closed 2023-04-04 by khassel