MagicMirrorOrg/MagicMirror #2408: Improve test suite by starting electron only when needed

rejas, 2021-01-06:
Speaking of: the test suite could be so much improved. Currently for every test we start electron, while in most cases we could just "virtually" start the module and test what getDom returns.

_Originally posted by @MichMich in https://github.com/MichMich/MagicMirror/issues/2401#issuecomment-753959303_

MichMich, 2021-01-06:
And while were at it. It might be worth rethinking the test suite we are using. I'm not a specialist on (javascript) testing. But the output of our current test is rather underwhelming. I like the output of [Jest](https://jestjs.io). But maybe there is a better alternative.

fewieden, 2021-01-06:
I had to work with multiple test frameworks for javascript in my company such as mocha, ava, jest, ...

My favorite test framework of all (the ones I used) is definitely Jest.

I used it for the last few days to test one of my modules from 0 to 100% coverage [MMM-soccer](https://github.com/fewieden/MMM-soccer). Since this worked quite nicely I intend to apply this also to my other modules.

The tests are mainly unit tests with a few integration tests. I didn't create any e2e tests like the ones we have for the default modules with electron. Instead, I created simple mocks for NodeHelper and Module, to avoid cloning magic mirror in the CI like in this [example](https://github.com/ianperrin/MMM-Strava/blob/ef0855979c0870c01511e8041bbcd41ece66e101/.travis.yml#L8-L11).

My suggestion would be to cover most logic as unit and integration test, which doesn't require to spin up electron, but using mocks where needed. In addition, electron can be used for specific tests like rendering or related to electron itself.

Maybe @roramirez can add some thoughts to it, as I think he created the initial setup.

MichMich, 2021-01-06:
This sounds so much better than what we currently have. It would also simplify building test, especially if we create some test tools users can use.

roramirez, 2021-01-13:
Hi guys!

The main idea to create the e2e was try the cover all most cases of the principal modules and find and get it  the bug in configuration and start of the app in Electron. At in this time (4 year ago) the tests for new feature/bugfix are 0 and the way most easily to cover from the module to the display in the html render for Electron was build a e2e test. 

The unit test are good but we try get and test what the user can see in the Mirror. The problem here the issue is for run a heavy load and many tests in the e2e in CI. Issue like dependencies between run/start/stop of app in middle of every test case have a lot of cost of resources and sometime the stop or start not run the best way.

Maybe the use of Jest can help to build new tests or replaces the current ones. But the replace the all tests suite can be a huge battle of work

khassel, 2021-04-27:
I did some tests with electron v12 and spectron v14 and 27 e2e tests are failing ...

Spectron is more or less unmaintained, see [this post](https://github.com/electron-userland/spectron/issues/896).

So I think it makes no sense to investigate in this. We should stay on electron v11 until we have a new test setup for e2e with Jest or whatever.

rejas, 2021-04-28:
Given the lack of responses to that Spectron post, switching to another test setup seems like a good idea in the long run. Have you had any experience with jest or whatever yet @khassel ?

fewieden, 2021-05-01:
@rejas jest is more of a replacement for mocha, since it is a testing framework.

The question is if spectron/electron is really necessary for the e2e tests, where we check the rendered results. We probably can also do snapshot testing of the result from the nunjuck rendering https://github.com/MichMich/MagicMirror/blob/491f5aa7769a12de7f20f201de0078704c158a8d/js/module.js#L84-L110 then we only need to mock the document createElement function

khassel, 2021-09-26:
Meanwhile I managed to move more tests using `electron` to use `server only`. Will provide a PR after https://github.com/MichMich/MagicMirror/pull/2664 is merged because its based on this (see [this branch](https://github.com/khassel/MagicMirror/tree/new-e2e)).

Because this new tests are using `jsdom` and therefore running in the browser all test configs used must be changed because `require` cannot be used anymore.

There are 2 module tests still using electron (calendar and weather). They are complex (using basic-auth or mock-data), will look into this in a following PR.

The other 2 remaining electron tests `env_spec` and `dev_console` could be removed without a replacement (there is a new `env_spec` under `e2e` with most functionality).

The question is, if we want still some electron tests or if we want to remove the spectron stuff completely.

rejas, 2021-09-27:
Great to hear about your progress, good job!

> The question is, if we want still some electron tests or if we want to remove the spectron stuff completely.

Would that mean we cannot test if MM runs in Electron without problems?

khassel, 2021-09-27:
> Would that mean we cannot test if MM runs in Electron without problems?

if we decide to have no electron tests, yes.

rejas, 2021-10-16:
Making good progress, awesome!

> > Would that mean we cannot test if MM runs in Electron without problems?
> 
> if we decide to have no electron tests, yes.

I guess basic electron fucntionality like starting and such should be covered by tests. What dou you think @MichMich ?

MichMich, 2021-10-16:
Yeah I think the aim is to test everything that's possible without the need of booting electron. And then have one suite of tests covering basic electron functionality.

khassel, 2021-11-26:
Spectron is dead, [here](https://github.com/electron-userland/spectron/issues/1045) is the deprecation announcment.

Did a first test with [playwright](https://github.com/Microsoft/playwright) which was mentioned in the spectron issue, this works already:

```js
const { _electron: electron } = require('playwright');

let electronApp;

describe("Electron app environment", function () {

        beforeAll(function () {
                // Set config sample for use in test
                process.env.MM_CONFIG_FILE = "tests/configs/env.js";
        });

        beforeEach(async function () {
                electronApp = await electron.launch({ args: ['js/electron.js'] });
        });

        afterEach(async function () {
                await electronApp.close();
        });

        it("should open a browserwindow", async function () {
                // Get the first window that the app opens, wait if necessary.
                const window = await electronApp.firstWindow();
                // Test the title.
                expect(await window.title()).toBe("MagicMirror²");
        });
});
```

So i will work on this to replace the remaining spectron tests with playwright.

MichMich, 2021-11-26:
Champ!

khassel, 2021-11-28:
From my side this can be closed.

rejas, 2022-01-14:
Should this be closed now that https://github.com/MichMich/MagicMirror/pull/2784 is merged?

rejas, 2022-01-14:
Oh, I opened it, so I guess I can ask myself :-) My verdict is yes :-)

closed 2022-01-14 by rejas

DanielHabenicht, 2022-02-03:
Hi @fewieden, 
> I created simple mocks for NodeHelper and Module, to avoid cloning magic mirror in the CI like in this [example](https://github.com/ianperrin/MMM-Strava/blob/ef0855979c0870c01511e8041bbcd41ece66e101/.travis.yml#L8-L11).

Did you publish the mocks anywhere?
If not, I am happy to publish a NPM package for it, so it can be readily used in other packages. (or provide a PR so it can be done from inside this repo?) 
Because I kind of run into the same problem with my own package testing.

fewieden, 2022-02-05:
@DanielHabenicht 

> Did you publish the mocks anywhere? (except inside you repo)

They are only in my repo https://github.com/fewieden/MMM-soccer/tree/master/__mocks__

> If not, I am happy to publish a NPM package for it (or you can do it yourself), so it can be readily used in other packages. Or I provide a PR so it can be done from inside this repo?

I wouldn't add those mocks to the core project. Then you would end up with the same problem I tried to avoid: Cloning the project in the CI.

Those mocks currently only contain what I needed to test my module and only work if you use the jest test framework.

Feel free to publish them as a package.

> Because I run into the same problem with my own package testing.

Do you have a link to your work in progress? You can also connect to me via the discord channel.