MagicMirrorOrg/MagicMirror #2409: Calendar Module logs at DEBUG level by default

0m4r, 2021-01-07:
## I found a bug in MagicMirror

**Platform**: 
 Raspberry 3b+ (but not relevant)

**Node Version**: 
v10.23.0 (but not relevant)

**MagicMirror Version**: 
2.14.0"

**Description**: 
Calendar Module logs at debug level

**Steps to Reproduce**: 
Configure the calendar module

**Expected Results**: 
no DEBUG logs are reported by default

**Actual Results**: 
DEBUG logs are reported by default

**Configuration**: 
A simple calendar configuration

**Additional Notes**: 
N/A

MichMich, 2021-01-07:
Are you on the `master` branch or on the `develop` branch? If I'm not mistaken this is already fixed on develop.

To try the develop branch, use the following commands:
```
git checkout develop
git pull
npm install
```

0m4r, 2021-01-07:
I am on the master branch. I'll give it a try with the develop one asap!
thanks for the quick answer. 👍🏻 

It could be nice to add a config setting for the log level, it could be useful to debug!

0m4r, 2021-01-07:
ok, it works on develop, and, I am not sure if it is a bug on master, but on this branch I can also actually see the calendar events.

closed 2021-01-07 by 0m4r

khassel, 2021-01-07:
duplicate of #2402

> It could be nice to add a config setting for the log level, it could be useful to debug!

this exists already, see [config sample](https://github.com/MichMich/MagicMirror/blob/master/config/config.js.sample#L31)

0m4r, 2021-01-10:
thank @khassel I know that exists, I was more like a debug level specific for the module :)