MagicMirrorOrg/MagicMirror #2410: Replace the default weather modules.

MichMich, 2021-01-08:
A while ago we started the development of the [new weather module](https://docs.magicmirror.builders/modules/weather.html) which is aimed to be the replacement for the current `currentweather` and `weatherforcast` modules.

Currently the `currentweather` and `weatherforcast`modules are still the default modules used in the config sample. Since the new `weather` module is now pretty mature, it might be an idea to move over to the new module in de config sample and phase out the development of the old modules. 

Any thoughts on this transition?

fewieden, 2021-01-08:
There is a file called deprecated. I would add support for modules there, currently, it only supports config options.

At start, I would send an alert if the module is active and add a log.

One release later I would replace the module content with a simple div wrapper saying that the module is deprecated and the user should switch to the new module.

fewieden, 2021-01-08:
Also as soon as the modules are deprecated all issues should refer to the new module and only if the issue exists on the new module there should be a fix provided. Same for pull requests.

MichMich, 2021-01-08:
> One release later I would replace the module content with a simple div wrapper saying that the module is deprecated and the user should switch to the new module.

I think that is a bit to harsh. There are a lot of people that don't want to change their config and not all features of the current modules are implemented yet. But stop supporting the old modules might help in people moving over. 

> Also as soon as the modules are deprecated all issues should refer to the new module and only if the issue exists on the new module there should be a fix provided. Same for pull requests.

I think we can keep merging bug fixing PR's to the old modules for a while but prevent any implementation of new features. Any issue regarding the old modules should refer to the new modules.

rejas, 2021-01-08:
Would it be an option to mark the modules deprecated (somehow) and after x month / 1 year move them into a seperate repo like MMM-LegacyCurrentWeather / MMM-LegacyWeatherforecast (working title)

There they could receive updates without being in the main Repo...

MichMich, 2021-01-08:
Yeah, that sounds reasonable. Although we can always look for the best option later. I don't think that's something we have to decide now. 

The first step is moving over to the new module in the default config. This won't affect existing users. 

Is the new module mature enough to take this step?

fewieden, 2021-01-09:
> I think that is a bit to harsh. There are a lot of people that don't want to change their config

I'm not sure if this is a big issue. Those people will also not update the mirror as long as it runs without an issue.

But new people and the ones actively playing will switch over.

> The first step is moving over to the new module in the default config. This won't affect existing users.

I think just replacing it in the config sample is not enough as a first step, because this only affects new users, but they don't know about missing features from the old modules. At least we should also add deprecated logs (affecting users that have issues and check logs) and send an alert on boot (affecting users that are updating their system and have a reboot). Otherwise, we never detect the missing features or existing issues on the new module without using the crowd as QA's.

buxxi, 2021-01-10:
What features are missing in the new module and have they been skipped intentionally?
I didn't notice any difference when I switched except for one more decimal for precipitation (but I mostly use the features that disables stuff that I don't care about).

The only "issue" I see is that the new weather-module uses fetch-api and promises (which I'm all for) on the client side which wont work for people using old iPads as MagicMirror-screens, but they should probably keep running an older version of MagicMirror anyway...

I agree with @fewieden that a logging and an warning notification should be displayed.

Maybe a bit overkill, but could we maybe make some kind of proxy from the old modules to the new:
1. map configuration parameters from the currentweather and weatherforecast into what they would have been in the new weather-module
2. load the new weather module after the old module
3. unload the old modulee

Or maybe do some preprocessing on the config file?

rejas, 2021-01-10:
I just came across one feature that I think is missing:
The compliments relies on the CURRENTWEATHER_DATA notification to display the compliments for s specific weather. Only the currentweather module is sending it, not the new one. Or was ist renamed or something and compliments needs to be updated?

Aikagi000, 2021-01-13:
I just discovered this module in light of Apple's acquisition of Darksky. So far I like it and I was able to recreate most the the layout I had. 

Feature requests:
1) Is it possible to have the hourly forecast setup so that it can show the hourly in increments? So say instead of  1pm, 2pm, 3pm  it can be 1pm, 3pm, 5pm ( or any adjustable increment amount). This way people can get an overview of however many hours without an entire list of entries.

2) Option to adjust colors in config? (or maybe clearly labeled in css?)

3) Icon sets? (the overcast icon looks like a moon over clouds to me)

Just a few thoughts off the top of my head, but I'm liking the direction this is going so far. Thank you for this!

rejas, 2021-01-14:
> Feature requests:

Are those 3 requests already in the "old" module? Then maybe we should open an issue with "help wanted" for each one?

buxxi, 2021-01-16:
Don't think any of them are in the old modules and probably should be posted as separate feature requests, but here's my thoughts on it:
1. Seems like a nice addition. But I think the hourly part of the weather provider needs a bit refactoring first (depends on OpenWeatherMap-configuration instead of making it an optional implementation, "fetchWeatherData" is a bit too generic) so I skipped implementing that for SMHI even though the original data is hourly.
2. Use custom.css, it's perfect for this. Lookup the icons class name at https://erikflowers.github.io/weather-icons/ and set the color for it (there are examples for this is in the forum). Should maybe be a part of the documentation for the weather module?
3. The weather icons are a font and not separate images, supporting multiple icon sets would probably require mapping the icon-values returned from the providers to the icon-sets classes. If it's just one icon that bothers you could override it in the custom.css too or create an issue here if we're using the wrong icon for that provider.

rejas, 2021-01-16:
> The only "issue" I see is that the new weather-module uses fetch-api and promises (which I'm all for) on the client side which wont work for people using old iPads as MagicMirror-screens, but they should probably keep running an older version of MagicMirror anyway...

Those OS arent supported anyway anymore. The js files already contains lots of es6 notations that some iOS dont understand.

rejas, 2021-01-18:
> Maybe a bit overkill, but could we maybe make some kind of proxy from the old modules to the new:

Yeah, to me that sounds like overkill :-) I think the parameters are not so complex that a user cannot do the switch himself with some help from a README in the new module and a deprecation notice in the old one.

buxxi, 2021-01-18:
It wasn't for the complexity I made that suggestion. More of a way to force everyone that uses the old modules to actually try the new one so the old ones could be removed quicker. But I agree, it shouldn't be needed.

Don't really see why we shouldn't be able to get rid of it quickly like this:
- Deprecate the modules in the next release, with logging if used and a notification on the mirror and updated documentation about the deprecation and how to migrate
- On the second release the old modules could show the notification permanent and remove all other functionality of the module and remove them from the documentation
- On the third release just remove them completely from both the code and the documentation

Aikagi000, 2021-01-19:
@buxxi Hey thanks for your answers! 

I'm gonna check out the css for the weather icons as you mentioned. I do think css stuff should be documented somehow for this so folks can know what to look for when customizing the css. Like maybe a list of the classes and what it relates to on the weather module display itself . I guess even if it's sort of general at least it would be a starting point they can start searching from.

As far as the sun looking like a moon issue. I now see why that is showing as it did to me. I dunno if it's maybe a bug or if that's just how it's programed, but the day/night versions of the icons are not following the sunrise/sunset time. Where I am, the sun currently rises at say 7:20 , by 8:00 when I'm leaving it still has a moon icon. Same thing at night, by time I get home it still has a sun icon even thought the sun has well set already.  I'm using the default 10 min update time.

MichMich, 2021-01-21:
Sorry for my slow response.

It would be okay to display a deprecated message in the logs. And maybe add some info in the source files: `This module is deprecated. Any additional feature will no longer be merged.` But other then this, I think we shouldn't be spending too much time on this change (skip the proxy idea, and the notification). So my suggestions for now are:

- Update de example config.
- Add a deprecated message to the log.
- Add info in the source file(s).
- Update the documentation.

If anyone wants to send some PR's for this, that would be awesome. Otherwise I'll make sure to do this before the next release.

buxxi, 2021-01-23:
I created a pull request for the first three (https://github.com/MichMich/MagicMirror/pull/2433).

buxxi, 2021-01-26:
I can try to make a pull request for the documentation too this weekend with a deprecation message and migration example for each of the modules.

MichMich, 2021-01-26:
That would be great! Thanks!

buxxi, 2021-02-06:
Created pull request for it (and two others for the documentation) but I guess you actually merge it when releasing new version since the documentation doesn't have develop-branch.

MichMich, 2021-02-06:
Yes, whenever I release a new version I go thought the Doc PR's.

rejas, 2021-04-18:
This forum post probably needs to be updated to: https://forum.magicmirror.builders/topic/343/default-modules

closed 2021-06-28 by stale[bot]

rejas, 2021-06-28:
So, are we really done with the replacing @MichMich ?

MichMich, 2021-06-28:
It looks like the stale boy thinks so. 😅 I don't see any issues we need to finish. Right?