MagicMirrorOrg/MagicMirror #2415: Latest Calendar Module

g1gsw, 2021-01-13:
I seem to have found a bug in the latest calendar Module in MagicMirror 2.14 the date is always MM/DD/YYY and does not change if I put any other information config.js, calendar.js shows it should be "MMM Do", my config is as below. 2.13 was fine and respected the config.
{
                        module: "calendar",
                        header: "Calendar",
                        position: "top_left",
                        config: {
                        dateFormat: "DD/MM/YYYY",
                        fullDayEventDateFormat: "DD/MM/YYYY",
                                calendars: [
                                        {
                                                symbol: "calendar",
                                                url: "https://calendar.google.com/calendar/ical/xxxxx.xxxxxx%40gmail.com/private-c157182d5621b40789936a9da5ed521b/basic.ics",
                                }
                                ]
                        }
                },
Name removed and replaced with x's.

MichMich, 2021-01-13:
Have you tried the dev branch to see if that solves the issue?

g1gsw, 2021-01-13:
How do I do that on my raspberry pi?  
My install script is:-
#!/bin/sh
bash -c  "$(curl -sL https://raw.githubusercontent.com/sdetweil/MagicMirror_scripts/master/raspberry.sh)"

and my update one is:-

#!/bin/sh
bash -c  "$(curl -sL https://raw.githubusercontent.com/sdetweil/MagicMirror_scripts/master/upgrade-script.sh)" apply

g1gsw, 2021-01-13:
Just had another look at it and found if I set the "timeFormat" to absolute or dateheaders then the date is in the correct format.

MichMich, 2021-01-13:
So, can I close this issue?

g1gsw, 2021-01-13:
Yes many thanks.

Colin

closed 2021-01-13 by MichMich