MagicMirrorOrg/MagicMirror #2419: MagicMirror becomes unresponsive

swvalenti194, 2021-01-15:
Running 2.14.0 on Buster desktop and only a restart of pi fixes the issue.  Getting this in the logs...any ideas?

`0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.14.8
3 info using node@v10.23.0
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle magicmirror@2.14.0~prestart: magicmirror@2.14.0
6 info lifecycle magicmirror@2.14.0~start: magicmirror@2.14.0
7 verbose lifecycle magicmirror@2.14.0~start: unsafe-perm in lifecycle true
8 verbose lifecycle magicmirror@2.14.0~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/home/pi/MagicMirror/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
9 verbose lifecycle magicmirror@2.14.0~start: CWD: /home/pi/MagicMirror
10 silly lifecycle magicmirror@2.14.0~start: Args: [ '-c',
10 silly lifecycle   'DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js' ]
11 silly lifecycle magicmirror@2.14.0~start: Returned: code: 1  signal: null
12 info lifecycle magicmirror@2.14.0~start: Failed to exec start script
13 verbose stack Error: magicmirror@2.14.0 start: `DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
13 verbose stack     at EventEmitter.emit (events.js:198:13)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:198:13)
13 verbose stack     at maybeClose (internal/child_process.js:982:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
14 verbose pkgid magicmirror@2.14.0
15 verbose cwd /home/pi/MagicMirror
16 verbose Linux 5.4.79-v7+
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v10.23.0
19 verbose npm  v6.14.8
20 error code ELIFECYCLE
21 error errno 1
22 error magicmirror@2.14.0 start: `DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js`
22 error Exit status 1
23 error Failed at the magicmirror@2.14.0 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]`

swvalenti194, 2021-01-15:
That looks crappy I attached the logs here
[2021-01-15T12_27_39_323Z-debug.log](https://github.com/MichMich/MagicMirror/files/5820532/2021-01-15T12_27_39_323Z-debug.log)

khassel, 2021-01-15:
AFAIS there is no real error message in the logs.

This can be caused by a module, so disabling one by one until it works is one approach.

ManuZz80, 2021-01-17:
Same problem. Nothing useful in the Logs. I know 2 people with the same problem. Crown reboots are the solution atm. Happens since the last update.

swvalenti194, 2021-01-17:
Glad I am not the only one...

rejas, 2021-01-17:
To maybe find out a little more could you guys and girls post the following:
- platform you are running MM on
- modules that are enabled 
?

ManuZz80, 2021-01-17:
Running on a pi3b+ and sd card
changed pi and sd card --> no improvement

After updating to the latest version of MM - updated modules at the same time were MMM-MotionDetector and MMM-PublicTransportBerlin

Running modules: 
default
MMM-AirQuality
MMM-bitcoin
MMM-MoonPhase
MMM-MotionDetector
MMM-PublicTransportBerlin

MichMich, 2021-01-17:
It would be super helpful if you could run it with default modules only to see if it is caused by the core or 3rd party modules.

ManuZz80, 2021-01-17:
I will, atm I have the previous version with all modules (up to date) running to see if this goes well. 
Then I will update it and throw all external modules out.

MichMich, 2021-01-21:
Any news on this?

ManuZz80, 2021-01-21:
Yes and no. I freshly installed the newest release and startet only with the default modules. All fine. Then I added one after another Module from the third party modules. All fine. The mirror respond as he should. No crashes no nothing. 

My guess is now that the error happens only with a updated core, maybe there are files which were not correctly overwritten/merged. 
I backed up all third party modules and reused them with the fresh install. So they are in the same state when I had the error. 

Solution for everyone with this bug, backup your config, delete your mm folder, download the files new, run npm install. Put your config back and enjoy!

MichMich, 2021-01-21:
Good to hear. Can I close this issue?

ManuZz80, 2021-01-21:
I would say yes. 
thx :)

closed 2021-01-21 by MichMich