MagicMirrorOrg/MagicMirror #2424: Calendar module enhancement: displaying full day events without time

uncleLem, 2021-01-16:
With relative timeformat, upcoming full day events are displayed like this: 

> Wednesday at 12:00 AM

which doesn't make much sense to me. I've tried to look into the documentation and the code, and it seems like there's no way to configure it at the moment. 

`calendar()` method from moments.js used in the relative time display can't format without time by default, but does allow using custom calendar formatters. They are a rather tricky, because due to different localizations one size won't fit all, but I believe I've nailed it by putting adapted locale-specific calendars into translation files.

Would you consider adding it to the main repository? The details of the change can be found [here](https://github.com/MichMich/MagicMirror/commit/5903adadf9af64b00fe9f21bebec98e58e773476). I've updated translation files for the languages I know, but others require at least minimal understanding of grammar, so I don't think I can't do it myself. Also, for configuring the calendar from translation files, I'm using coreTranslations directly because in this case I don't want the fallback to English version, but rather to default localized calendar from moments.js.

I am also thinking about using `dateFormat` from config to replace the default `"sameElse": "L"`, allowing customization in relative format too, what do you think of it?

MichMich, 2021-01-21:
If it's not too much work, would you mind creating a PR so we can see your proposed change?

uncleLem, 2021-01-23:
Created pull request #2434 

Sorry for the failed checks, I can't figure out what's wrong at the moment and I hope I'll be able to check them a bit later this weekend, but I hope that would be enough for now to explain the idea?

Alvinger, 2021-01-25:
@uncleLem you can use the config option **fullDayEventDateFormat** to set the format for full day events. That will allow you to skip the time part.

uncleLem, 2021-01-25:
@Alvinger `fullDayEventDateFormat` is used in absolute time format, and it's just a date format, while "calendars" used by moment.js in relative time format are more complex. So, even though this could be a workaround, it's not exactly what I'm going for, I personally find relative time format more appealing and I believe this aspect can be enhanced.

Alvinger, 2021-01-25:
@uncleLem I agree that there is always room for improvement. I also noticed that moment.js calendar() now has a config option for various formatting options. However, as you say, the options are tricky to implement due to localization. Also, it wouldn't solve your issue as full day events still would be displayed with time.

The way it currently works is "by design" to avoid issues with backward compatibility.

What I could do in the short term is to let "fullDayEventDateFormat" apply to "relative" events as well. How's that?

Alvinger, 2021-01-26:
@uncleLem I just had a thought:

What if we introduced a new option, let's call it `momentCalendarFormat`, that can contain the formatting options for moment.calendar()? Then we wouldn't have to worry about translation as each user can implement their own language directly in the config.

It is also relatively easy to implement.

uncleLem, 2021-01-26:
> Also, it wouldn't solve your issue as full day events still would be displayed with time.

@Alvinger Sorry, I am not sure what you mean by that. There are some screenshots in the PR, showing no awkward time markers in full day events in what I'm proposing. Are we talking about the same thing?

> What I could do in the short term is to let "fullDayEventDateFormat" apply to "relative" events as well. How's that?

If I understand correctly, that would effectively turn relative time format into the absolute one, wouldn't it? I was thinking about further customization of the calendar's `sameElse` parameter with `dateFormat` or `fullDayEventDateFormat`, but I'd like to settle on specifics of how calendars should be stored before doing that. There's some discussion about implementation's specifics starting in comments to the PR, you are welcome to join if you want.

> What if we introduced a new option, let's call it momentCalendarFormat, that can contain the formatting options for moment.calendar()? Then we wouldn't have to worry about translation as each user can implement their own language directly in the config.

This indeed would be easier to implement, but not easier to use, because some languages require at least basic JavaScript knowledge to implement a functioning calendar. I'm not very familiar with user base of this project, is this something to be expected? And would it even be worth implementing? I believe a person with enough JS knowledge to use moment.js' calendar with functions might as well take a step further and inject it directly in their copy of calendar.js like I did it in my first PoC/draft version of this.

Alvinger, 2021-01-27:
@uncleLem I do understand and to some extent agree with you.

But there are considerations to be made: some users might actually *want* the start and end time for a full day event and some users might *want* to know when a full day event finishes. How would they be able to get that with your proposed changes? The calendar module is widely used and backward compatibility is important. One could change default behavior but a user should be able to "restore" behavior by setting options.

uncleLem, 2021-01-27:
> The calendar module is widely used and backward compatibility is important. One could change default behavior but a user should be able to "restore" behavior by setting options.

@Alvinger calendar.js:382 - in current implementation it falls back to default `.calendar()` method if there's no custom calendar available. I can add a config parameter to enforce it.

khassel, 2022-10-03:
seems to be fixed with the above PR's, so should be closed @uncleLem @MichMich

PR #2737 (2021-12-21): Resubmit of "Relative full day events"
Took #2704 and merged develop into it to see if the tests run through now.

Would fix #2424
PR #2704 (2021-12-21): Relative full day events
Fixes #2424
- Don't display start time of full date events when using relative dates.
- Use the fullDateEventDateFormat when showing full day events more than a week out
closed 2022-10-04 by MichMich