MagicMirrorOrg/MagicMirror #2425: mess calendar module

hangorazvan, 2021-01-18:
you higgledy-piggledy the calendar again!

timeFormat: "relative",
[image]

timeFormat: "absolute",
[image]

rejas, 2021-01-18:
Could you elaborate on what is messed up?

khassel, 2021-01-18:
looks like relative and absolute are swapped

hangorazvan, 2021-01-19:
> looks like relative and absolute are swapped

exactly

MichMich, 2021-01-21:
@hangorazvan Could you send a PR to fix this?

hangorazvan, 2021-01-24:
> @hangorazvan Could you send a PR to fix this?

No, I don’t have fork for PR, I use my own repository

rejas, 2021-01-24:
> > @hangorazvan Could you send a PR to fix this?
> 
> No, I don’t have fork for PR, I use my own custom repository

Did you fix it in there with a commit which we could copy/paste maybe?

hangorazvan, 2021-01-24:
No, for now I use "absolute" for "relative" and vice versa

rejas, 2021-01-25:
I see. Maybe @alvinger can help us out, since he seemed to have touched / refactored that code in November?

Alvinger, 2021-01-25:
@hangorazvan Could you please provide me with the complete configuration for the calendar in question? I do use both absolute and relative extensively without any of the issues you describe.
Once I have the configuration I will take a look!

hangorazvan, 2021-01-25:
{
			module: "calendar",
			position: "top_left",
			header: "Calendar evenimente și aniversări",
			disabled: false,
			config: {
				maximumEntries: 16,
				maximumNumberOfDays: 365,
				limitDays: 0,
				displaySymbol: true,
				defaultSymbol: "calendar",
				showLocation: false,
				displayRepeatingCountTitle: true,
				defaultRepeatingCountTitle: "",
				maxTitleLength: 25,
				maxLocationTitleLength: 25,
				wrapEvents: false,
				wrapLocationEvents: false,
				maxTitleLines: 3,
				maxEventTitleLines: 3,
				fetchInterval: 60 * 1000,
				fade: true,
				fadePoint: 0.25,
				colored: false,
				coloredSymbolOnly: true,
				urgency: 10,
				timeFormat: "absolute", 		// for the moment to show relative
				dateFormat: "MMM Do",				
				dateEndFormat: "LT",
				fullDayEventDateFormat: "ddd D MMM",
				showEnd: false,
				getRelative: 24,
				hidePrivate: false,
				hideOngoing: false,
				tableClass: "qsmall",
				customEvents: [],
				titleReplace: {},
				locationTitleReplace: {},
				broadcastEvents: true,
				excludedEvents: [],
				sliceMultiDayEvents: false,
				broadcastPastEvents: false,
				nextDaysRelative: true,

				calendars: [
					{
						symbol: "calendar-check-o", symbolClass: "skyblue", titleClass: "skyblue", timeClass: "skyblue", color: "normal",
						url: "https://calendar.google.com/calendar/ical/ro.romanian%23holiday%40group.v.calendar.google.com/public/basic.ics"
					},
					{
						symbol: "moon", symbolClass: "normal", titleClass: "normal", timeClass: "normal", color: "normal",
						url: "https://calendar.google.com/calendar/ical/ht3jlfaac5lfd6263ulfh4tql8%40group.calendar.google.com/public/basic.ics"
					},
					{
						symbol: "registered", symbolClass: "lightgreen", titleClass: "lightgreen", timeClass: "lightgreen", color: "normal",
						url: "https://calendar.google.com/calendar/ical/.../basic.ics"
					},
					{
						symbol: "product-hunt", symbolClass: "tomato", titleClass: "tomato", timeClass: "tomato", color: "normal",
						url: "https://calendar.google.com/calendar/ical/.../basic.ics"
					},
					{
						symbol: "birthday-cake", symbolClass: "gold", titleClass: "gold", timeClass: "gold", color: "normal",
						url: "https://calendar.google.com/calendar/ical/.../basic.ics"
					},
					{
						symbol: "film", symbolClass: "magenta", titleClass: "magenta", timeClass: "magenta", color: "normal",
						url: "https://calendar.google.com/calendar/ical/.../basic.ics"
					},
					{
						symbol: "suitcase", symbolClass: "coral", titleClass: "coral",timeClass: "coral", color: "normal",
						url: "https://calendar.google.com/calendar/ical/.../basic.ics"
					},
				]
			}
		},

hangorazvan, 2021-01-27:
@Alvinger Sorry for my frustration but you don't understand, I don't care and I don't want to change "urgency" or "nextDayRelative". I need relative dates, not absolute. Until now I have same settings and display correctly, after update the relative and absolute are switched.

sdetweil, 2021-01-27:
i can only say, i did not change the display side of calendar

look back thru the change log to find it

hangorazvan, 2021-01-27:
> i can only say, i did not change the display side of calendar
> 
> look back thru the change log to find it

It really doesn't matter, I presented the current situation

Alvinger, 2021-01-27:
@hangorazvan @sdetweil I think this discussion should be moved to the forums as it is not an issue but rather a feature request.

_My final words on the matter (although I'm happy to discuss in the forums):_

Even though @sdetweil and I aimed for backwards compatibility while fixing a lot of the bugs in the calendar module, especially relating to recurring events, it is entirely possible that some use cases will have been altered.  One display change I know happened is that previously even `absolute` dates became relative for the last two days. Now, that doesn't happen unless you set `nextDaysRelative` to true.

One can describe the three display modes as follows:
`dateheaders` group events into days and presents each day under one header.
`relative` could be described as `moment default` as it simply displays the date in the default format that moment.js uses.
`absolute` could be described as `custom` as it allows you to customize the display by setting options.

If you're happy with the way moment formats dates, choose `relative`.
If you want to have absolute dates only, choose `absolute`, set `urgency` and `getRelative` to 0 (zero) and `nextDaysRelative` to false.
If you want absolute dates to become relative at one point in time or another, set `urgency`, `getRelative`, and `nextDaysRelative` to appropriate values.

hangorazvan, 2021-01-27:
I open topic on forum
https://forum.magicmirror.builders/topic/14460/mess-calendar-module/1?_=1611742196667

hangorazvan, 2021-01-27:
unfortunately no one understood what I want to highlight although I also posted pictures, but instead I understood what you mean, so with absolute time format you can play for relative dates on urgency, but for all time absolute at line 366 in calendar.js instead
`timeWrapper.innerHTML = this.capFirst(moment(event.startDate, "x").calendar());`
must be
`timeWrapper.innerHTML = this.capFirst(moment(event.startDate, "x").fromNow());`
see the diference between first and third picture

closed 2021-01-27 by hangorazvan