MagicMirrorOrg/MagicMirror #2426: problem saying config folder not found

adamhrastich, 2021-01-18:
/* Magic Mirror Config Sample
 *
 * By Michael Teeuw https://michaelteeuw.nl
 * MIT Licensed.
 *
 * For more information on how you can configure this file
 * See https://github.com/MichMich/MagicMirror#configuration
 *
 */

var config = {
	address: "localhost", 	// Address to listen on, can be:
							// - "localhost", "127.0.0.1", "::1" to listen on loopback interface
							// - another specific IPv4/6 to listen on a specific interface
							// - "0.0.0.0", "::" to listen on any interface
							// Default, when address config is left out or empty, is "localhost"
	electronOptions:  {
		webPreferences: {
			webviewTag: true
		}
	},
	port: 8080,
	basePath: "/", 	// The URL path where MagicMirror is hosted. If you are using a Reverse proxy
					// you must set the sub path here. basePath must end with a /
	ipWhitelist: ["127.0.0.1", "::ffff:127.0.0.1", "::1"], 	// Set [] to allow all IP addresses
															// or add a specific IPv4 of 192.168.1.5 :
															// ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.1.5"],
															// or IPv4 range of 192.168.3.0 --> 192.168.3.15 use CIDR format :
															// ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.3.0/28"],

	useHttps: false, 		// Support HTTPS or not, default "false" will use HTTP
	httpsPrivateKey: "", 	// HTTPS private key path, only require when useHttps is true
	httpsCertificate: "", 	// HTTPS Certificate path, only require when useHttps is true

	language: "en",
	logLevel: ["INFO", "LOG", "WARN", "ERROR"], // Add "DEBUG" for even more logging
	timeFormat: 12,
	units: "imperial",
	// serverOnly:  true/false/"local" ,
	// local for armv6l processors, default
	//   starts serveronly and then starts chrome browser
	// false, default for all NON-armv6l devices
	// true, force serveronly mode, because you want to.. no UI on this device
	modules: [

		{
			module: "alert",
		},
		{
			module: "updatenotification",
			position: "top_bar"
		},
		{
			module: "clock",
			position: "top_left"
		},
		{
			module: "calendar",
			header: "calendar",
			position: "top_left",
			config: {
				calendars: [
					{
						symbol: "calendar-check",
						url: "https://calendar.google.com/calendar/ical/adamhrastich%40gmail.com/private-7d73f3437121432a5b14623cb97139ee/basic.ics"		}
				]
			}
		},
		{
			module: "currentweather",
			position: "top_right",
			config: {
				location: "Menomonie",
				locationID: "5262634", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
				appid: "c2647e77b86f90ed7922d3cb499204d0"
			}
		},
		{
			module: "weatherforecast",
			position: "top_bar",
			header: "Weather Forecast",
			config: {
				location: "Menomonie",
				locationID: "5262634", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
				appid: "c2647e77b86f9ed7922d3cb499204d0"
			}
		},
		{
			module: "newsfeed",
			position: "bottom_bar",
			config: {
				feeds: [
					{
						title: "New York Times",
						url: "https://rss.nytimes.com/services/xml/rss/nyt/HomePage.xml"
					}
				],
				showSourceTitle: true,
				showPublishDate: false,
				broadcastNewsFeeds: true,
				broadcastNewsUpdates: true
			}
		},
		{
  module: "MMM-GoogleAssistant",
  position: "fullscreen_above",
  config: {
    debug: true,
    assistantConfig: {
      lang: "en-US",
      projectId: "", // Required to use gaction.
      modelId: "", // (OPTIONAL for gaction)
      instanceId: "", // (OPTIONAL for gaction)
      latitude: 44.876480,
      longitude: -91.929703,
    },
    responseConfig: {
      useScreenOutput: true,
      screenOutputCSS: "screen_output.css",
      screenOutputTimer: 5000,
      screenRotate: false,
      activateDelay: 250,
      useAudioOutput: true,
      useChime: true,
      newChime: false,
      useNative: false,
      playProgram: "mpg321"
    },
    micConfig: { // put there configuration generated by auto-installer
      recorder: "arecord",
      device: "plughw:2",
    },
    snowboy: {
      usePMDL: false,
      audioGain: 2.0,
      Frontend: true,C
      Model: "jarvis",
      Sensitivity: null
    },
    A2DServer: {
      useA2D: true,
      stopCommand: "stop",
      useYouTube: true,
      youtubeCommand: "youtube",
      displayResponse: true
    },
    recipes: [ "with-MMM-TelegramBot.js" ],
    NPMCheck: {
      useChecker: true,
      delay: 10 * 60 * 1000,
      useAlert: true
    }
  }
},
{
  module: "MMM-Assistant2Display",
  position: "top_left",
  config: {
    debug:false,
    useYoutube: true,
    links: {
      useLinks: false,
      displayDelay: 60 * 1000,
      scrollStep: 25,
      scrollInterval: 1000,
      scrollStart: 5000,
      scrollActivate: false,
      verbose: false
    },
    photos: {
      usePhotos: true,
      displayDelay: 10 * 1000
    },
    volume: {
      useVolume: false,
      volumePreset: "ALSA",
      myScript: null
    },
    briefToday: {
      useBriefToday: true,
      welcome: "brief Today"
    },
    screen: {
      useScreen: false,
      delay: 5 * 60 * 1000,
      turnOffDisplay: true,
      mode: 1,
      ecoMode: true,
      displayCounter: true,
      displayBar: false,
      displayStyle: "Text",
      text: "Auto Turn Off Screen:",
      detectorSleeping: false,
      governorSleeping: false,
      displayLastPresence: true,
      LastPresenceText: "Last Presence:",
      delayed: 0
    },
    touch: {
      useTouch: false,
      mode: 3
    },
    pir: {
      usePir: false,
      gpio: 21,
      reverseValue: false
    },
    governor: {
      useGovernor: false,
      sleeping: "powersave",
      working: "ondemand"
    },
    internet: {
      useInternet: true,
      displayPing: false,
      delay: 2* 60 * 1000,
      scan: "google.fr",
      command: "pm2 restart 0",
      showAlert: true
    },
    cast: {
      useCast: true,
      castName: "MagicMirror_A2D",
      port: 8569
    },
    spotify: {
      useSpotify: false,
      useIntegred: false,
      useLibrespot: false,
      connectTo: null,
      playDelay: 3000,
      minVolume: 10,
      maxVolume: 100,
      updateInterval: 1000,
      idleInterval: 10000,
      username: "",
      password: "",
      CLIENT_ID: "",
      CLIENT_SECRET: "",
      deviceDisplay: "Listening on",
      usePause: true,
      typeArtist: "artist",
      typePlaylist: "playlist",
      typeAlbum: "album",
      typeTrack: "track"
    },
    NPMCheck: {
      useChecker: true,
      delay: 10 * 60 * 1000,
      useAlert: true
    }
  }
},
	]
};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}

rejas, 2021-01-19:
Hi. Could you give a little more information: Where does the error appear? What else does it report?

adamhrastich, 2021-01-19:
> Hi. Could you give a little more information: Where does the error appear? What else does it report?

When I do the start up it says the config is missing or the code is wrong

rejas, 2021-01-19:
Running config_check on your config file results in this output:

[19.01.2021 19:13.28.793] [ERROR] Your configuration file contains syntax errors :(
[19.01.2021 19:13.28.793] [ERROR] Line 138 column 6: Parsing error: Unexpected token Model

There seems to be a C too much here:

```
					Frontend: true,C
					Model: "jarvis",
```

@MichMich I didnt find any mention on the config:check script. Shouldnt this be added to the README / UI to help people?

MichMich, 2021-01-19:
Good idea! Feel free to send some PR's.

closed 2021-01-19 by adamhrastich

rejas, 2021-01-20:
PR is submitted :-)