MagicMirrorOrg/MagicMirror #2429: sendSocketNotification send twice from node_helper

goedh452, 2021-01-21:
Hardware: Raspberry Pi 4
Magic Mirror version: 2.14.0
Node version: 10.23.1

Problem: It seems that the node_helper processes a socket notification twice. This goes for all modules I installed that have a node_helper. I added some logging to the .js of the module itself and to the node_helper. It just sends a socket notification to the helper and the helper sends it back. While the socket notification to the helper is send once, the helper sends 2 socket notifications back.

.js of the module. The getData function is called by the scheduled interval every 10 seconds.
```javascript
getData: function(url) {
  self = this;
  var requestData = new XMLHttpRequest();
  requestData.onreadystatechange = function() {
    if (this.readyState === 4) {
      if (this.status === 200) {
        console.log("Send notification");
        self.sendSocketNotification("MMM-DOMO-DATA-LOADED", { data: JSON.parse(this.responseText) } );
      } else {
        Log.error(self.name + ": Could not load data.");
        console.log("Did not load data");
      }
    };
  }
  requestData.open("GET", url, true);
  requestData.send();
},
```

node_helper:
```javascript
socketNotificationReceived: function (notification, payload) {
  self = this;
  if ( notification == "MMM-DOMO-DATA-LOADED") {
    console.log("Send back");
    self.sendSocketNotification("MMM-DOMO-DATA-RECEIVED", payload.data);
  }
},
```

The logging shows:
[21.01.2021 20:05.11.989] [LOG]   Send notification
[21.01.2021 20:05.12.228] [LOG]   Send back
[21.01.2021 20:05.12.547] [LOG]   Send back

So somehow the notification that is send to the helper is received and processed twice. Any suggestion how to prevent this from happening? Or am I missing something? This module is the only module in the config.json.

MichMich, 2021-01-21:
Where do you see that `send notification` log? That is client side so should be in your developers tools console, right? The other 2 in your terminal?

goedh452, 2021-01-21:
That’s correct. I copy pasted them but they come from different places.

MichMich, 2021-01-21:
Ok. Just to be sure: you are sure "send notification" isn't printed twice? Chrome doesnt show multiple logs with the same text. It puts a number (2) in front of it. 

I want to be absolutely sure you aren't sending the notification twice.

goedh452, 2021-01-21:
Yes I’m sure. There is no 2 in front of the log message. Very wise to check! Let me now if I can check anything to help!

khassel, 2021-01-21:
tried to debug this with the `default/newsfeed` module.

Results:
- at the first time loading mm in the browser the event is only logged once in the server logs
- waiting for the next regular reload interval the event is only logged once in the server logs
- refreshing the browser the event is logged twice in the server logs

AFAIK this is expected behaviour because when refreshing the browser the server side callback function is called explicitly when getting the fetcher and in the normal way.

Tested this also with v2.13.0, same behaviour.

@goedh452  can you confirm the result above? If not can you provide a module example?

goedh452, 2021-01-22:
Thanks for your response! I tested it and in the process found my mistake. Normally I code and test directly on the mirror itself, but the past few days I tested it in a browser on the laptop. However, the mirror itself is also operational, so basically there are 2 clients. Both send 1 notification to the helper and the helper (of course) processes both and thus sends 2 notifications back. Now I run MM in server mode and everything is fine.

I appreciate you both took the time to look into this and will continu to support the project!

closed 2021-01-22 by goedh452