MagicMirrorOrg/MagicMirror #2436: bug with electron the mirror wont start

antoinegv7, 2021-01-25:
when i npm start the MagicMirror i get this:
    '/home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js',
    undefined
  ]
}
[25.01.2021 18:44.25.949] [ERROR] MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
[25.01.2021 18:44.25.950] [ERROR] If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
[25.01.2021 18:44.26.258] [LOG]   Launching application.
[25.01.2021 18:44.26.286] [WARN]  (electron) The default value of app.allowRendererProcessReuse is deprecated, it is currently "false".  It will change to be "true" in Electron 9.  
[image]

but it's wired because my config file is all right, the issue is fixed when i take the config file sample and put it in the config file and reboot the rasberry and the n put my personal config file, save it and restart the MagicMirror, evrything is working but as soon as i reboot my rasberry pi it appears again as electron and the magic mirror wont start and also i am connected to internet.
i see that there is that line " code: 'MODULE_NOT_FOUND',
  requireStack: [
'/home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js'
" but i have never tutched electron before and i started having this whole electron issue when i downloaded the MMM-screencast module.
here is my config file if it can help 
[config magic mirror 2.txt](https://github.com/MichMich/MagicMirror/files/5868669/config.magic.mirror.2.txt)

 

[image]

## I'm sure this is a bug

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)

## I'm having troubles installing or configuring MagicMirror

Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

## I found a bug in the MagicMirror installer

If you are facing an issue or found a bug while trying to install MagicMirror via the installer please report it in the respective GitHub repository:
[https://github.com/sdetweil/MagicMirror_scripts](https://github.com/sdetweil/MagicMirror_scripts)

## I found a bug in the MagicMirror Docker image

If you are facing an issue or found a bug while running MagicMirror inside a Docker container please create an issue in the GitHub repository of the MagicMirror Docker image:
[https://github.com/bastilimbach/docker-MagicMirror](https://github.com/bastilimbach/docker-MagicMirror)

---

## I found a bug in MagicMirror

Please make sure to only submit reproducible issues. You can safely remove everything above the dividing line.
When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version _and_ your hardware (Raspberry Pi 2/3, Windows, Mac, Linux, System V UNIX).

**Node Version**: Make sure it's version 8 or later.

**MagicMirror Version**: Please let us now which version of MagicMirror you are running. It can be found in the `package.log` file.

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce**: List the step by step process to reproduce the issue.

**Expected Results**: Describe what you expected to see.

**Actual Results**: Describe what you actually saw.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

rejas, 2021-01-25:
Maybe the sd card is defective? Check if something written to it is still there after a reboot.

antoinegv7, 2021-01-25:
thanks i am going to try that but it is a brand new SanDisk Extreme

antoinegv7, 2021-01-26:
i've tried with others sd cards and i still have the issue.also i realized that i've made a mistake in the description, it's about the MMM-remotecontrol modules and not screencast, it's when the module is installed that the issue is happening, when y remove this module (and just comment it in the config file) the issue is gone

rejas, 2021-01-27:
Could you post your config @antoinegv7 so I can take alook myself ?

antoinegv7, 2021-01-27:
/* Magic Mirror Config Sample
 *
 * By Michael Teeuw https://michaelteeuw.nl
 * MIT Licensed.
 *
 * For more information on how you can configure this file
 * See https://github.com/MichMich/MagicMirror#configuration
 *
 */

var config = {
	//address: "localhost",
	 address: "0.0.0.0",
							// Address to listen on, can be:
							// - "localhost", "127.0.0.1", "::1" to listen on loopback interface
							// - another specific IPv4/6 to listen on a specific interface
							// - "0.0.0.0", "::" to listen on any interface
							// Default, when address config is left out or empty, is "localhost"
	port: 8080,
	basePath: "/", 	// The URL path where MagicMirror is hosted. If you are using a Reverse proxy
					// you must set the sub path here. basePath must end with a /
	ipWhitelist: ["[]", "[]"], 	// Set [] to allow all IP addresses
															// or add a specific IPv4 of 192.168.1.5 :
															// ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.1.5"],
															// or IPv4 range of 192.168.3.0 --> 192.168.3.15 use CIDR format :
															// ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.3.0/28"],

	useHttps: false, 		// Support HTTPS or not, default "false" will use HTTP
	httpsPrivateKey: "", 	// HTTPS private key path, only require when useHttps is true
	httpsCertificate: "", 	// HTTPS Certificate path, only require when useHttps is true

	language: "fr",
	logLevel: ["INFO", "LOG", "WARN", "ERROR"], // Add "DEBUG" for even more logging
	timeFormat: 24,
	units: "metric",
	// serverOnly:  true/false/"local" ,
	// local for armv6l processors, default
	//   starts serveronly and then starts chrome browser
	// false, default for all NON-armv6l devices
	// true, force serveronly mode, because you want to.. no UI on this device

	modules: [
			{
					module: 'MMM-PIR-Sensor',
					config: {
						sensorpin:2,
						powerSavingDelay: 950,
						// See 'Configuration options' for more information.
					}
				}
			]
			/*{
					module: 'MMM-Screencast',
					position: 'bottom_right', // This position is for a hidden <div /> and not the screencast window
					config: {
						position: 'bottomRight',
						height: 300,
						width: 500,
					}
		},*/
		{
			module: "alert",
		},
		{
			module: "updatenotification",
			position: "top_bar"
		},
		{
			module: "clock",
			position: "top_left"
		},
		{
			module: "calendar",
			header: "Calandar",
			position: "top_left",
			config: {
				calendars: [
					{
						symbol: "calendar-check",
						url: "https://calendar.google.com/calendar/ical/antoine.gervais.hamelin%40gmail.com/private-845db19714af8636a2e03882c6544602/basic.ics"					}	
				],
				maximumEntries:7,
				showLocation:true,
			}
		},
		/*{
			module: 'MMM-Remote-Control',
			 uncomment the following line to show the URL of the remote control on the mirror
			// position: 'bottom_left',
			// you can hide this module afterwards from the remote control itself
			config: {
				apiKey: '8893ee613333428a917ca6a93ff6768d'
		},
		},

		/*{
			module: "compliments",
			position: "lower_third"
		},
		{
			module: "currentweather",
			position: "top_right",
			config: {
				location: "New York",
				locationID: "5128581", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
				appid: "YOUR_OPENWEATHER_API_KEY"
			}
		},
		{
			module: "weatherforecast",
			position: "top_right",
			header: "Weather Forecast",
			config: {
				location: "New York",
				locationID: "5128581", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
				appid: "YOUR_OPENWEATHER_API_KEY"
			}
		},*/
		{
			disabled: false,
			module: 'MMM-WeatherOrNot',
			position: 'top_right',
			config: {
			  location: "le-mans",                // See instructions
			  locationCode: "48d010d20",              // See instructions
			  languages: "fr",                          // See Languages list
			  tempUnits: "C",                           // F or C
			  font: "Tahoma",                           // See Font list
			  textColor: "#ffffff",                     // Hex color codes.
			  htColor: "#ffffff",                       // high temp color. Hex color codes.
			  ltColor: "#00dfff",                       // low temp color. Hex color codes.
			  sunColor: "#febc2f",                      // Hex color codes.
			  moonColor: "#dfdede",                     // Hex color codes.
			  cloudColor: "#dfdede",                    // Hex color codes.
			  cloudFill: "#1f567c",                     // Hex color codes.
			  rainColor: "#93bffe",                     // Hex color codes.
			  snowColor: "#dfdede",                     // Hex color codes.
			  height: "5750px",                          // module is responsive to changes
			  width: "250px",                          // module is responsive to changes
			  label: "Le Mans",                        // Location seems logical . .
			  label2: "Sarthe",                  // . . . or anything you like
			  days: "7",                                // 3, 5 or 7
			  theme: "Retro Sky",                           // See Themes list *** theme overrides bgColor. ***
			  bgColor: "#000000",                       // theme overrides bgColor.
			  icons: "Climacons Animated",              // Iconvault, Climacons or Climacons Animated
			  animationSpeed: 3500,
			  updateInterval: 10 * 60 * 1000,
			}
		},
		{
			module: "newsfeed",
			position: "bottom_bar",
			config: {
				feeds: [
					{
						title: "Le Monde",
						url: "https://www.lemonde.fr/international/rss_full.xml"
					},
					{
						title: "Ouest France",
						url: "https://www.ouest-france.fr/rss.xml" 
					},
				],
				showSourceTitle: true,
				showPublishDate: true,
				broadcastNewsFeeds: true,
				broadcastNewsUpdates: true
			}
		},
	]
};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}

antoinegv7, 2021-01-27:
[image]
[image]
[config magic mirror 2.txt](https://github.com/MichMich/MagicMirror/files/5882095/config.magic.mirror.2.txt)
[image]

antoinegv7, 2021-01-27:
/* Magic Mirror Config Sample
 *
 * By Michael Teeuw https://michaelteeuw.nl
 * MIT Licensed.
 *
 * For more information on how you can configure this file
 * See https://github.com/MichMich/MagicMirror#configuration
 *
 */

var config = {
	//address: "localhost",
	 address: "0.0.0.0",
							// Address to listen on, can be:
							// - "localhost", "127.0.0.1", "::1" to listen on loopback interface
							// - another specific IPv4/6 to listen on a specific interface
							// - "0.0.0.0", "::" to listen on any interface
							// Default, when address config is left out or empty, is "localhost"
	port: 8080,
	basePath: "/", 	// The URL path where MagicMirror is hosted. If you are using a Reverse proxy
					// you must set the sub path here. basePath must end with a /
	ipWhitelist: ["[]", "[]"], 	// Set [] to allow all IP addresses
															// or add a specific IPv4 of 192.168.1.5 :
															// ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.1.5"],
															// or IPv4 range of 192.168.3.0 --> 192.168.3.15 use CIDR format :
															// ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.3.0/28"],

	useHttps: false, 		// Support HTTPS or not, default "false" will use HTTP
	httpsPrivateKey: "", 	// HTTPS private key path, only require when useHttps is true
	httpsCertificate: "", 	// HTTPS Certificate path, only require when useHttps is true

	language: "fr",
	logLevel: ["INFO", "LOG", "WARN", "ERROR"], // Add "DEBUG" for even more logging
	timeFormat: 24,
	units: "metric",
	// serverOnly:  true/false/"local" ,
	// local for armv6l processors, default
	//   starts serveronly and then starts chrome browser
	// false, default for all NON-armv6l devices
	// true, force serveronly mode, because you want to.. no UI on this device

	modules: [
			/*{
					module: 'MMM-PIR-Sensor',
					config: {
						sensorpin:2,
						powerSavingDelay: 950,
						// See 'Configuration options' for more information.
					}
				}
			]
			/*{
					module: 'MMM-Screencast',
					position: 'bottom_right', // This position is for a hidden <div /> and not the screencast window
					config: {
						position: 'bottomRight',
						height: 300,
						width: 500,
					}
		},*/
		{
			module: "alert",
		},
		{
			module: "updatenotification",
			position: "top_bar"
		},
		{
			module: "clock",
			position: "top_left"
		},
		{
			module: "calendar",
			header: "Calandar",
			position: "top_left",
			config: {
				calendars: [
					{
						symbol: "calendar-check",
						url: "https://calendar.google.com/calendar/ical/antoine.gervais.hamelin%40gmail.com/private-845db19714af8636a2e03882c6544602/basic.ics"					}	
				],
				maximumEntries:7,
				showLocation:true,
			}
		},
		{
			module: 'MMM-Remote-Control',
			// uncomment the following line to show the URL of the remote control on the mirror
			 position: 'bottom_left',
			// you can hide this module afterwards from the remote control itself
			config: {
				apiKey: '8893ee613333428a917ca6a93ff6768d'
		},
		},

		/*{
			module: "compliments",
			position: "lower_third"
		},
		{
			module: "currentweather",
			position: "top_right",
			config: {
				location: "New York",
				locationID: "5128581", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
				appid: "YOUR_OPENWEATHER_API_KEY"
			}
		},
		{
			module: "weatherforecast",
			position: "top_right",
			header: "Weather Forecast",
			config: {
				location: "New York",
				locationID: "5128581", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
				appid: "YOUR_OPENWEATHER_API_KEY"
			}
		},*/
		{
			disabled: false,
			module: 'MMM-WeatherOrNot',
			position: 'top_right',
			config: {
			  location: "le-mans",                // See instructions
			  locationCode: "48d010d20",              // See instructions
			  languages: "fr",                          // See Languages list
			  tempUnits: "C",                           // F or C
			  font: "Tahoma",                           // See Font list
			  textColor: "#ffffff",                     // Hex color codes.
			  htColor: "#ffffff",                       // high temp color. Hex color codes.
			  ltColor: "#00dfff",                       // low temp color. Hex color codes.
			  sunColor: "#febc2f",                      // Hex color codes.
			  moonColor: "#dfdede",                     // Hex color codes.
			  cloudColor: "#dfdede",                    // Hex color codes.
			  cloudFill: "#1f567c",                     // Hex color codes.
			  rainColor: "#93bffe",                     // Hex color codes.
			  snowColor: "#dfdede",                     // Hex color codes.
			  height: "5750px",                          // module is responsive to changes
			  width: "250px",                          // module is responsive to changes
			  label: "Le Mans",                        // Location seems logical . .
			  label2: "Sarthe",                  // . . . or anything you like
			  days: "7",                                // 3, 5 or 7
			  theme: "Retro Sky",                           // See Themes list *** theme overrides bgColor. ***
			  bgColor: "#000000",                       // theme overrides bgColor.
			  icons: "Climacons Animated",              // Iconvault, Climacons or Climacons Animated
			  animationSpeed: 3500,
			  updateInterval: 10 * 60 * 1000,
			}
		},
		{
			module: "newsfeed",
			position: "bottom_bar",
			config: {
				feeds: [
					{
						title: "Le Monde",
						url: "https://www.lemonde.fr/international/rss_full.xml"
					},
					{
						title: "Ouest France",
						url: "https://www.ouest-france.fr/rss.xml" 
					},
				],
				showSourceTitle: true,
				showPublishDate: true,
				broadcastNewsFeeds: true,
				broadcastNewsUpdates: true
			}
		},
	]
};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}

rejas, 2021-01-27:
This config  invalid, please run "npm run config:check" in your MagicMirror directory to see the error

antoinegv7, 2021-01-27:
[image]

khassel, 2021-01-27:
replace `ipWhitelist: ["[]", "[]"], // Set [] to allow all IP addresses` with `ipWhitelist: [],` as already mentioned in the comment of this line.

antoinegv7, 2021-01-27:
ti loocks like the issue is fixed

antoinegv7, 2021-01-27:
thanks verry much

antoinegv7, 2021-01-30:
i've tried but it does not work, also here is my autostart file if it can help
[image]
[image]

[image]

rejas, 2021-01-30:
Have you run "npm install" in the mmm-pi-sensor directory?

antoinegv7, 2021-01-30:
it's wired,  i still have a black screen with "This device is not allowed to access your mirror.
Please check your config.js or config.js.sample to change this." but here is my startup of the mirror wich looks good:
npm start

> magicmirror@2.14.0 start /home/pi/MagicMirror
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

[30.01.2021 19:27.54.057] [LOG]   Starting MagicMirror: v2.14.0
[30.01.2021 19:27.54.063] [LOG]   Loading config ...
[30.01.2021 19:27.54.069] [LOG]   Loading module helpers ...
[30.01.2021 19:27.54.072] [LOG]   No helper found for module: alert.
[30.01.2021 19:27.54.208] [LOG]   Initializing new module helper ...
[30.01.2021 19:27.54.209] [LOG]   Module helper loaded: updatenotification
[30.01.2021 19:27.54.210] [LOG]   No helper found for module: clock.
[30.01.2021 19:27.54.595] [LOG]   Initializing new module helper ...
[30.01.2021 19:27.54.596] [LOG]   Module helper loaded: calendar
[30.01.2021 19:27.54.597] [LOG]   No helper found for module: MMM-Remote-Control.
[30.01.2021 19:27.54.598] [LOG]   No helper found for module: MMM-WeatherOrNot.
[30.01.2021 19:27.54.619] [LOG]   Initializing new module helper ...
[30.01.2021 19:27.54.620] [LOG]   Module helper loaded: newsfeed
[30.01.2021 19:27.56.478] [LOG]   Initializing new module helper ...
[30.01.2021 19:27.56.479] [LOG]   Module helper loaded: MMM-GoogleAssistant
[30.01.2021 19:27.56.480] [LOG]   No helper found for module: MMM-Assistant2Display.
[30.01.2021 19:27.56.480] [LOG]   All module helpers loaded.
[30.01.2021 19:27.56.566] [LOG]   Starting server on port 8080 ... 
[30.01.2021 19:27.56.574] [LOG]   Server started ...
[30.01.2021 19:27.56.575] [LOG]   Connecting socket for: updatenotification
[30.01.2021 19:27.56.576] [LOG]   Connecting socket for: calendar
[30.01.2021 19:27.56.576] [LOG]   Starting node helper for: calendar
[30.01.2021 19:27.56.577] [LOG]   Connecting socket for: newsfeed
[30.01.2021 19:27.56.578] [LOG]   Starting node helper for: newsfeed
[30.01.2021 19:27.56.579] [LOG]   Connecting socket for: MMM-GoogleAssistant
[30.01.2021 19:27.56.580] [LOG]   Sockets connected & modules started ...
[30.01.2021 19:27.56.589] [ERROR] Whoops! There was an uncaught exception...
[30.01.2021 19:27.56.595] [ERROR] Error: listen EADDRINUSE: address already in use 0.0.0.0:8080
    at Server.setupListenHandle [as _listen2] (net.js:1300:14)
    at listenInCluster (net.js:1348:12)
    at doListen (net.js:1487:7)
    at processTicksAndRejections (internal/process/task_queues.js:81:21) {
  code: 'EADDRINUSE',
  errno: 'EADDRINUSE',
  syscall: 'listen',
  address: '0.0.0.0',
  port: 8080
}
[30.01.2021 19:27.56.597] [ERROR] MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
[30.01.2021 19:27.56.597] [ERROR] If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
[30.01.2021 19:27.56.882] [LOG]   Launching application.
[30.01.2021 19:27.56.926] [WARN]  (electron) The default value of app.allowRendererProcessReuse is deprecated, it is currently "false".  It will change to be "true" in Electron 9.  For more information please check https://github.com/electron/electron/issues/18397
[image]

antoinegv7, 2021-01-30:
for the pir sensor yes

antoinegv7, 2021-01-30:
even when i run the sample config file i have this
[image]

antoinegv7, 2021-01-31:
i think that it comes from the autostart

rejas, 2021-01-31:
>  [30.01.2021 19:27.56.595] [ERROR] Error: listen EADDRINUSE: address already in use 0.0.0.0:8080

Is there already an instance of MM running that blocks that port?

Anyway, I do think that you can find better help on how to configure your Mirror on the official forum https://forum.magicmirror.builders/

MichMich, 2021-01-31:
Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2021-01-31 by MichMich

antoinegv7, 2021-01-31:
i cant post any issues, it tels me to login to be able to post but when logged in i cant post there is not any button

antoinegv7, 2021-02-05:
!