MagicMirrorOrg/MagicMirror #2441: forum having trouble. 500 error

sdetweil, 2021-01-29:
## I'm not sure if this is a bug

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)

## I'm having troubles installing or configuring MagicMirror

Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

## I found a bug in the MagicMirror installer

If you are facing an issue or found a bug while trying to install MagicMirror via the installer please report it in the respective GitHub repository:
[https://github.com/sdetweil/MagicMirror_scripts](https://github.com/sdetweil/MagicMirror_scripts)

## I found a bug in the MagicMirror Docker image

If you are facing an issue or found a bug while running MagicMirror inside a Docker container please create an issue in the GitHub repository of the MagicMirror Docker image:
[https://github.com/bastilimbach/docker-MagicMirror](https://github.com/bastilimbach/docker-MagicMirror)

---

## I found a bug in MagicMirror

Please make sure to only submit reproducible issues. You can safely remove everything above the dividing line.
When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version _and_ your hardware (Raspberry Pi 2/3, Windows, Mac, Linux, System V UNIX).

**Node Version**: Make sure it's version 8 or later.

**MagicMirror Version**: Please let us now which version of MagicMirror you are running. It can be found in the `package.log` file.

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce**: List the step by step process to reproduce the issue.

**Expected Results**: Describe what you expected to see.

**Actual Results**: Describe what you actually saw.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

memoriasIT, 2021-01-29:
I am able to reach the forum but not the main page.
However, the forum is unusable for the most part.

Console spits out this error:
Firefox can’t establish a connection to the server at wss://forum.magicmirror.builders/socket.io/?EIO=3&transport=websocket&sid=<sid>

MichMich, 2021-01-30:
Thanks for the report. Could you check if it's any better now (after a refresh)?

memoriasIT, 2021-01-30:
Still not working, and now I can't even reach it. Also getting a error 500.
Also, nginx is leaking the version. I think this should be patched as it can be useful for hackers. The patch is very easy.

MichMich, 2021-01-30:
What is the full url you are trying to load?

MichMich, 2021-01-30:
Will do some maintenance. But this might mean the forum is down for the coming hour. Hope this will solve the issue afterwards. "ll let you know when it is done.

memoriasIT, 2021-01-30:
Sure, no pressure, I was just trying to help :)

MichMich, 2021-01-30:
I've updated Node & NodeBB. Let me know if it solves the issue.

khassel, 2021-01-30:
Forum is up and running again 👍

sdetweil, 2021-01-31:
different color themes not available. some chats appear to be hung.

MichMich, 2021-02-02:
Themes should now be available again, @sdetweil! Can you give some more info on the chats issue?

sdetweil, 2021-02-02:
dont know for sure. think it might have been themes. couldnt see buttons. looks like its cleared now

MichMich, 2021-02-02:
Good to hear! Can I close this issue for now?

sdetweil, 2021-02-02:
no, sorry, we cant upload pics now, get 'invalid files'

sdetweil, 2021-02-05:
new user gets error regiatring on forum. see discord support channel. also sound from forum not working. contact mykle1

MichMich, 2021-02-06:
The registration issue is solved. Still working on the image issue.

MichMich, 2021-02-06:
Images are working as well. Any other issues?

sdetweil, 2021-02-08:
github registration on forum not sending verify email. see discord support

MomentumV, 2021-02-08:
I experienced a lack of verify email when registering with github account. Comments were made in Discord General Channel. I successfully registerd an account by email registration using my same gmail account without the period in it. Not sure if that's relevant.

closed 2021-03-04 by sdetweil