MagicMirrorOrg/MagicMirror #2453: Enhancement suggestion: Multiple instances of the same module

pierrephi, 2021-02-10:
Hello, 

Not a bug but a feature suggestion, managing several pages (MMM-pages) on my magic mirror, I find the need to have a module behaving differently on the different pages (eg large display on the home screen, compact mode in another)

In the forums, a recurrent solution is to clone the modules to a different module directory with a different name, fiddle with the js/css file name and possibly other files depending on the module. Not really satisfactory as updating the module as it evolves becomes cumbersome.

I was thinking of something like 

```
		{
			module: "text-clock-homescreen",
			moduleOnFile: "MMM-text-clock",
			position: "middle_center",
			config: {
				compact: false,
				size: "large",
				language: ['jp','fr','en','es'],
				languageAlternationInterval: 10
			}
		}, 
		{
			module: "text-clock-compact",
			moduleOnFile: "MMM-text-clock",
			position: "top_center",
			config: {
				compact: true,
				size: "medium",
				language: ['jp','fr','en','es'],
				languageAlternationInterval: 10
			}
		}, 
```
I haven't looked at the internals of MagicMirror but just wanted to run this by the active community to see if this made sense at all to others or had already been considered and discarded.

Happy to contribute to this of course.

MichMich, 2021-02-10:
Afaik this is already possible, but depends on the implementation of MMM-pages (not a core module). You can add the module multiple times to your config with different configs, and use the classes property to assign it to different pages.

MichMich, 2021-02-10:
Try this implementation:
```
		{
			module: "MMM-text-clock",
			classes: "text-clock-homescreen",
			position: "middle_center",
			config: {
				compact: false,
				size: "large",
				language: ['jp','fr','en','es'],
				languageAlternationInterval: 10
			}
		}, 
		{
			module: "MMM-text-clock",
			classes: "text-clock-compact",
			position: "top_center",
			config: {
				compact: true,
				size: "medium",
				language: ['jp','fr','en','es'],
				languageAlternationInterval: 10
			}
		}, 
```

pierrephi, 2021-02-10:
Thanks for that swift answer @MichMich , looks like exactly what I need, I will check at enhancing MMM-pages then!

MichMich, 2021-02-10:
This should be possible with the current mmm-pages version. You only need to alter your config.

sdetweil, 2021-02-10:
it really is up to the module to implement support for multiple instances. not hard, but does require some coding change

pierrephi, 2021-02-10:
Thanks @michmich was right it work out of the box with classes in MMM-pages I've got it up and running now :)

closed 2021-02-10 by MichMich