MagicMirrorOrg/MagicMirror #2455: I get error when try to run the magic mirror at first time

CherryChethan, 2021-02-12:
after copy config.js file i get this error 
When am give this code npm run start i get the error like this bellow the maic mirror is not open

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! magicmirror@2.14.0 start: `DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the magicmirror@2.14.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2021-02-12T10_46_13_145Z-debug.log

rejas, 2021-02-12:
Hi @CherryChethanm did you copy the sample config.js?

JaytheMCgamer, 2021-02-12:
hmmm.Looks like the start script is broken.Try re-installing MagicMirror

Smiche, 2021-02-12:
If you're running this on windows remove  `DISPLAY="${DISPLAY:=:0}"` from package.json start script

CherryChethan, 2021-02-12:
@rejas

> Hi @CherryChethanm did you copy the sample config.js?

yes i did that all the steps

CherryChethan, 2021-02-12:
> hmmm.Looks like the start script is broken. Try re-installing MagicMirror

I did this one also and I try a lot of time but it's not working

CherryChethan, 2021-02-12:
> If you're running this on windows remove `DISPLAY="${DISPLAY:=:0}"` from package.json start script

no M NOT running on the windows now am flash the sd card again am instaling the raspberry os bellow you are seeing know as the same thing
Raspberry Pi OS with desktop and recommended software
Release date: January 11th 2021
Kernel version: 5.4
Size: 2,863MB
Show SHA256 file integrity hash:

Smiche, 2021-02-12:
Can you provide the contents of  the debug log: /home/pi/.npm/_logs/2021-02-12T10_46_13_145Z-debug.log

sdetweil, 2021-02-12:
what pi is this? pi 0 has no electron version anymore.

the system is running in graphical desktop mode  correct?

is xwindows running, do this command
pgrep Xorg
should return a number

CherryChethan, 2021-02-13:
> pgrep Xorg

Where should i run this code

CherryChethan, 2021-02-13:
> Can you provide the contents of the debug log: /home/pi/.npm/_logs/2021-02-12T10_46_13_145Z-debug.log

0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'run', 'start' ]
2 info using npm@6.14.11
3 info using node@v10.23.3
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle magicmirror@2.14.0~prestart: magicmirror@2.14.0
6 info lifecycle magicmirror@2.14.0~start: magicmirror@2.14.0
7 verbose lifecycle magicmirror@2.14.0~start: unsafe-perm in lifecycle true
8 verbose lifecycle magicmirror@2.14.0~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/home/pi/MagicMirror/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
9 verbose lifecycle magicmirror@2.14.0~start: CWD: /home/pi/MagicMirror
10 silly lifecycle magicmirror@2.14.0~start: Args: [ '-c',
10 silly lifecycle   'DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js' ]
11 silly lifecycle magicmirror@2.14.0~start: Returned: code: 1  signal: null
12 info lifecycle magicmirror@2.14.0~start: Failed to exec start script
13 verbose stack Error: magicmirror@2.14.0 start: `DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
13 verbose stack     at EventEmitter.emit (events.js:198:13)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:198:13)
13 verbose stack     at maybeClose (internal/child_process.js:982:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
14 verbose pkgid magicmirror@2.14.0
15 verbose cwd /home/pi/MagicMirror
16 verbose Linux 5.4.51-v7l+
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "run" "start"
18 verbose node v10.23.3
19 verbose npm  v6.14.11
20 error code ELIFECYCLE
21 error errno 1
22 error magicmirror@2.14.0 start: `DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js`
22 error Exit status 1
23 error Failed at the magicmirror@2.14.0 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]

sdetweil, 2021-02-13:
pgrep Xorg

open a terminal window, or ssh session

same place u would run npm start

CherryChethan, 2021-02-13:
> open a terminal window, or ssh session
pi@raspberrypi:~ $ cd MagicMirror/
pi@raspberrypi:~/MagicMirror $ pgrep xorg
pi@raspberrypi:~/MagicMirror $ pgrep Xorg
533
pi@raspberrypi:~/MagicMirror $

aidynme, 2021-03-10:
the same problem,
any info?

MichMich, 2021-03-13:
Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2021-03-13 by MichMich

EiderDiaz, 2021-05-17:
i have the same issue, when i npm run start 
- im using a raspberry pi 3 B+
- i install the recomended OS 
- thanks in advance 
i get the error :
`> magicmirror@2.15.0 start /home/pi/MagicMirror
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

/home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGSEGV
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! magicmirror@2.15.0 start: `DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the magicmirror@2.15.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2021-05-17T08_32_19_264Z-debug.log
`

log file: 
`0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'run', 'start' ]
2 info using npm@6.14.13
3 info using node@v14.17.0
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle magicmirror@2.15.0~prestart: magicmirror@2.15.0
6 info lifecycle magicmirror@2.15.0~start: magicmirror@2.15.0
7 verbose lifecycle magicmirror@2.15.0~start: unsafe-perm in lifecycle true
8 verbose lifecycle magicmirror@2.15.0~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/home/pi/MagicMirror/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
9 verbose lifecycle magicmirror@2.15.0~start: CWD: /home/pi/MagicMirror
10 silly lifecycle magicmirror@2.15.0~start: Args: [
10 silly lifecycle   '-c',
10 silly lifecycle   'DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js'
10 silly lifecycle ]
11 silly lifecycle magicmirror@2.15.0~start: Returned: code: 1  signal: null
12 info lifecycle magicmirror@2.15.0~start: Failed to exec start script
13 verbose stack Error: magicmirror@2.15.0 start: `DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
13 verbose stack     at EventEmitter.emit (events.js:376:20)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:376:20)
13 verbose stack     at maybeClose (internal/child_process.js:1055:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
14 verbose pkgid magicmirror@2.15.0
15 verbose cwd /home/pi/MagicMirror
16 verbose Linux 5.10.17-v7+
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "run" "start"
18 verbose node v14.17.0
19 verbose npm  v6.14.13
20 error code ELIFECYCLE
21 error errno 1
22 error magicmirror@2.15.0 start: `DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js`
22 error Exit status 1
23 error Failed at the magicmirror@2.15.0 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]`

EiderDiaz, 2021-05-17:
@MichMich i do not see any solution to the problem here, maybe you should consider to reopen the issue :)

MichMich, 2021-05-17:
As said: please refer to http://forum.magicmirror.builders for support and troubleshooting.