MagicMirrorOrg/MagicMirror #2456: I get error in MMM - google Assistance

CherryChethan, 2021-02-13:
@MichMich 
Hope all are doing well i get this error can any one help me i as facing this issues from last 4 days i try clear it but i dont know and am fresher to this concept please resolve this
 
when am run the mmm google assistance i get this error
after complet all steps when am give this cmd

```
cd ~/MagicMirror/modules/MMM-GoogleAssistant
node auth_and_test.js
```

internal/modules/cjs/loader.js:638
throw err;
^

Error: Cannot find module ‘@bugsounet/google-assistant’
at Function.Module._resolveFilename (internal/modules/cjs/loader.js:636:15)
at Function.Module._load (internal/modules/cjs/loader.js:562:25)
at Module.require (internal/modules/cjs/loader.js:692:17)
at require (internal/modules/cjs/helpers.js:25:18)
at Object. (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/auth_and_test.js:7:25)
at Module._compile (internal/modules/cjs/loader.js:778:30)
at Object.Module._extensions…js (internal/modules/cjs/loader.js:789:10)
at Module.load (internal/modules/cjs/loader.js:653:32)
at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
at Function.Module._load (internal/modules/cjs/loader.js:585:3)
@MichMich @bugsounet  please help me guys

bugsounet, 2021-02-13:
Hi, this issue is not related to MagicMirror but on my GA module

Please open an issue in @bugsounet modules forum [there](http://forum.bugsounet.fr) or in [MagicMirror Forum](https://forum.magicmirror.builders)

Note: i don't help in MagicMirror Forum. Only on my forum

@bugsounet

CherryChethan, 2021-02-13:
It showing some error i already post in that this is the url of that please check and give the solution
[](https://forum.magicmirror.builders/topic/14614/mmm-google-assistant-i-get-error)

@bugsounet

bugsounet, 2021-02-13:
> Note: i don't help in MagicMirror Forum. Only on my forum
> 
> @bugsounet

CherryChethan, 2021-02-13:
how can i open the issues in your @bugsounet forum can you send open issues to link please if you don't mine

MichMich, 2021-02-13:
As stated this is not a core issue. Closing.

closed 2021-02-13 by MichMich