MagicMirrorOrg/MagicMirror #2459: Current weather only display sunrise and never sunset

mcnahum, 2021-02-16:
I had to fully re-install my MagicMirror and I notice that the current Weather module is only displaying the sunrise time and never the sunset one... 
in my config I just have the location / location ID / appid 
did I forgot something?

MichMich, 2021-02-16:
We will switch to the new weather modules in the next release. Could you check if those solve your issue?

mcnahum, 2021-02-16:
sure, just let me know how to test it

rejas, 2021-04-17:
Hi @mcnahum sorry for the late response. You can use the new weather modules for the current weather with a config like this:

		{
			module: "weather",
			position: "top_left",
			config: {
				type: "current",
				location: "Berlin",
				locationID: "xxxxxxxxxx",  //ID from http://www.openweathermap.org/help/city_list.txt
				apiKey: "xxxxxx"
			}
		}

Have you upgraded to v2.15 yet?

mcnahum, 2021-04-25:
I just updated to 2.15 ( not sure why the module didn't told me there was a new version)
let see if the sunrise / sunset stay stable

closed 2021-07-08 by stale[bot]