MagicMirrorOrg/MagicMirror #2462: Add ability to hide time in clock module

prapicault, 2021-02-20:
Hi, I would like for the clock module to have the ability to hide the time (so it only displays the date). I have already changed the code in my installation but it makes keeping up with master a bit annoying. 
Would you be interested in a contribution that would add a `showTime` configuration option to the default clock module? This option when set to `false` would not render the time at all.

sdetweil, 2021-02-20:
you could use css to turn off the time by overriding the class set to disolay:none;

prapicault, 2021-02-20:
I probably could (though tbh, it did not even occur to be me try) but it does not seem to be as discoverable as a configuration for other users.

MichMich, 2021-02-23:
Feel free to send a PR. 👍🏻

rejas, 2021-04-18:
I looked into the clock module to see if hiding the time was possible.
Yes it is, but the module is loaded with different options for all kind of things which made me think if it could be reworked a little. But then one or two options might change so I wanted forst to ask @MichMich what the official stance on removing / chaning options in the default module is?

MichMich, 2021-04-18:
I don't think we should overbloat the clock module (I think it already is a bit). If someone really wants super specific features, it's better to opt for a third party module. 

I'm not a big fan for removing options and essentially introducing breaking changes.

rejas, 2022-10-03:
@prapicault @AndyHazz can this issue be closed since your problems should be fixed in the last year ;-) ? if yes @MichMich can do it

AndyHazz, 2022-10-03:
Fine by me, I've had no issues since switching to use showDate: false

PR #2641 (2021-08-31): Actually use showTime parameter in clock module
Actually implements and fixes #2462. 

My first PR (#2627) was faulty with regards to the new parameter. Only found out after checking back on the issue :-(

Now showTime is set to true as per default and disabling it hide it when the digital clock is shown.
closed 2022-10-06 by khassel