MagicMirrorOrg/MagicMirror #2467: How can i add my Nextcloud Calendars?

poltrian, 2021-02-27:
Nowadays it's good to have your data at home and not on Google or Apple. That's why I have a Nextcloud at home. How can I display the appointments in Magic Mirror?

sdetweil, 2021-02-27:
you will have to download them to a local file... 
lots of folks do this..
i know the forums have people doing this..
https://forum.magicmirror.builders

poltrian, 2021-02-27:
I tried these, put the local path in the cinfig file, but the ics would not be shown. Most online calendars have no ics, most you gent a link like http:zztztt.com/caldav.php/user/8787765654 or http:zztztt.com/caldav.php/user/calendarname

No other solution possible?

sdetweil, 2021-02-27:
local path must be in MagicMirror folder tree.

mm web server root is the mm foldr  so /modules...... whereever.. it works..

of fike:///....

poltrian, 2021-02-27:
Thanks, these is working very well 

url: "https://nextcloud.MYDOMAIN.xyz/remote.php/dav/calendars/USER/personal/?export",
user: "USER",
pass: "PASSW",
method: "basic",

closed 2021-03-03 by MichMich